DIGImend/digimend-kernel-drivers #392: Huion H1060P in openSuse

Giorgio-IT, 2020-05-16:
I installed v9 driver in openSuse Leap 15.1 using dkms but my Huion H1060P doesn't works at all. 
This is what I get in terminal during installation:

`dkms add .

Creating symlink /var/lib/dkms/digimend/9/source ->
                 /usr/src/digimend-9

DKMS: add completed.
dkms build digimend/9

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.12.14-lp151.28.48-default KVERSION=4.12.14-lp151.28.48-default...
cleaning build area...

DKMS: build completed.
dkms install digimend/9

hid-kye.ko:
Running module version sanity check.
Error! Module version 9 for hid-kye.ko
is not newer than what is already found in kernel 4.12.14-lp151.28.48-default (260609848385A8CF1FD1F11).
You may override by specifying --force.

hid-uclogic.ko:
Running module version sanity check.
Error! Module version 9 for hid-uclogic.ko
is not newer than what is already found in kernel 4.12.14-lp151.28.48-default (AA1A15BA9F5974F2B544EE4).
You may override by specifying --force.

hid-polostar.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.12.14-lp151.28.48-default/updates/

hid-viewsonic.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.12.14-lp151.28.48-default/updates/

depmod....

DKMS: install completed.
install -D -m 0755 hid-rebind /lib/udev/hid-rebind
install -D -m 0644 udev.rules /lib/udev/rules.d/90-digimend.rules
install -D -m 0644 xorg.conf /usr/share/X11/xorg.conf.d/50-digimend.conf
udevadm control --reload`

kgozdziewski, 2020-05-16:
Hi, 

I have this tablet (H1060P) partly working with the wacom kernel driver (my kernel is 5.4.17). There are issues, however. The working area is restricted to  32768x32768 points, though  the xsetwacom utility recognizes the tablet and makes it possible to change this parameter to the native 33020x50800 points. 

The tablet appears as in logs as
hid-generic 0003:256C:006D.0009: hiddev2: USB HID v1.11 Device [HID 256c:006d] on usb-0000:15:00.0-2/input0

I have red somewhere that the digimend drivers 9 may depend on the USB configuration, and some people report that one downgraded to version 8, the tablet may work (I did not check this).

Regards,

spbnick, 2020-05-17:
@Giorgio-IT, this is a known issue, please see https://github.com/DIGImend/digimend-kernel-drivers#dkms-issues-preventing-correct-installation

@kgozdziewski, it looks like the digimend drivers are not handling your tablet (and Wacom kernel drivers won't work with your tablet at all). Please open a separate issue, as yours is unrelated to the one @Giorgio-IT reports, and somebody will help you.

Closing as a known issue with a workaround.

closed 2020-05-17 by spbnick

Giorgio-IT, 2020-05-19:
Thanks for answer.
I run the following command:
`sudo sed -i \
         -e '/^get_module_verinfo()/,+3 s/\<unset res$\|\<res=()$/res=("" "" "")/' \
         /usr/sbin/dkms`
that's the only fix I'm able to apply but tablet don't works.
Removed and reinstalled driver don't change, tablet still not working.