DIGImend/digimend-kernel-drivers #395: Huion 1060P works with USB2.0 but not USB3.0, linux kernel 5.4.x

kgozdziewski, 2020-05-19:
I have a few comments, since there is a lot of confusion regarding the Huion H1060P tablet, which is missing in the documentation.  This tablet is often confused with H1060Plus. 

I spent some time figuring out how to make the H1060P working.

Actually, the tablet works with the latest digimend-kernel drivers. I had to disable USB3.0 in my Bios however (this is a know issue reported in many places, regarding newer kernels, 5.x), and  to remove the Wacom module from my kernel. I installed the xwacom utility and xorg driver, as advised on the occasion of a few Huion models working with the Wacom driver:

dev-libs/libwacom
x11-drivers/xf86-input-wacom

Then I downloaded the github digimend-driver, compiled it and loaded the modules. After that the tablet appears as  Nuvoton WPM USB, 256c:006d.

The digmend driver properly sets the resolution spanning the whole working area (unlike the wacom driver, which reduced it to roughly a half), and at least 6 buttons may be re/defined with the xsetwacom utility, such as

xsetwacom --set "HID 256c:006d Pad pad" Button  1 "key ctrl z"

I still have only some problem to define the pressure curve properly but at least the pressure seems to be recognised. I found it working with gimp and xournal/xournalpp.

Hope this helps. The USB3.0 issue is still present, but it may be somewhat reduced.

Many thanks for the driver, it's great that such a fine hardware may be used with linux.

spbnick, 2020-05-19:
Thank you for the report, @kgozdziewski! Just have to add that the USB3.0 issue might be fixed in the latest kernel releases, and the kernel Wacom driver has nothing to do with your or any non-Wacom tablets, so no need to remove it. What's handling your tablet in the absence of a specialized driver is the generic HID driver (`hid-generic`). It's trying to do its best with however the tablet describes itself according to the (rather vague) HID specification, with such descriptions being quite poor as well normally.

kgozdziewski, 2020-05-19:
Nikolai, many thanks for your support and time, now I understand.

bsamorodov, 2020-06-03:
Untill the kernel is fixed one may be interested at the [workaround](https://github.com/DIGImend/digimend-kernel-drivers/issues/405#issuecomment-637844659).

HTH, bsam