DIGImend/digimend-kernel-drivers #396: Huion Q620M gets rocognized as mouse not as stylus/touch device

ticotaco72, 2020-05-21:
Hello,
I installed digimend from source using DKMS on Fedora 31 KDE.
I don't know if this is intended, but much of software behave with it like a mouse. Krita works with pen pressure ( i don't know how to check if tilt works). Xournal++ doesn't work with pen pressure nor tilt.
In other software tablet works like a mouse too. E.g. in Firefox when i touch and swipe through page, it selects text instead of scrolling the page down.
I would expect wheel to work in the second direction, but it doesn't matter.

Anyway big thanks for this amazing work.

ticotaco72, 2020-05-21:
hhhh

Code
Issues 165

lparolari, 2020-12-23:
Any news on this?

ticotaco72, 2023-11-23:
Device gets recognised as a stylus tablet and also this issue probably shouldn't have been reported right here.

closed 2023-11-23 by ticotaco72