DIGImend/digimend-kernel-drivers #398: Huion Kamvas 13 (2020) issues

darkbeanbbq, 2020-05-21:
Hi, I've recently acquired the new [Huion Kamvas 13](https://www.huion.com/pen_display/Kamvas/kamvas-13.html) (2020) Pen display tablet and in my attempt to configure it and to contribute to the project with its information, here I am.
I'm creating a new issue because the Kamvas 13 (2020) differs from the Kamvas pro 13 talked about in issue #214 (different pen, and different ID). 

So it "worked" out of the box on a week old Ubuntu 20.04 install. By "worked" I mean that the stylus makes the cursor move correctly and the pressure sensor is working, but the tablets buttons don't work (the pen's buttons do) and it frequently "disconnects" and I have to turn it off and on again (often sometimes several times) to get it to work again and show up in `xinput list`.
Sometimes it works great for an hour or two, and sometimes it just can't stop disconnecting.

So in my quest to make it fully functional I installed the digimend drivers from source (because of the #387 issue on Ubuntu 20.04). Since then nothing really changed except that my tablet changed names in `xinput list` and it now shows up in `xsetwacom --list`.

I went over to the [Collecting tablet diagnostics page](https://digimend.github.io/support/howto/trbl/diagnostics/) to help me understand how my tablet works and to share the data with you here.

---
## Model
```
$ lsusb
Bus 003 Device 030: ID 256c:006d  Tablet Monitor
```
---
## USB descriptors
### descriptors
```
$ sudo lsusb -v -d 256c:006d
Bus 003 Device 030: ID 256c:006d  Tablet Monitor
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x256c 
  idProduct          0x006d 
  bcdDevice            0.00
  iManufacturer           5 (error)
  iProduct                6 (error)
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      18
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
[... 141 lines cut ...]
```
$ sudo usbhid-dump -es -m 256c:006d | tee pen_pressure.txt
Starting dumping interrupt transfer stream
with 1 minute timeout.

003:044:002:Failed to set infinite idle duration: Input/Output Error
003:044:002:Failed to attach to the kernel driver: Entity not found
003:044:001:Failed to attach to the kernel driver: Entity not found
003:044:000:Failed to release: No such device (it may have been disconnected)
003:044:000:Failed to attach to the kernel driver: No such device (it may have been disconnected)
```
```
$ sudo usbhid-dump -es -m 256c:006d | tee pen_pressure.txt
Starting dumping interrupt transfer stream
with 1 minute timeout.

003:060:001:Failed to set report protocol: Input/Output Error
003:060:002:Failed to attach to the kernel driver: Entity not found
003:060:001:Failed to release: No such device (it may have been disconnected)
003:060:001:Failed to attach to the kernel driver: No such device (it may have been disconnected)
003:060:000:Failed to release: No such device (it may have been disconnected)
003:060:000:Failed to attach to the kernel driver: No such device (it may have been disconnected)
```

I've learned a lot doing this diagnostic but I still can't understand most of this. Any help would be greatly appreciated :)

TheFlagCourier, 2020-05-23:
I have the same tablet, and get a similar experience on Fedora 32. I do not have the random-disconnect issue on mine. Thought I'd share the frame-buttons usbhid-dump output. These were taken without installing the kernel drivers, on Fedora's kernel-5.6.13-300.fc32.x86_64
[frame-buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4670877/frame-buttons.txt)

darkbeanbbq, 2020-05-27:
Hey @TheFlagCourier and thanks for your feedback. I wasn't sure those disconnections were normal, so I tested the tablet on a Windows partition and the experience there is much worse. I recorded a video of the problem and sent it to Huion support to find a fix or get a new cable or tablet.

I got the buttons to work with `xsetwacom --set 'Tablet Monitor Pad pad' Button 1 "key 1"` found on the  [linuxwacom/xf86-input-wacom wiki](https://github.com/linuxwacom/xf86-input-wacom/wiki/Tablet-Configuration-1:-xsetwacom-and-xorg.conf).

Apart from my disconnection problem everything seems functional on the Huion Kamvas 13, it's just that it needs to be scripted manually by yourself.

TheFlagCourier, 2020-06-09:
@darkbeanbbq thanks for the update - and I'll definitely look into that button setup.