DIGImend/digimend-kernel-drivers #399: Huion 1060 Pro not working on Kubuntu 20.04 on Kernel 5.6.13

wingtatyau, 2020-05-22:
Hello,
I have installed the digimend-kernel-drivers from source with DKMS. Seems the tablet is recognized since I can see it from `xsetwacom list`. However, I cannot use it to move the cursor and the buttons are not working. I checked #318 which says the tablet Huion 1060 Plus should work with kernel 5.6.8. I wonder if it should be the same for Huion 1060 Pro as I am already using kernel 5.6.13.

Could you please help to check if I missed anything? Thanks in advance.

**Information:**
I am running Kubuntu 20.04 on kernel 5.6.13
```
$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="20.04 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

$ uname -sr
Linux 5.6.13-050613-generic
```
The tablet can be found in `xsetwacom list`.
```
$ xsetwacom list
PenTablet  Pen stylus                   id: 10  type: STYLUS    
PenTablet  Pad pad                      id: 11  type: PAD       
PenTablet  Dial pad                     id: 12  type: PAD       
PenTablet  Touch Strip pad              id: 13  type: PAD    
```
usbhid-dump can detect the tablet.
```
$ sudo usbhid-dump -es -m 256c:006e
Starting dumping interrupt transfer stream
with 1 minute timeout.

001:002:001:STREAM             1590121322.543154
 0A C0 F4 73 C0 2B 00 00

001:002:001:STREAM             1590121324.437463
 0A C0 74 20 65 7C 00 00

001:002:001:STREAM             1590121324.441334
 0A C0 79 2F 6E 28 00 00

001:002:001:STREAM             1590121324.443347
 0A C0 79 2F 6E 28 00 00

001:002:001:STREAM             1590121324.445326
 0A C0 79 2F 4A 28 00 00

001:002:001:STREAM             1590121324.449473
 0A C0 79 2F 2C 28 00 00

001:002:001:STREAM             1590121324.451326
 0A C0 85 2F 10 28 00 00

001:002:001:STREAM             1590121324.453322
 0A C0 85 2F F7 27 00 00
```

Output of `xinput --list`
```
$ xinput --list
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ ITE8708 CIR transceiver                   id=18   [slave  pointer  (2)]
⎜   ↳ BluetoothMouse3600 Consumer Control       id=20   [slave  pointer  (2)]
⎜   ↳ BluetoothMouse3600 Mouse                  id=22   [slave  pointer  (2)]
⎜   ↳ PenTablet  Dial pad                       id=10   [slave  pointer  (2)]
⎜   ↳ PenTablet  Pad pad                        id=11   [slave  pointer  (2)]
⎜   ↳ PenTablet  Touch Strip pad                id=12   [slave  pointer  (2)]
⎜   ↳ PenTablet  Pen stylus                     id=13   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Video Bus                                 id=7    [slave  keyboard (3)]
    ↳ Power Button                              id=8    [slave  keyboard (3)]
    ↳ Sleep Button                              id=9    [slave  keyboard (3)]
    ↳ USB Keyboard                              id=14   [slave  keyboard (3)]
    ↳ USB Keyboard System Control               id=15   [slave  keyboard (3)]
    ↳ USB Keyboard Consumer Control             id=16   [slave  keyboard (3)]
    ↳ USB Keyboard                              id=17   [slave  keyboard (3)]
    ↳ ITE8708 CIR transceiver                   id=19   [slave  keyboard (3)]
    ↳ BluetoothMouse3600 Consumer Control       id=21   [slave  keyboard (3)]
```
uclogic-probe output and `/etc/X11/xorg.conf.d/50-tablet.conf`:

```
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 060b:7a0d Solid Year USB Keyboard
Bus 001 Device 004: ID 8087:0aaa Intel Corp. 
Bus 001 Device 002: ID 256c:006e  PenTablet 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ sudo uclogic-probe 1 2 | uclogic-decode 
    |                Product: PenTablet 
 64 |        Params block #1: ??????
    |                          Max X: 40000
    |                          Max Y: 25000
    |                   Max pressure: 2047
    |                     Resolution: 4000
    |
 79 |         Internal model: HA60-F400
 7b |         Buttons status: HK On
 c8 |        Params block #2: ??a?????
    |                          Max X: 40000
    |                          Max Y: 25000
    |                   Max pressure: 2047
    |                     Resolution: 4000
    |
 c9 |       Firmware version: HUION_T151_160608
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd

$ sudo cat /etc/X11/xorg.conf.d/50-tablet.conf 
Section "InputClass"
    Identifier "Tablet"
    Driver "wacom"
    MatchDevicePath "/dev/input/event*"
    MatchUSBID "256c:006e
EndSection
```
evtest output:

```
$ sudo evtest 
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Sleep Button
/dev/input/event1:      Power Button
/dev/input/event2:      Power Button
/dev/input/event3:      PenTablet  Mouse
/dev/input/event4:      PenTablet  Pen
/dev/input/event5:      PenTablet  Pad
/dev/input/event6:      HDA Intel PCH Mic
/dev/input/event7:      HDA Intel PCH Front Headphone
/dev/input/event8:      HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event9:      HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event10:     USB Keyboard
/dev/input/event11:     USB Keyboard System Control
/dev/input/event12:     USB Keyboard Consumer Control
/dev/input/event13:     USB Keyboard
/dev/input/event14:     USB Keyboard
/dev/input/event15:     Video Bus
/dev/input/event16:     ITE8708 CIR transceiver
/dev/input/event17:     PenTablet  Touch Strip
/dev/input/event18:     PenTablet  Dial
/dev/input/event19:     HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event20:     HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event21:     HDA Intel PCH HDMI/DP,pcm=10
/dev/input/event22:     BluetoothMouse3600 Mouse
/dev/input/event23:     BluetoothMouse3600 Consumer Control
Select the device event number [0-23]: 4
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
Input device name: "PenTablet  Pen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 320 (BTN_TOOL_PEN)
    Event code 330 (BTN_TOUCH)
    Event code 331 (BTN_STYLUS)
    Event code 332 (BTN_STYLUS2)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max    40000
      Resolution     157
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max    25000
      Resolution     157
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     2047
    Event code 26 (ABS_TILT_X)
      Value      0
      Min      -60
      Max       60
      Resolution      57
    Event code 27 (ABS_TILT_Y)
      Value      0
      Min      -60
      Max       60
      Resolution      57
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
```

spbnick, 2020-05-24:
@wingtatyau, do you get any output from `xinput test "PenTablet  Pen stylus"` when moving your pen? How about evtest output for the `PenTablet  Pen` device?

wingtatyau, 2020-05-25:
Hi @spbnick , `xinput test "PenTablet  Pen stylus"` does not show anything when I move the pen.
```
$ xinput test "PenTablet  Pen stylus"
^C
```

There is no evtest output for the `PenTablet  Pen` device.
```
$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Sleep Button
/dev/input/event1:      Power Button
/dev/input/event2:      Power Button
/dev/input/event3:      USB Keyboard
/dev/input/event4:      USB Keyboard System Control
/dev/input/event5:      USB Keyboard Consumer Control
/dev/input/event6:      HDA Intel PCH Mic
/dev/input/event7:      HDA Intel PCH Front Headphone
/dev/input/event8:      HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event9:      HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event10:     USB Keyboard
/dev/input/event11:     USB Keyboard
/dev/input/event12:     BluetoothMouse3600 Mouse
/dev/input/event13:     BluetoothMouse3600 Consumer Control
/dev/input/event14:     PenTablet  Pen
/dev/input/event15:     Video Bus
/dev/input/event16:     ITE8708 CIR transceiver
/dev/input/event17:     PenTablet  Pad
/dev/input/event18:     PenTablet  Touch Strip
/dev/input/event19:     HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event20:     HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event21:     HDA Intel PCH HDMI/DP,pcm=10
/dev/input/event22:     PenTablet  Dial
/dev/input/event23:     PenTablet  Mouse
Select the device event number [0-23]: 14
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
Input device name: "PenTablet  Pen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 320 (BTN_TOOL_PEN)
    Event code 330 (BTN_TOUCH)
    Event code 331 (BTN_STYLUS)
    Event code 332 (BTN_STYLUS2)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max    40000
      Resolution     157
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max    25000
      Resolution     157
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     2047
    Event code 26 (ABS_TILT_X)
      Value      0
      Min      -60
      Max       60
      Resolution      57
    Event code 27 (ABS_TILT_Y)
      Value      0
      Min      -60
      Max       60
      Resolution      57
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
```

spbnick, 2020-05-25:
Thank you. Could you please post the output of `cat /sys/modules/hid_uclogic/version`?

wingtatyau, 2020-05-25:
There is no such file.
```
$ cat /sys/modules/hid_uclogic/version
cat: /sys/modules/hid_uclogic/version: No such file or directory
```

spbnick, 2020-05-25:
OK, that means you're not actually using the downloaded driver.
Please check you don't have an issue described in https://github.com/DIGImend/digimend-kernel-drivers#dkms-issues-preventing-correct-installation

If not, please post your entire installation session here.

wingtatyau, 2020-05-26:
I checked that I don't have an issue described in https://github.com/DIGImend/digimend-kernel-drivers#dkms-issues-preventing-correct-installation

I am installing from source with DKMS.

```
$ git clone https://github.com/DIGImend/digimend-kernel-drivers.git
Cloning into 'digimend-kernel-drivers'...
remote: Enumerating objects: 87, done.
remote: Counting objects: 100% (87/87), done.
remote: Compressing objects: 100% (53/53), done.
remote: Total 1443 (delta 48), reused 57 (delta 34), pack-reused 1356
Receiving objects: 100% (1443/1443), 435.95 KiB | 249.00 KiB/s, done.
Resolving deltas: 100% (942/942), done.
```

I checked that both the linux-headers and dkms are already installed.

```
$ apt list --installed | grep linux-headers

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

linux-headers-5.4.0-29-generic/focal-updates,focal-security,now 5.4.0-29.33 amd64 [installed,automatic]
linux-headers-5.4.0-29/focal-updates,focal-updates,focal-security,focal-security,now 5.4.0-29.33 all [installed,automatic]
linux-headers-5.4.0-31-generic/focal-updates,focal-security,now 5.4.0-31.35 amd64 [installed]
linux-headers-5.4.0-31/focal-updates,focal-updates,focal-security,focal-security,now 5.4.0-31.35 all [installed]
linux-headers-5.6.13-050613-generic/now 5.6.13-050613.202005141310 amd64 [installed,local]
linux-headers-5.6.13-050613/now 5.6.13-050613.202005141310 all [installed,local]
linux-headers-generic/focal-updates,focal-security,now 5.4.0.31.36 amd64 [installed,automatic]
```
```
$ apt list --installed | grep dkms

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

dkms/focal,focal,now 2.8.1-5ubuntu1 all [installed,automatic]
virtualbox-dkms/focal,now 6.1.6-dfsg-1 amd64 [installed,automatic]
```

Then I did `sudo make dkms_install`.

```
$ cd digimend-kernel-drivers/
$ sudo make dkms_install
dkms add .

Creating symlink /var/lib/dkms/digimend/10/source ->
                 /usr/src/digimend-10

DKMS: add completed.
dkms build digimend/10

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.6.13-050613-generic KVERSION=5.6.13-050613-generic...
Signing module:
 - /var/lib/dkms/digimend/10/5.6.13-050613-generic/x86_64/module/hid-kye.ko
 - /var/lib/dkms/digimend/10/5.6.13-050613-generic/x86_64/module/hid-viewsonic.ko
 - /var/lib/dkms/digimend/10/5.6.13-050613-generic/x86_64/module/hid-uclogic.ko
 - /var/lib/dkms/digimend/10/5.6.13-050613-generic/x86_64/module/hid-polostar.ko
Secure Boot not enabled on this system.
cleaning build area...

DKMS: build completed.
dkms install digimend/10

hid-kye.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.6.13-050613-generic/updates/dkms/

hid-uclogic.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.6.13-050613-generic/updates/dkms/

hid-polostar.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.6.13-050613-generic/updates/dkms/

hid-viewsonic.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.6.13-050613-generic/updates/dkms/

depmod...

DKMS: install completed.
install -D -m 0644 depmod.conf /etc/depmod.d/digimend.conf
depmod -a
set -e -x; \
if test -e /usr/lib/dracut/dracut.conf.d; then \
    install -m 0644 dracut.conf /usr/lib/dracut/dracut.conf.d/90-digimend.conf; \
    dracut --force; \
fi
+ test -e /usr/lib/dracut/dracut.conf.d
install -D -m 0755 hid-rebind /lib/udev/hid-rebind
install -D -m 0644 udev.rules /lib/udev/rules.d/90-digimend.rules
udevadm control --reload
install -D -m 0644 xorg.conf /usr/share/X11/xorg.conf.d/50-digimend.conf
install -D -m 0755 digimend-debug /usr/sbin/digimend-debug
```

After that, I unloaded the previous versions of the drivers and reconnected the tablet.
```
$ sudo modprobe -r hid-kye hid-uclogic hid-polostar hid-viewsonic
```

I also created the file `/etc/X11/xorg.conf.d/50-tablet.conf` and rebooted my machine.
```
$ sudo cat /etc/X11/xorg.conf.d/50-tablet.conf 
Section "InputClass"
    Identifier "Tablet"
    Driver "wacom"
    MatchDevicePath "/dev/input/event*"
    MatchUSBID "256c:006e
EndSection
```

spbnick, 2020-05-26:
Looks good, although creating `50-tablet.conf` is not necessary, drivers ship with their own X.org configuration.

Could you please post the output of `/sbin/modinfo hid-uclogic`?

spbnick, 2020-05-26:
`No original module exists within this kernel` is somewhat suspicious. Is `5.6.13-050613-generic` the kernel you're running? What does your `uname -r` show?

wingtatyau, 2020-05-26:
Yes I am running `5.6.13-050613-generic`.
```
$ uname -r
5.6.13-050613-generic
```

Output of `/sbin/modinfo hid-uclogic`:
```
$ /sbin/modinfo hid-uclogic
filename:       /lib/modules/5.6.13-050613-generic/updates/dkms/hid-uclogic.ko
version:        10
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     957BF5A50EF2C284749D987
alias:          hid:b0003g*v000028BDp00000042
alias:          hid:b0003g*v000028BDp00000094
alias:          hid:b0003g*v000028BDp00000075
alias:          hid:b0003g*v000028BDp00000055
alias:          hid:b0003g*v000028BDp00000071
alias:          hid:b0003g*v000028BDp00000074
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000047
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006D
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid,usbhid
retpoline:      Y
name:           hid_uclogic
vermagic:       5.6.13-050613-generic SMP mod_unload 
sig_id:         PKCS#7
signer:         chris-pc Secure Boot Module Signature key
sig_key:        67:EB:0C:BC:74:03:16:12:E8:05:AE:E0:D3:2D:F7:B8:F4:67:DF:9B
sig_hashalgo:   sha512
signature:      35:C3:BA:BB:73:46:58:D4:C5:43:E2:D3:78:48:7F:25:09:23:BD:B3:
                AC:E5:CC:CE:47:54:A3:C2:B7:21:9D:17:8C:AB:17:B2:22:4F:B5:12:
                B9:7B:E5:26:BC:EE:22:BC:A3:55:84:4F:19:96:5C:23:D5:4B:9B:67:
                3A:B4:28:F2:1A:27:F4:B2:89:2D:74:51:5E:FE:11:C4:B4:41:A7:E0:
                5F:27:1F:E2:D0:62:5E:BC:4A:08:4B:6E:70:10:9F:04:32:4D:77:C9:
                A2:C9:77:11:1E:02:E1:EA:67:63:9B:77:B8:89:3A:EA:29:42:35:E2:
                F2:BA:C7:75:44:63:C9:7A:35:85:75:C2:84:58:4D:9C:29:E7:2F:69:
                FC:0C:D8:6C:54:BF:AF:2F:96:9C:C0:90:E9:A2:F0:20:ED:97:51:4F:
                63:15:22:23:25:84:FA:6C:06:E6:3F:19:43:E6:27:F8:4F:A1:1C:19:
                5F:14:FB:4D:CC:69:DF:8D:58:4F:31:45:C7:BF:5C:A7:33:9F:33:49:
                49:B3:D4:FD:0E:F0:E0:68:4B:E2:CC:D9:97:21:F0:C9:80:2F:C0:0B:
                E8:47:92:73:46:12:EE:00:F2:1B:2C:F6:82:DC:39:13:B8:0E:70:75:
                6B:49:07:36:55:76:E9:12:01:6E:A0:0A:C8:B7:4D:F0
```

spbnick, 2020-05-26:
Alright, looks good. Could you do `sudo modprobe -r hid-uclogic && sudo modprobe hid-uclogic && cat /sys/module/hid_uclogic/version`?

wingtatyau, 2020-05-26:
```
$ sudo modprobe -r hid-uclogic && sudo modprobe hid-uclogic && cat /sys/module/hid_uclogic/version
10
```

spbnick, 2020-05-26:
Alright, now you've got the right module loaded, check if your tablet works. After that we can figure out why it wasn't loaded automatically.

wingtatyau, 2020-05-26:
Umm my tablet still does not work. I tried `xinput test "PenTablet  Pen stylus"` again and there is no output.
```
$ xinput test "PenTablet  Pen stylus"
^C
```

spbnick, 2020-05-26:
Assuming `evtest` doesn't work either, please download [`digimend-debug`](https://raw.githubusercontent.com/DIGImend/digimend-kernel-drivers/master/digimend-debug), execute `sudo ./digimend-debug on`, reconnect your tablet, and post the relevant piece of your `dmesg` output.

wingtatyau, 2020-05-26:
The `evtest` doesn't work.
```
$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Sleep Button
/dev/input/event1:      Power Button
/dev/input/event2:      Power Button
/dev/input/event3:      Video Bus
/dev/input/event4:      USB Keyboard
/dev/input/event5:      USB Keyboard System Control
/dev/input/event6:      USB Keyboard Consumer Control
/dev/input/event7:      USB Keyboard
/dev/input/event8:      USB Keyboard
/dev/input/event9:      BluetoothMouse3600 Mouse
/dev/input/event10:     BluetoothMouse3600 Consumer Control
/dev/input/event11:     HDA Intel PCH Mic
/dev/input/event12:     HDA Intel PCH Front Headphone
/dev/input/event13:     PenTablet  Pen
/dev/input/event14:     PenTablet  Pad
/dev/input/event15:     PenTablet  Touch Strip
/dev/input/event16:     ITE8708 CIR transceiver
/dev/input/event17:     PenTablet  Dial
/dev/input/event18:     PenTablet  Mouse
/dev/input/event19:     HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event20:     HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event21:     HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event22:     HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event23:     HDA Intel PCH HDMI/DP,pcm=10
Select the device event number [0-23]: 13
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
Input device name: "PenTablet  Pen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 320 (BTN_TOOL_PEN)
    Event code 330 (BTN_TOUCH)
    Event code 331 (BTN_STYLUS)
    Event code 332 (BTN_STYLUS2)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max    40000
      Resolution     157
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max    25000
      Resolution     157
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     2047
    Event code 26 (ABS_TILT_X)
      Value      0
      Min      -60
      Max       60
      Resolution      57
    Event code 27 (ABS_TILT_Y)
      Value      0
      Min      -60
      Max       60
      Resolution      57
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
[... 229 lines cut ...]
[11360.356036] hid_uclogic:uclogic_params_frame_hid_dbg:92: uclogic 0003:256C:006E.001A:                .desc_size = 0
[11360.356037] hid_uclogic:uclogic_params_frame_hid_dbg:93: uclogic 0003:256C:006E.001A:                .id = 0
[11360.356038] hid_uclogic:uclogic_params_frame_hid_dbg:94: uclogic 0003:256C:006E.001A:                .suffix = (null)
[11360.356038] hid_uclogic:uclogic_params_frame_hid_dbg:95: uclogic 0003:256C:006E.001A:                .re_lsb = 0
[11360.356039] hid_uclogic:uclogic_params_frame_hid_dbg:96: uclogic 0003:256C:006E.001A:                .dev_id_byte = 0
[11360.356040] hid_uclogic:uclogic_params_frame_hid_dbg:97: uclogic 0003:256C:006E.001A:                .touch_byte = 0
[11360.356041] hid_uclogic:uclogic_params_frame_hid_dbg:98: uclogic 0003:256C:006E.001A:                .touch_max = 0
[11360.356041] hid_uclogic:uclogic_params_frame_hid_dbg:99: uclogic 0003:256C:006E.001A:                .touch_flip_at = 0
[11360.356042] hid_uclogic:uclogic_params_frame_hid_dbg:101: uclogic 0003:256C:006E.001A:               .bitmap_dial_byte = 0
[11360.356043] hid_uclogic:uclogic_params_hid_dbg:127: uclogic 0003:256C:006E.001A:     },
[11360.356044] hid_uclogic:uclogic_params_hid_dbg:125: uclogic 0003:256C:006E.001A:     {
[11360.356044] hid_uclogic:uclogic_params_frame_hid_dbg:91: uclogic 0003:256C:006E.001A:                .desc_ptr = 000000004465ea41
[11360.356045] hid_uclogic:uclogic_params_frame_hid_dbg:92: uclogic 0003:256C:006E.001A:                .desc_size = 0
[11360.356046] hid_uclogic:uclogic_params_frame_hid_dbg:93: uclogic 0003:256C:006E.001A:                .id = 0
[11360.356047] hid_uclogic:uclogic_params_frame_hid_dbg:94: uclogic 0003:256C:006E.001A:                .suffix = (null)
[11360.356048] hid_uclogic:uclogic_params_frame_hid_dbg:95: uclogic 0003:256C:006E.001A:                .re_lsb = 0
[11360.356048] hid_uclogic:uclogic_params_frame_hid_dbg:96: uclogic 0003:256C:006E.001A:                .dev_id_byte = 0
[11360.356049] hid_uclogic:uclogic_params_frame_hid_dbg:97: uclogic 0003:256C:006E.001A:                .touch_byte = 0
[11360.356050] hid_uclogic:uclogic_params_frame_hid_dbg:98: uclogic 0003:256C:006E.001A:                .touch_max = 0
[11360.356051] hid_uclogic:uclogic_params_frame_hid_dbg:99: uclogic 0003:256C:006E.001A:                .touch_flip_at = 0
[11360.356051] hid_uclogic:uclogic_params_frame_hid_dbg:101: uclogic 0003:256C:006E.001A:               .bitmap_dial_byte = 0
[11360.356052] hid_uclogic:uclogic_params_hid_dbg:127: uclogic 0003:256C:006E.001A:     }
[11360.356053] hid_uclogic:uclogic_params_hid_dbg:130: uclogic 0003:256C:006E.001A: }
[11360.356054] uclogic 0003:256C:006E.001A: interface is invalid, ignoring
```

spbnick, 2020-05-26:
Thank you, everything looks fine here as well. Could you please see if you get any output from `/sys/kernel/debug/hid/0003:256C:006E.*/events` files when running `cat` on them as root, while running `sudo evtest` on any of the tablet devices and doing pen strokes?

wingtatyau, 2020-05-26:
I there is no output from `/sys/kernel/debug/hid/0003:256C:006E.*/events` when I am running `sudo evtest` on the tablet devices and doing pen strokes. I tried all the devices starting with "PenTablet".

Terminal 1 as root:
```
# cat /sys/kernel/debug/hid/0003:256C:006E.*/events

```

Terminal 2 as my own user:
```
$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Sleep Button
/dev/input/event1:      Power Button
/dev/input/event2:      Power Button
/dev/input/event3:      Video Bus
/dev/input/event4:      USB Keyboard
/dev/input/event5:      USB Keyboard System Control
/dev/input/event6:      USB Keyboard Consumer Control
/dev/input/event7:      USB Keyboard
/dev/input/event8:      USB Keyboard
/dev/input/event9:      BluetoothMouse3600 Mouse
/dev/input/event10:     BluetoothMouse3600 Consumer Control
/dev/input/event11:     HDA Intel PCH Mic
/dev/input/event12:     HDA Intel PCH Front Headphone
/dev/input/event13:     PenTablet  Pen
/dev/input/event14:     PenTablet  Pad
/dev/input/event15:     PenTablet  Touch Strip
/dev/input/event16:     ITE8708 CIR transceiver
/dev/input/event17:     PenTablet  Dial
/dev/input/event18:     PenTablet  Mouse
/dev/input/event19:     HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event20:     HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event21:     HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event22:     HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event23:     HDA Intel PCH HDMI/DP,pcm=10
Select the device event number [0-23]: 13
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
Input device name: "PenTablet  Pen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 320 (BTN_TOOL_PEN)
    Event code 330 (BTN_TOUCH)
    Event code 331 (BTN_STYLUS)
    Event code 332 (BTN_STYLUS2)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max    40000
      Resolution     157
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max    25000
      Resolution     157
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     2047
    Event code 26 (ABS_TILT_X)
      Value      0
      Min      -60
      Max       60
      Resolution      57
    Event code 27 (ABS_TILT_Y)
      Value      0
[... 209 lines cut ...]
/dev/input/event23:     HDA Intel PCH HDMI/DP,pcm=10
Select the device event number [0-23]: 18
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
Input device name: "PenTablet  Mouse"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
    Event code 275 (BTN_SIDE)
    Event code 276 (BTN_EXTRA)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
    Event code 6 (REL_HWHEEL)
    Event code 8 (REL_WHEEL)
    Event code 11 (REL_WHEEL_HI_RES)
    Event code 12 (REL_HWHEEL_HI_RES)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
```

spbnick, 2020-05-26:
> I there is no output from `/sys/kernel/debug/hid/0003:256C:006E.*/events` when I am running `sudo evtest` on the tablet devices and doing pen strokes. I tried all the devices starting with "PenTablet".
> 
> Terminal 1 as root:
> 
> ```
> # cat /sys/kernel/debug/hid/0003:256C:006E.*/events
> ```

Is there more than one line in the output of `ls -l /sys/kernel/debug/hid/0003:256C:006E.*/events`? If there is, then the command above would only try the first file. If there is, please try every of those files separately.

wingtatyau, 2020-05-27:
There are 3 directories for `256C:006E` under `/sys/kernel/debug/hid/`.
```
# ls -l /sys/kernel/debug/hid/0003\:256C\:006E*/events
-r-------- 1 root root 0 May 27 23:20 /sys/kernel/debug/hid/0003:256C:006E.0030/events
-r-------- 1 root root 0 May 27 23:20 /sys/kernel/debug/hid/0003:256C:006E.0031/events
-r-------- 1 root root 0 May 27 23:20 /sys/kernel/debug/hid/0003:256C:006E.0032/events
```

There is output when I cat `/sys/kernel/debug/hid/0003:256C:006E.0031/events` while running `evtest` for `PenTablet  Mouse`

Terminal 1 as root:
```
# cat /sys/kernel/debug/hid/0003:256C:006E.0031/events

report (size 8) (numbered) =  0a c0 04 5a e7 0b 00 00
Digitizers.TipSwitch = 0
Digitizers.BarrelSwitch = 0
Digitizers.Eraser = 0
Digitizers.Invert = 0
Digitizers.SecondaryTipSwitch = 0
Digitizers.BarrelSwitch = 0
Digitizers.InRange = 1
GenericDesktop.X = 23044
GenericDesktop.Y = 3047
Digitizers.TipPressure = 0

report (size 8) (numbered) =  0a c0 9a 6b ec 19 00 00
Digitizers.TipSwitch = 0
Digitizers.BarrelSwitch = 0
Digitizers.Eraser = 0
Digitizers.Invert = 0
Digitizers.SecondaryTipSwitch = 0
Digitizers.BarrelSwitch = 0
Digitizers.InRange = 1
GenericDesktop.X = 27546
GenericDesktop.Y = 6636
Digitizers.TipPressure = 0

report (size 8) (numbered) =  0a c0 a6 6b de 19 00 00
Digitizers.TipSwitch = 0
Digitizers.BarrelSwitch = 0
Digitizers.Eraser = 0
Digitizers.Invert = 0
Digitizers.SecondaryTipSwitch = 0
Digitizers.BarrelSwitch = 0
Digitizers.InRange = 1
GenericDesktop.X = 27558
GenericDesktop.Y = 6622
Digitizers.TipPressure = 0

report (size 8) (numbered) =  0a c0 b3 6b d5 19 00 00
Digitizers.TipSwitch = 0
Digitizers.BarrelSwitch = 0
Digitizers.Eraser = 0
Digitizers.Invert = 0
Digitizers.SecondaryTipSwitch = 0
Digitizers.BarrelSwitch = 0
Digitizers.InRange = 1
GenericDesktop.X = 27571
GenericDesktop.Y = 6613
Digitizers.TipPressure = 0

...
```

Terminal 2 as my own user:
```
$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Sleep Button
/dev/input/event1:      Power Button
/dev/input/event2:      Power Button
/dev/input/event3:      Video Bus
/dev/input/event4:      BluetoothMouse3600 Mouse
/dev/input/event5:      BluetoothMouse3600 Consumer Control
/dev/input/event6:      USB Keyboard
/dev/input/event7:      USB Keyboard System Control
/dev/input/event8:      USB Keyboard Consumer Control
/dev/input/event9:      USB Keyboard
/dev/input/event10:     USB Keyboard
/dev/input/event11:     HDA Intel PCH Mic
/dev/input/event12:     HDA Intel PCH Front Headphone
/dev/input/event13:     PenTablet  Pen
/dev/input/event14:     PenTablet  Pad
/dev/input/event15:     PenTablet  Touch Strip
/dev/input/event16:     ITE8708 CIR transceiver
/dev/input/event17:     PenTablet  Dial
/dev/input/event18:     PenTablet  Mouse
/dev/input/event19:     HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event20:     HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event21:     HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event22:     HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event23:     HDA Intel PCH HDMI/DP,pcm=10
Select the device event number [0-23]: 18
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
Input device name: "PenTablet  Mouse"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
    Event code 275 (BTN_SIDE)
    Event code 276 (BTN_EXTRA)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
    Event code 6 (REL_HWHEEL)
    Event code 8 (REL_WHEEL)
    Event code 11 (REL_WHEEL_HI_RES)
    Event code 12 (REL_HWHEEL_HI_RES)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
```