DIGImend/digimend-kernel-drivers #401: Not able to install digimend-dkms_9_all.deb on ubuntu 20.04

kunalvshah, 2020-05-24:
Hi All,

I saw few previous post on the same subject  and it seems my error is different. 

```
desktop:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04 LTS
Release:	20.04
Codename:	focal

Linux desktop 5.4.0-31-generic #35-Ubuntu SMP Thu May 7 20:20:34 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

desktop:~/Downloads$ sudo dpkg -i digimend-dkms_9_all.deb 
[sudo] password for xxxxxxx: 
Selecting previously unselected package digimend-dkms.
(Reading database ... 250206 files and directories currently installed.)
Preparing to unpack digimend-dkms_9_all.deb ...
Unpacking digimend-dkms (9) ...
Setting up digimend-dkms (9) ...
Loading new digimend-9 DKMS files...
Building for 5.4.0-31-generic
Building initial module for 5.4.0-31-generic
ERROR (dkms apport): unable to determine source package for digimend-dkms
Error! Bad return status for module build on kernel: 5.4.0-31-generic (x86_64)
Consult /var/lib/dkms/digimend/9/build/make.log for more information.
dpkg: error processing package digimend-dkms (--install):
 installed digimend-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 digimend-dkms

```

and 
/var/lib/dkms/digimend/9/build/make.log shows

```
DKMS make.log for digimend-9 for kernel 5.4.0-31-generic (x86_64)
Sunday 24 May 2020 04:46:58 PM IST
make -C /lib/modules/5.4.0-31-generic/build SUBDIRS=/var/lib/dkms/digimend/9/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-31-generic'
  DESCEND  objtool
/usr/src/linux-headers-5.4.0-31-generic/tools/build/Makefile.build:37: /usr/src/linux-headers-5.4.0-31-generic/tools/build/Build.include: No such file or directory
make[5]: *** No rule to make target '/usr/src/linux-headers-5.4.0-31-generic/tools/build/Build.include'.  Stop.
make[4]: *** [Makefile:43: /usr/src/linux-headers-5.4.0-31-generic/tools/objtool/fixdep-in.o] Error 2
make[3]: *** [/usr/src/linux-headers-5.4.0-31-generic/tools/build/Makefile.include:5: fixdep] Error 2
make[2]: *** [Makefile:67: objtool] Error 2
make[1]: *** [Makefile:1823: tools/objtool] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-31-generic'
make: *** [Makefile:22: modules] Error 2

```

I have installed all required modules and headers and still getting this error. Not sure what more should I do to fix this.

DaviRamosAndrade, 2020-05-24:
Work's on kernel version 5.3.0.53

kunalvshah, 2020-05-25:
If I don't use deb and install kernel module directly, it works fine. So the way I solved was 
1) download the soruce code from github 
'''
git clone https://github.com/DIGImend/digimend-kernel-drivers.git

sudo apt-get install -y "linux-headers-$(uname -r)"

sudo apt-get install -y dkms

sudo make dkms_install

#below command is just to ensure I don't have modules already loaded
sudo modprobe -r hid-kye hid-uclogic hid-polostar hid-viewsonic
'''
and it works !!!

But .deb definitely has some issues.

closed 2020-05-25 by kunalvshah