DIGImend/digimend-kernel-drivers #404: install Digimend driver in openSuse

Giorgio-IT, 2020-05-26:
Installation of v 9 driver in openSuse Leap 15.1 using dkms doesn't works, as described in https://github.com/DIGImend/digimend-kernel-drivers/issues/392 . 
The only fix I'm able to apply doesn't solve the problem.
I ask for reopening that issue but it seems impossible then I opened a new one.

spbnick, 2020-05-26:
@Giorgio-IT, please post your complete installation session.

Giorgio-IT, 2020-05-26:
Removed and reinstalled v 9 driver:

`
sudo make dkms_install
[sudo] password di root: 
dkms add .

Creating symlink /var/lib/dkms/digimend/9/source ->
                 /usr/src/digimend-9

DKMS: add completed.
dkms build digimend/9

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.12.14-lp151.28.48-default KVERSION=4.12.14-lp151.28.48-default...
cleaning build area...

DKMS: build completed.
dkms install digimend/9

hid-kye.ko:
Running module version sanity check.
 - Original module
   - Found /lib/modules/4.12.14-lp151.28.48-default/extra/hid-kye.ko
   - Storing in /var/lib/dkms/digimend/original_module/4.12.14-lp151.28.48-default/x86_64/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.12.14-lp151.28.48-default/updates/

hid-uclogic.ko:
Running module version sanity check.
 - Original module
   - Found /lib/modules/4.12.14-lp151.28.48-default/extra/hid-uclogic.ko
   - Storing in /var/lib/dkms/digimend/original_module/4.12.14-lp151.28.48-default/x86_64/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/4.12.14-lp151.28.48-default/updates/

hid-polostar.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.12.14-lp151.28.48-default/updates/

hid-viewsonic.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.12.14-lp151.28.48-default/updates/

depmod...

DKMS: install completed.
install -D -m 0755 hid-rebind /lib/udev/hid-rebind
install -D -m 0644 udev.rules /lib/udev/rules.d/90-digimend.rules
install -D -m 0644 xorg.conf /usr/share/X11/xorg.conf.d/50-digimend.conf
udevadm control --reload
`

Giorgio-IT, 2020-07-11:
Is there any hope to get an answer? 
Thanks.