DIGImend/digimend-kernel-drivers #405: "HUION new 1600 Plus" fails kernel 5.3 ubuntu 18.04 LTS but works older kernel 4.15.0-101

walterwatson, 2020-05-26:
Hello , new tablet here fails on 5.3 series kernel on 64bit ubuntu 18.04.

I can get it to work on 4.15 tho, but only with: hid_generic

Is there any way to get this to work on 5.X series linux kernels?

Using 4.15 isn't an option for me longterm as the 5.x series has better hardware support for other devices I have.

What I tried for 5.3 kernel:
- official release deb
- compile official release source
- compile latest branch
- Plug the tablet into a USB 2.0, 3.0 and 3.1 ports on my computer - since I noticed in some bug reports that might be an issue as well.

All of the above failed.

If I downgrade to 4.15 the USB 2.0 /3.0 is a non-issue.  It works in both of them fine.

In kernel 4.15 all buttons on this device are registered as buttons 1-16 in 'xev'.

Is there any way to get this to work on 5.X series linux kernels?

Thank you for your consideration.

walterwatson, 2020-05-26:
This is under 4.15 as well, can't seem to rotate this pad or pen
`
xsetwacom list
HID 256c:006e Pen stylus        	id: 9	type: STYLUS    
HID 256c:006e Pad pad           	id: 10	type: PAD       
$ xsetwacom set "Pen" rotate half
Cannot find device 'Pen'.
$ xsetwacom set "Pad" rotate half
Cannot find device 'Pad'.
`

bsamorodov, 2020-06-02:
There are two bugs opened (with a solution) at kernel.org:
- https://bugzilla.kernel.org/show_bug.cgi?id=206897
- https://bugzilla.kernel.org/show_bug.cgi?id=207065

For those who don't want to patch kernel there is a woraround though (cherry-picked from neighbour issues). One should force a USB 3.0 port to work in USB 2.0 mode in Linux: https://www.systutorials.com/how-to-force-a-usb-3-0-port-to-work-in-usb-2-0-mode-in-linux/ .

1. Detect USB controllers.
`% lspci -nn | grep USB`
`00:14.0 USB controller [0c03]: Intel Corporation Wildcat Point-LP USB xHCI Controller [8086:9cb1] (rev 03)`
`00:1d.0 USB controller [0c03]: Intel Corporation Wildcat Point-LP USB EHCI Controller [8086:9ca6] (rev 03)`

2. Set USB-2.0 mode (use vendor and device ID pair in square brakets below). For me the first pair did the trick:
`% sudo setpci -H1 -d 8086:9cb1 d0.l=0`
`% xsetwacom --list devices`
`HID 256c:006e Pen stylus                id: 9   type: STYLUS`
`HID 256c:006e Pad pad                   id: 10  type: PAD`

If one should revert, set the last value to 1:
`% sudo setpci -H1 -d 8086:9cb1 d0.l=1`

pisimpson, 2020-06-16:
If I use your xsetwacom --list devices
i get
exactly the same
HID 256c:006e Pen stylus id: 9 type :STYLUS
HID 256c:006e Pad pad id:7 type : PAD

But I still dont get any output from either when I xinput --test 9

So what am I doing wrong?