DIGImend/digimend-kernel-drivers #407: how to handle pad and pen which is on same interface

DONofDESTROY, 2020-05-28:
hey everyone!!! 
I managed to make my tablet form generic keyboard emulation to generic buttons which closes this issue #403
but I face one major problem here 
My tablet is parblo A640 and when initializing the tablet it changes form separate interface for pen and pad to a single interface

**Before initializing**

- PEN(aka stylus)
```
002:006:000:STREAM             1589459888.689554
09 00 21 0F 8E 0D 00 00

```

- PAD
```
002:006:002:STREAM             1589460338.833529
03 03 2D 00 00 00 00 00
```
**after initializing**

- PEN
```
001:007:000:STREAM             1590512843.627644
07 80 AA 33 BE 18 00 00 00 00
```

- PAD
```
001:008:000:STREAM              1590513020.311517
07 E0 02 01 04 00 00 00 00 00
```
**did you notice something ?** 

both uses same interface **0**
and I have written report descriptor for both of the strings ie. for both PEN and PAD
and have tested both PEN and PAD separately both are working fine and  all the buttons are remap able  in both events

_so why this issue_

**did you notice  another thing?**

both  uses same report id's too(whyyyyyyy.....)
so I can use one at a time ie.
either PEN at interface **0** or PAD at  that interface 
how can I overcome this and make my tablet pad work with pen at same time

spbnick, 2020-05-28:
The way hid-uclogic deals with that is by rewriting the report ID for button reports.

DONofDESTROY, 2020-05-28:
that means should I enter some number instead of 0 in the frame_init_with_desc() 
function and in alter the report id in the report descriptor

DONofDESTROY, 2020-06-25:
still didn't get any idea about that any articles about how to change the interface will be appreciated