DIGImend/digimend-kernel-drivers #408: GAOMON G620 descriptors

fgmaggi, 2020-06-01:
here a link to amazon where i bought the item

https://www.amazon.it/GAOMON-Pollici-Tavoletta-Batteria-Disegnare/dp/B07R49KYCF/ref=pd_lpo_147_t_0/258-1290032-1114723?_encoding=UTF8&pd_rd_i=B07R49KYCF&pd_rd_r=c5c964ed-6522-4a5c-85b9-69a34adcb47d&pd_rd_w=O3E13&pd_rd_wg=LTAQF&pf_rd_p=3445fbfe-ec7c-415a-923d-008a660451c3&pf_rd_r=R7Z1YZ4YNSTRRV9KWF52&psc=1&refRID=R7Z1YZ4YNSTRRV9KWF52

here below data I grabbed via usbhid-dump

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4711193/descriptors.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4711194/frame_buttons.txt)
[frame_wheel_srolling.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4711195/frame_wheel_srolling.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4711196/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4711197/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4711198/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4711199/pen_pressure.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4711200/pen_tilt.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4711201/probe.txt)

I hope some of you will write the driver for this tablet

DavidHusicka, 2020-06-17:
I have this tablet and drivers in this repo take care of it. The drivers work perfectly for me. Do you have any issues with this tablet.

clay53, 2020-07-06:
> I have this tablet and drivers in this repo take care of it. The drivers work perfectly for me. Do you have any issues with this tablet.

I'm unable to use this driver so I can't test it for myself but the confusion probably comes from it not being in the supported device list.
(also title should be changed to say S620)

DavidHusicka, 2020-07-06:
> Edit: I can't use it because of bug in Kernel 5.4

I have Kernel 5.4.0-7634-generic and everything works correctly. The driver for this tablet (Gaomon S620) is already in the kernel for me.

DavidHusicka, 2020-07-06:
@clay53 what's your issue?

clay53, 2020-07-06:
@DavidHusicka #387

DavidHusicka, 2020-07-06:
@clay53 the digimend drivers should be part of the kernel. Therefore you don't need to install anything. 

Both pen pressure sensitivity and area changing works for me without installing third party software. Please follow the README file if you want to change anything regarding the tablet.

If you want to change your tablet area, it is necessary to use xorg server (instead of Wayland).

clay53, 2020-07-06:
@DavidHusicka hm? Are digimend drivers pre-installed? How do I check this because my tablet definitely does not work or is reports that it is using it. I was having trouble installing the drivers but are you saying I don't need to?

DavidHusicka, 2020-07-06:
@clay53 it should be preinstalled. First check whether your tablet has working sensitivity (try to draw in Krita or similar tool. Krita works the best for me). If it's working, proceed to configure your xorg to recognize your tablet. After xorg is aware of your tablet, you are able to change your area with xsetwacom utility (described in the README). I'm joined your discord so I can help you there.

spbnick, 2020-08-01:
Thank you for your contributions, everyone! Closing as the tablet seems to be supported.

closed 2020-08-01 by spbnick