DIGImend/digimend-kernel-drivers #409: New Huion 1060 Plus issue

sekoenig-de, 2020-06-01:
Hi,

I have an issue with my Huion tablet (model NEW 1060 PLUS), which may be the same or at least related to other issues reported here.  When I try to use it with my current system (Thinkpad P53, Debian Bullseye (testing) with kernel 5.6.14) the tablet is only recognized as a storage device (8 GB). When I run lsusb, I see one entry without description:

Bus 007 Device 012: ID 256c:006e

I believe this is the tablet.  I have the latest digimend drivers installed directly from the repository here via the dkms_install make target, which runs just fine.

Half a year ago or so the same tablet worked fine with my Thinkpad T420 and whatever kernel Debian testing was using at the time.

My BIOS does not offer to disable USB3.0, which I have read in some of the comments here, so unfortunately I cannot try that.  Is there anything else I can do to help debug the problem?

Best regards,

Sebastian

bsamorodov, 2020-06-02:
Untill the kernel is fixed you may be interested at the [workaround](https://github.com/DIGImend/digimend-kernel-drivers/issues/405#issuecomment-637844659).

HTH, bsam

sekoenig-de, 2020-06-03:
Thanks, bsam!  I found that the patch mentioned in the workaround made it into the kernel mainline with the recent 5.7 release, and I am happy to confirm that after compiling and booting that kernel the problem is indeed resolved now.

closed 2020-06-03 by sekoenig-de