DIGImend/digimend-kernel-drivers #411: Need instruction to purge the digimend dkms files after failed build

binarydepth, 2020-06-04:
Every time I install a package DKMS tried to build the drivers and fails.

```
------------------------------
Setting up digimend-dkms (9) ...
Removing old digimend-9 DKMS files...

------------------------------
Deleting module version: 9
completely from the DKMS tree.
------------------------------
Done.
Loading new digimend-9 DKMS files...
Building for 5.4.0-33-generic
Building initial module for 5.4.0-33-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/digimend-dkms.0.crash'
Error! Bad return status for module build on kernel: 5.4.0-33-generic (x86_64)
Consult /var/lib/dkms/digimend/9/build/make.log for more information.
dpkg: error processing package digimend-dkms (--configure):
 installed digimend-dkms package post-installation script subprocess returned error exit status 10
Setting up cpufrequtils (008-1.1) ...
update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults
update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults
Processing triggers for systemd (245.4-4ubuntu3.1) ...
Processing triggers for man-db (2.9.1-1) ...
Processing triggers for libc-bin (2.31-0ubuntu9) ...
Errors were encountered while processing:
 digimend-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
------------------------------
```

binarydepth, 2020-06-04:
sudo dkms uninstall -m digimend9 -v 9
Error! There are no instances of module: digimend9
located in the DKMS tree.

binarydepth, 2020-06-04:
sudo dkms uninstall -m digimend -v 9
Error! The module digimend 9 is not currently installed.
This module is not currently ACTIVE for kernel 5.4.0-33-generic (x86_64).

maz3max, 2020-06-04:
I did this: `sudo rm -rf /var/lib/dkms/digimend-kernel-drivers/`.
Seemed to work.

spbnick, 2020-06-06:
Would be a good question to ask at https://github.com/dell/dkms/issues or on your distro's support forums.

binarydepth, 2020-06-07:
> Would be a good question to ask at https://github.com/dell/dkms/issues or on your distro's support forums.

You are right!

binarydepth, 2020-06-11:
For the ones in Ubuntu I got the answer on AskUbuntu, do this: `dpkg --remove --force-remove-reinstreq digimend-dkms`