DIGImend/digimend-kernel-drivers #414: Can't load the driver automaticly

ltfan, 2020-06-09:
I install the driver successfully.
When i run the commond :
~~~~~~~~~~~~~
lsusb
Bus 001 Device 007: ID 5543:0041 UC-Logic Technology Corp. Genius PenSketch 6x8 Tablet
~~~~~~~~~~~~~
I  can't draw any line in krita .Only I can move the point with my tablet pen. 
I must to run the commond :
~~~~~~~~~~~~~~~~~~
sudo uclogic-probe 1 7
~~~~~~~~~~~~~~~~~~
Then I can draw lines with my tabliet pen.
How shoud I do ?

spbnick, 2020-06-09:
@ltfan, your tablet is not supported. The fact that it starts working after `uclogic-probe` is a happy coincidence, I would say. If you want it supported, please post the diagnostics as described in https://digimend.github.io/support/howto/trbl/diagnostics/

ltfan, 2020-06-10:
@spbnick 
My tablet is ugee rainbow,it is a very old product of UGEE,about fifteen years ago.
I don't think to be worth developing a new driver for the old tablet .
Anyway, thanks a lot.
At least , I can write note by hand with my ugee rainbow.

closed 2020-06-11 by ltfan