DIGImend/digimend-kernel-drivers #416: Huion 420 reports BTN_STYLUS2 events instead of BTN_TOOL_RUBBER on tip down

Xyene, 2020-06-11:
Hi,

I'm running the drivers in kernel 5.4.0 with a Huion 420 tablet whose pen has two buttons, one of which is a mode switch to eraser on tip down. However, it seems this device (and possibly also the H420, which AFAIK is the same pad but with 3 buttons on it?) has buggy firmware: it performs the virtual proximity out / in switch to eraser mode, but then sends `BTN_STYLUS2` events instead of `BTN_TOOL_RUBBER`.

[Here](https://gitlab.freedesktop.org/libinput/libinput/-/issues/494) is a thread from libinput describing this which has a bit more background and event traces. The relevant bits are on button press while tip is down:

```
  - evdev:
    - [  1, 924005,   4,   4,      30] # EV_MSC / MSC_SCAN                 30 (obfuscated)
    - [  1, 924005,   1, 330,       0] # EV_KEY / BTN_TOUCH                 0
    - [  1, 924005,   3,  24,       0] # EV_ABS / ABS_PRESSURE              0 (-2047)
    - [  1, 924005,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +56ms
  - evdev:
    - [  1, 932004,   4,   4,      30] # EV_MSC / MSC_SCAN                 30 (obfuscated)
    - [  1, 932004,   1, 330,       1] # EV_KEY / BTN_TOUCH                 1
    - [  1, 932004,   4,   4,      30] # EV_MSC / MSC_SCAN                 30 (obfuscated)
    - [  1, 932004,   1, 332,       1] # EV_KEY / BTN_STYLUS2               1
    - [  1, 932004,   3,  24,     647] # EV_ABS / ABS_PRESSURE            647 (+647)
    - [  1, 932004,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +8ms
```

which I think look fine except for `BTN_STYLUS2` instead of `BTN_TOOL_RUBBER`.

I'm happy to provide any other info that might be useful here.

spbnick, 2020-06-12:
@Xyene, I assume that switching to eraser tool is how the windows drivers treat the top pen button of your tablet.

All tablet drivers I wrote, always report it as BTN_STYLUS2, which is translated to the right mouse button by userspace drivers, because that's how most tablets intend it to be used.

If you want it to switch the tool, there's a number of ways that could be accomplished, but a quick fix might be to try to remap it to a shortcut which switches to that tool in your drawing program. That could be done with xsetwacom.

Otherwise this needs work in either the kernel or the userspace drivers.

Xyene, 2020-06-12:
Thanks for the quick reply! I'm afraid I was not very clear about what's going on, especially with regards to orientation... so here's an annotated Huion 420 pen :slightly_smiling_face: 

[image]

When the pen is in proximity, but _not_ touching the pad, pressing _B_ reports `BTN_STYLUS2`, and pressing _A_ reports `BTN_STYLUS` (I'm assuming _B_ is what you mean by "top pen button"?). When the pen is touching the pad, pressing _B_ again reports `BTN_STYLUS2`. I think this is still fine.

The problem is when pressing _A_ while the pen _is_ touching the pad. What happens is:

* the pen reports going out of proximity
* in the next event frame, it:
    * reports coming back into proximity
    * reports `BTN_STYLUS2` being pressed (!!, not `BTN_STYLUS` as it does when it's not down)

This sounds a lot like [the behavior of the Surface 3 pens](https://who-t.blogspot.com/2014/09/stylus-behaviour-on-microsoft-surface-3.html), except the transition is reported as `BTN_STYLUS2` instead of `BTN_TOOL_RUBBER`.

Assuming a mapping of `BTN_STYLUS2` &rarr; eraser, and `BTN_STYLUS` &rarr; something else, this ends up making using button _A_ for anything very finicky: if you try to stop erasing while the pen is down, you can either:

* lift the pen, in which case the moment the pen stops touching the pad but is still in proximity, the pad will revert to reporting `BTN_STYLUS` again and scribble with whatever tool is mapped to that while you're lifting the pen; or
* keep the pen touched down and release the button, in which case you scribble with the regular pen tool

It's actually a bit worse than that in practice, since I think there's also some pressure requirement (roughly > 0.2, it seems) before the tablet switches sending `BTN_STYLUS2` instead of `BTN_STYLUS`... so you can be in contact with the tablet surface and using `BTN_STYLUS`'s tool, but if you apply just a bit too much (> 0.2) pressure, it switches to `BTN_STYLUS2` and you start erasing instead.