DIGImend/digimend-kernel-drivers #418: Raspberry Pi & Huion New1060Plus

pisimpson, 2020-06-14:
I have installed digimend-dkms_9_all.deb on my RaspberryPi 4 Model2 Kernel 4.19.50-v7l+
the install was error free and i plugged my USB based Huion New 1060 Plus into the USB3 port.
I checked with xinput that the tablet and pen stylus were present, all seemed well.

My prefered graphics app is MyPaint version 1.20 and using edit>edit preferences>devices all looks fine with the Pen Stylus and Tablet shown as input devices with Any Tas and scroll 'zoom' available. Exactly the same as my optical mouse.

However when I try to use the graphics tablet and Pen, Nothing happens!!

I checked the tablet by connecting it to my Windows 10 laptop, and it's fine working with MS paint 3d.

It just seems to totally dead when plugged into the debian Raspberry 4.

Any suggestions on where to go next?

jplloyd, 2020-06-14:
Now I wasn't 100% clear on this when we [discussed it in the MyPaint forum](https://community.mypaint.org/t/which-graphic-tablets-models-work-on-mypaint-1-2-0/269/39), but I think it's important to clarify.

When you say "totally dead" and "nothing happens" - does that mean: 

1) Moving the stylus across the tablet does not move the cursor (not in MyPaint, or not anywhere)?

2) Placing/tapping the stylus on the tablet does not produce a click - i.e. you cannot open a menu or click on an icon using the stylus?

pisimpson, 2020-06-14:
Yes it means ... Nothing   moving the stylus doesn't do anything no cursor
movement in MyPaint or anywhere else.
Placing tapping the stylus does not produce a click or any kind of control.

It's as if it wasn't there.

Pi

pisimpson, 2020-06-16:
I see my tablet is registered under the supported tablet with version 8,  I've got version -9 installed I'm assuming that's not the problem?

pisimpson, 2020-06-16:
I've obviously installed something wrong I get no output from the tablet?  

[image]

spbnick, 2020-06-16:
@pisimpson, do you get anything out of `sudo evtest` with the `Pen` device selected, when moving your stylus?

pisimpson, 2020-06-16:
Hi Nikolai  no I dont' get anything from evtest with either the stylus/pen
or the pad selected.

The mouse is fine and gives plenty of feedback.

Pi

pisimpson, 2020-06-17:
I've been incorporation a SNES Gameboy handset and noticed that
occasionally the system needs rebooting to to get evtest to work on the USB
Gameboy port, could my tablet not giving an output be related?

spbnick, 2020-06-17:
> Is there someway i can check for raw data?
Yes. Try running `sudo usbhid-dump -es -m VID:PID`, where `VID` is your tablet's vendor ID and `PID` is its product ID.

pisimpson, 2020-06-17:
[image: 20200617_181302.jpg]

absolutly nothing coming in ........  i tried pressing the pen on the pad
etc. nothing.

>

spbnick, 2020-06-18:
This looks like a hardware failure. Does the same command work e.g. for a USB mouse or a USB keyboard? Does the tablet work on any other computer? Is the pen charged?

pisimpson, 2020-06-18:
The pen is fully charged and when I plug the pad into my windows 10 laptop
it works perfectly using MS Paint 3D.....There's nothing wrong with the
Huion hardware.

It must be something getting in the way on the Linux box.  I've tried
different USB ports, I can swap the mouse and the pad over and the mouse
consistently works fine, the pad never works.

So I can only assume the Raspberry Pi hardware is also fine. It is a
software fault.

I'm musing on my next move.

spbnick, 2020-06-18:
Is there anything strange in `dmesg` output?

pisimpson, 2020-06-18:
It does mention the pen andd pad:
 4.712725] sd 0:0:0:0: Attached scsi generic sg0 type 0
[    4.849461] input: HID 256c:006e Pen as
/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4.4/1-1.4.4.1/1-1.4.4.1:1.0/0003:256C:006E.0005/input/input4
[    4.851175] input: HID 256c:006e Pad as
/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4.4/1-1.4.4.1/1-1.4.4.1:1.0/0003:256C:006E.0005/input/input5
[    4.864094] uclogic 0003:256C:006E.0005: input,hidraw4: USB HID v1.11
Keypad [HID 256c:006e] on usb-0000:01:00.0-1.4.4.1/input0
[    4.865004] uclogic 0003:256C:006E.0006: No inputs registered, leaving

pisimpson, 2020-06-18:
lsusb -t looks interesting?
pi@raspberrypi:~ $ lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 1: Dev 5, If 0, Class=Human Interface Device,
Driver=usbhid, 1.5M
            |__ Port 1: Dev 5, If 1, Class=Human Interface Device,
Driver=usbhid, 1.5M
            |__ Port 2: Dev 7, If 0, Class=Human Interface Device,
Driver=usbhid, 1.5M
            |__ Port 4: Dev 9, If 0, Class=Human Interface Device,
Driver=usbhid, 1.5M
        |__ Port 4: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 4: Dev 8, If 0, Class=Mass Storage,
Driver=usb-storage, 480M
            |__ Port 1: Dev 6, If 0, Class=Human Interface Device,
Driver=usbhid, 12M
            |__ Port 1: Dev 6, If 1, Class=Human Interface Device,
Driver=usbhid, 12M

noob117, 2020-08-26:
How did you install the digimend driver? Is there a special version for arm processors?

yzotov, 2021-01-01:
I've just installed fresh Raspberry OS on new Raspberry Pi 4, than I've installed deb package from digimend v10 release and connected Huion Pro 12 to it. When I connected it to USB-3 port the pen didn't work, when I just reconnected to USB-2 port the pen started to work properly.