DIGImend/digimend-kernel-drivers #420: H1161 Works Great!  .. a little nit ..

tpoliquin, 2020-06-19:
Running Linux 16.04. 4.8.0-51-generic. Used the dkms install (June18 2020)
it works great! Remapped to left hand operation, and the buttons. Fantastic in Krita!
Thank you!

I do have a nit issue.
The hover distance feels big to me ( about an inch).

It appears that there is a parameter in xsetwacom,
`CursorProximity`
When I try to get the parameter it complains ..

```
xsetwacom` --get "HUION Huion Tablet Pad pad"  CursorProximity
Property 'Wacom Proximity Threshold' does not exist on device.
xsetwacom --get "HUION Huion Tablet Pen stylus"  CursorProximity`
Property 'Wacom Proximity Threshold' does not exist on device.

```
Any thoughts?

Again, it's a nit issue and I'm super happy right now with
the way it works.

_FWIW .. I bought the "2019" H1161 off Amazon. There appeared
to be a "2020" version which the comments implied wouldn't
work in Linux. I contacted Huion and they said that there was
only one version._

For fun I've included my 'tabset' script for Krita ..
```
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 3 "key +ctrl +shift a -ctrl -shift b"
sleep 0.5
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 1 "key +ctrl R -ctrl"
sleep 0.5
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 2 "key t"
sleep 0.5
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 8 "key +space"
sleep 0.5
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 9 "key +ctrl z -ctrl"
sleep 0.5
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 10 "key +ctrl +shift z -shift -ctrl"
sleep 0.5
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 11 "key +ctrl +space"
sleep 0.5
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 12 3
sleep 0.5
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 13 "key b"
sleep 0.5
xsetwacom --set 'HUION Huion Tablet Pad pad' Button 14 "key t"
sleep 0.5
xsetwacom --set 'HUION Huion Tablet Pen stylus' Rotate half
sleep 0.5
xsetwacom --set 'HUION Huion Tablet Pen stylus' suppress 30

```
The 'supress 30' makes tool tips work better (allows jitter in pen movement)

I know, I know,  .. I'm lazy and haven't tuned the sleep length .. or even if
it's necessary. I was just anxious to get it working and drawing!

Tom

spbnick, 2020-06-20:
Hi @tpoliquin, thank you for the report. I personally never touched H1161, so can't say if it should work or not, or which versions there are, but I'm glad yours does.

You cannot set proximity threshold, because none of the non-Wacom tablets support configuring it, i.e. there's no support in their firmware or maybe even hardware.

tpoliquin, 2020-06-21:
Thanks for the quick reply!

I'll adapt.

Since I'm using the tablet left handed I added these two lines to my setup script
to flip the touch strip which now works great.
```
xsetwacom set "HUION Huion Tablet Touch Strip pad" "AbsWheelUp" "+5"
xsetwacom set "HUION Huion Tablet Touch Strip pad" "AbsWheelDown" "+4"
```
Thanks again,

Tom