DIGImend/digimend-kernel-drivers #422: Fedora32 + ACS patch override -> dkms install issue

robert-clayton, 2020-06-21:
Sorry if this is lacking some information, I initially set my Huion 16 Pro up last week.

I cloned master and followed the README steps to do a dkms install, but this is the output I receive:

```sh
[robert@localhost digimend-kernel-drivers]$ sudo make dkms_install  
dkms add .

Creating symlink /var/lib/dkms/digimend/10/source ->
                 /usr/src/digimend-10

DKMS: add completed.
dkms build digimend/10

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j12 KERNELRELEASE=5.6.18-301.acspatch.fc32.x86_64 KVERSION=5.6.18-301.acspatch.fc32.x86_64...
cleaning build area...

DKMS: build completed.
dkms install digimend/10

hid-kye.ko.xz:
Running module version sanity check.
 - Original module
   - Found /lib/modules/5.6.18-301.acspatch.fc32.x86_64/extra/hid-kye.ko.xz
   - Storing in /var/lib/dkms/digimend/original_module/5.6.18-301.acspatch.fc32.x86_64/x86_64/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/5.6.18-301.acspatch.fc32.x86_64/extra/

hid-uclogic.ko.xz:
Running module version sanity check.
 - Original module
   - Found /lib/modules/5.6.18-301.acspatch.fc32.x86_64/extra/hid-uclogic.ko.xz
   - Storing in /var/lib/dkms/digimend/original_module/5.6.18-301.acspatch.fc32.x86_64/x86_64/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/5.6.18-301.acspatch.fc32.x86_64/extra/

hid-polostar.ko.xz:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/5.6.18-301.acspatch.fc32.x86_64/extra/

hid-viewsonic.ko.xz:
Running module version sanity check.
 - Original module
   - Found /lib/modules/5.6.18-301.acspatch.fc32.x86_64/extra/hid-viewsonic.ko.xz
   - Storing in /var/lib/dkms/digimend/original_module/5.6.18-301.acspatch.fc32.x86_64/x86_64/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/5.6.18-301.acspatch.fc32.x86_64/extra/
Adding any weak-modules
/etc/dracut.conf.d/local.conf:add_drivers += "vfio vfio_iommu_type1 vfio_pci vfio_virqfd"
/etc/dracut.conf.d/local.conf:install_items += "/sbin/vfio-pci-override.sh /usr/bin/find /usr/bin/dirname"

dracut: WARNING: <key>+=" <values> ": <values> should have surrounding white spaces!
dracut: WARNING: This will lead to unwanted side effects! Please fix the configuration file.

/etc/dracut.conf.d/local.conf: line 1: add_drivers: command not found
/etc/dracut.conf.d/local.conf: line 2: install_items: command not found
/etc/dracut.conf.d/local.conf:add_drivers += "vfio vfio_iommu_type1 vfio_pci vfio_virqfd"
/etc/dracut.conf.d/local.conf:install_items += "/sbin/vfio-pci-override.sh /usr/bin/find /usr/bin/dirname"

dracut: WARNING: <key>+=" <values> ": <values> should have surrounding white spaces!
dracut: WARNING: This will lead to unwanted side effects! Please fix the configuration file.

/etc/dracut.conf.d/local.conf: line 1: add_drivers: command not found
/etc/dracut.conf.d/local.conf: line 2: install_items: command not found
/etc/dracut.conf.d/local.conf:add_drivers += "vfio vfio_iommu_type1 vfio_pci vfio_virqfd"
/etc/dracut.conf.d/local.conf:install_items += "/sbin/vfio-pci-override.sh /usr/bin/find /usr/bin/dirname"

dracut: WARNING: <key>+=" <values> ": <values> should have surrounding white spaces!
dracut: WARNING: This will lead to unwanted side effects! Please fix the configuration file.

/etc/dracut.conf.d/local.conf: line 1: add_drivers: command not found
/etc/dracut.conf.d/local.conf: line 2: install_items: command not found

depmod....

DKMS: install completed.
install -D -m 0644 depmod.conf /etc/depmod.d/digimend.conf
depmod -a
set -e -x; \
if test -e /usr/lib/dracut/dracut.conf.d; then \
    install -m 0644 dracut.conf /usr/lib/dracut/dracut.conf.d/90-digimend.conf; \
    dracut --force; \
fi
+ test -e /usr/lib/dracut/dracut.conf.d
+ install -m 0644 dracut.conf /usr/lib/dracut/dracut.conf.d/90-digimend.conf
+ dracut --force
/etc/dracut.conf.d/local.conf:add_drivers += "vfio vfio_iommu_type1 vfio_pci vfio_virqfd"
/etc/dracut.conf.d/local.conf:install_items += "/sbin/vfio-pci-override.sh /usr/bin/find /usr/bin/dirname"

dracut: WARNING: <key>+=" <values> ": <values> should have surrounding white spaces!
dracut: WARNING: This will lead to unwanted side effects! Please fix the configuration file.

/etc/dracut.conf.d/local.conf: line 1: add_drivers: command not found
/etc/dracut.conf.d/local.conf: line 2: install_items: command not found

```
I initially ignored it as everything on the tablet seemed to work other than the touch strip after I set up a key mapping script, figuring I'd find the issue later. Well, I've exhausted my (poor) ability to figure out the issue. 

Various related outputs:
```sh
[robert@localhost ~]$ xsetwacom --list devices
HID 256c:006d Pen stylus                id: 10  type: STYLUS    
HID 256c:006d Pad pad                   id: 11  type: PAD       
HID 256c:006d Touch Strip pad           id: 12  type: PAD
```
```sh
[robert@localhost ~]$ xinput list
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ HID 256c:006d Pen stylus                  id=10   [slave  pointer  (2)]
⎜   ↳ HID 256c:006d Pad pad                     id=11   [slave  pointer  (2)]
⎜   ↳ HID 256c:006d Touch Strip pad             id=12   [slave  pointer  (2)]
⎜   ↳ HID 256c:006d Dial                        id=13   [slave  pointer  (2)]
⎜   ↳ Razer Razer Naga Chroma                   id=14   [slave  pointer  (2)]
⎜   ↳ Razer Razer Naga Chroma Consumer Control  id=16   [slave  pointer  (2)]
⎜   ↳ Logitech G815 RGB MECHANICAL GAMING KEYBOARD Mouse        id=21   [slave  pointer  (2)]
```
When I tried to see the results of testing the touch strip's inputs, I got nothing.
```sh
[robert@localhost ~]$ xinput test "HID 256c:006d Touch Strip pad"
^C
```

Any help would be appreciated.

robert-clayton, 2020-06-21:
This might deserve a separate issue, but it's just `sudo dnf -y kernel-devel` for Fedora-based systems now.

Popolon, 2020-09-28:
Please could you report a full diagnostics as in http://digimend.github.io/support/howto/trbl/diagnostics/
I have a Kamvas pro 20, have about the same issues, I added #463 for my version this is probably the same protocol in both, as they are from the same product line.

 Shinmera : I use ArchLinux too, you can find all the tool needed in arch. package usbutils and AUR package uclogic-tools

Popolon, 2020-09-29:
@Shinmera I managed to have most of things working (strips + most buttons + tilt) by installing on Arch Linux, the AUR package digimend-kernel-drivers-dkms-git  https://aur.archlinux.org/packages/digimend-kernel-drivers-dkms-git/

The PKGBUILD receipe is clear and what is made inside could be reused on other distributions.

Shinmera, 2020-10-05:
I've updated everything to the latest arch (5.8.13-arch1-1) and reinstalled the package, but it didn't improve the situation. Touch strip still does not report any xinput events on any of the devices.

I've attached all the diagnostic files.
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5326279/descriptors.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5326280/frame_buttons.txt)
[frame_strip.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5326281/frame_strip.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5326283/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5326284/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5326285/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5326286/pen_pressure.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5326287/pen_tilt.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5326288/probe.txt)

spbnick, 2020-10-05:
Does any one of you get anything out of `evtest` for the "Touch Strip" device?

Shinmera, 2020-10-05:
Yes, I get events for the strip device (first moving down, then up):
```
Input device name: "HID 256c:006d Touch Strip"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 256 (BTN_0)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max        1
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max        1
    Event code 8 (ABS_WHEEL)
      Value      0
      Min        0
      Max        7
      Resolution       3
    Event code 40 (ABS_MISC)
      Value      0
      Min        0
      Max      255
Properties:
Testing ... (interrupt to exit)
Event: time 1601888878.645171, type 3 (EV_ABS), code 40 (ABS_MISC), value 15
Event: time 1601888878.645171, -------------- SYN_REPORT ------------
Event: time 1601888878.983132, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 1
Event: time 1601888878.983132, -------------- SYN_REPORT ------------
Event: time 1601888879.177166, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 2
Event: time 1601888879.177166, -------------- SYN_REPORT ------------
Event: time 1601888879.427168, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 3
Event: time 1601888879.427168, -------------- SYN_REPORT ------------
Event: time 1601888879.621173, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 4
Event: time 1601888879.621173, -------------- SYN_REPORT ------------
Event: time 1601888879.835169, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 5
Event: time 1601888879.835169, -------------- SYN_REPORT ------------
Event: time 1601888880.159170, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 6
Event: time 1601888880.159170, -------------- SYN_REPORT ------------
Event: time 1601888880.535172, type 3 (EV_ABS), code 40 (ABS_MISC), value 0
Event: time 1601888880.535172, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 0
Event: time 1601888880.535172, -------------- SYN_REPORT ------------
Event: time 1601888882.539133, type 3 (EV_ABS), code 40 (ABS_MISC), value 15
Event: time 1601888882.539133, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 6
Event: time 1601888882.539133, -------------- SYN_REPORT ------------
Event: time 1601888882.755131, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 5
Event: time 1601888882.755131, -------------- SYN_REPORT ------------
Event: time 1601888882.981133, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 4
Event: time 1601888882.981133, -------------- SYN_REPORT ------------
Event: time 1601888883.165171, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 3
Event: time 1601888883.165171, -------------- SYN_REPORT ------------
Event: time 1601888883.347168, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 2
Event: time 1601888883.347168, -------------- SYN_REPORT ------------
Event: time 1601888883.529194, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 1
Event: time 1601888883.529194, -------------- SYN_REPORT ------------
Event: time 1601888883.691172, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 0
Event: time 1601888883.691172, -------------- SYN_REPORT ------------
Event: time 1601888883.971171, type 3 (EV_ABS), code 40 (ABS_MISC), value 0
Event: time 1601888883.971171, -------------- SYN_REPORT ------------
```

spbnick, 2020-10-05:
The evtest output looks OK. Please try the latest release (v10) and see if it works. Post your `xorg.0.log` here or the equivalent `journal` output, if it doesn't.

spbnick, 2020-10-05:
Also please see the release notes regarding the limited support fro touch strips/rings.

Popolon, 2020-10-06:
I tried v10 on ArchLinux, and Huion Kamvas Pro 20 (see #463), and I have no more cursor. Ptxconf, see the device, but the mouse cursor isn't moved anymore by stylus. I come back to previous release (10.rc0.r1 (g691dae8)) and it works again.

spbnick, 2020-10-07:
@Popolon, that is troubling news. Could you post an `evtest` output for your stylus with v10?

Popolon, 2020-10-08:
I have the same problem with the current one (10.rc0.r1) one time yesterday. This time, I power on tablet before or at the same time than computer, and i tried to use pen, before XFCE launch endend. That was perhaps the case too the day of my try. I will retry with v10, waiting for full boot before powering on the tablet.

Popolon, 2020-10-08:
OK, it worked this time. I need to investigate why for, when i touch the screen with stylus before the end, of the startup sequence (and at wich stage ?) the stlus is no more detected at all. What I remember, is that PTXConf, seen is as stylus, I used the 50-wacom.conf to map it to wacom tablet. I will make more try in different cases.

Popolon, 2020-10-08:
So I just tried more cases :
* With /etc/X11/xorg.conf.d/50-wacom.conf (with the Huion Kamava pro 20 usb ID) OK and the name is Stylus, not Pen in this case in Ptxconf.
* When I just quit the session and restart the XFCE session, the stylus keep set in the tablet space.
* I stop/start-ed lxdm in a terminal, everything worked fine, the stylus was mapped on the 2 screen (Ptxconf and XFCE, as far I know has no mean to memorize it).
* I rebooted the computer with the tablet on, using pen near the screen the whole boot sequence to try to have the bug the earliest possible, everything worked fine, and strangely (perhaps because by defaut, before XFCE start, the screens are set as mirror, the pencil was set in my computer main screen only,  and can't access the tablet. but still work. at the end of the boot, I set it correctly with PTXconf, everything works finely.
I don't remember what I done when I had the bug yesterday, but at least the new version of the driver works finely. This is as most a common bug, or a bug somwhere in Xinput or tablet itself...
By default the bug is not seen with evtest/PTXconf, i need to move the pen near the screen to make it discovered and the device appear.

Popolon, 2020-10-09:
again the bug today, I just started the tablet after about a day of usage of the system. the stylus is not seen even if I move him near from the screen. Strangely device 21 is missing 
<pre>
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Power Button
/dev/input/event1:	Power Button
/dev/input/event2:	PC Speaker
/dev/input/event3:	Eee PC WMI hotkeys
/dev/input/event4:	Video Bus
/dev/input/event5:	HDA Intel PCH Front Mic
/dev/input/event6:	HDA Intel PCH Rear Mic
/dev/input/event7:	HDA Intel PCH Line
/dev/input/event8:	HDA Intel PCH Line Out Front
/dev/input/event9:	HDA Intel PCH Line Out Surround
/dev/input/event10:	HDA Intel PCH Line Out CLFE
/dev/input/event11:	HDA Intel PCH Line Out Side
/dev/input/event12:	HDA Intel PCH Front Headphone
/dev/input/event13:	HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event14:	HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event15:	Logitech USB Optical Mouse
/dev/input/event16:	NOVATEK USB Keyboard
/dev/input/event17:	NOVATEK USB Keyboard System Control
/dev/input/event18:	NOVATEK USB Keyboard Consumer Control
/dev/input/event19:	NOVATEK USB Keyboard
/dev/input/event20:	Razer Razer Electra V2 USB Consumer Control
/dev/input/event22:	Tablet Monitor
/dev/input/event23:	Tablet Monitor Pad
/dev/input/event24:	Tablet Monitor Touch Strip
/dev/input/event25:	Tablet Monitor Dial
Select the device event number [0-25]: 
</pre>
I turn off the tablet and turn on again, I can move the cursor  with stylus, but don't see the stylus on evtest, and there is one less device (the tablet monitor devices  was reassigned:
<pre>
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Power Button
/dev/input/event1:	Power Button
/dev/input/event2:	PC Speaker
/dev/input/event3:	Eee PC WMI hotkeys
/dev/input/event4:	Video Bus
/dev/input/event5:	HDA Intel PCH Front Mic
/dev/input/event6:	HDA Intel PCH Rear Mic
/dev/input/event7:	HDA Intel PCH Line
/dev/input/event8:	HDA Intel PCH Line Out Front
/dev/input/event9:	HDA Intel PCH Line Out Surround
/dev/input/event10:	HDA Intel PCH Line Out CLFE
/dev/input/event11:	HDA Intel PCH Line Out Side
/dev/input/event12:	HDA Intel PCH Front Headphone
/dev/input/event13:	HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event14:	HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event15:	Logitech USB Optical Mouse
/dev/input/event16:	NOVATEK USB Keyboard
/dev/input/event17:	NOVATEK USB Keyboard System Control
/dev/input/event18:	NOVATEK USB Keyboard Consumer Control
/dev/input/event19:	NOVATEK USB Keyboard
/dev/input/event20:	Razer Razer Electra V2 USB Consumer Control
/dev/input/event21:	Tablet Monitor
/dev/input/event22:	Tablet Monitor Pad
/dev/input/event23:	Tablet Monitor Touch Strip
/dev/input/event24:	Tablet Monitor Dial
Select the device event number [0-24]: 
</pre>

Popolon, 2020-10-09:
But at the same time I am able to configure it with PTXconf.

See [image]!

Popolon, 2020-10-16:
With driver 10.rc0.r1.g691dae8 Gnome see the tablet, and allow to configure it.
with driver 10.r0.gec5c16d doesn't see the tablet, and then doesn't allow to configure it.

evtest with 10.rc0.r1.g691dae8:
<pre>
/dev/input/event22:	Tablet Monitor Pen
/dev/input/event23:	Tablet Monitor Pad
/dev/input/event24:	Tablet Monitor Touch Strip
/dev/input/event25:	Tablet Monitor Dial
Select the device event number [0-25]: 22      
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
Input device name: "Tablet Monitor Pen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 320 (BTN_TOOL_PEN)
    Event code 330 (BTN_TOUCH)
    Event code 331 (BTN_STYLUS)
    Event code 332 (BTN_STYLUS2)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max    86970
      Resolution     200
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max    47752
      Resolution     200
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     8191
    Event code 26 (ABS_TILT_X)
      Value      0
      Min      -60
      Max       60
      Resolution      57
    Event code 27 (ABS_TILT_Y)
      Value      0
      Min      -60
      Max       60
      Resolution      57
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
</pre>

evtest with 10.r0.gec5c16d (october version)
<pre>
/dev/input/event22:	Tablet Monitor
/dev/input/event23:	Tablet Monitor Pad
/dev/input/event24:	Tablet Monitor Touch Strip
/dev/input/event25:	Tablet Monitor Dial
Select the device event number [0-25]: 22
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
Input device name: "Tablet Monitor"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 320 (BTN_TOOL_PEN)
    Event code 330 (BTN_TOUCH)
    Event code 331 (BTN_STYLUS)
    Event code 332 (BTN_STYLUS2)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value  24967
      Min        0
      Max    86970
      Resolution     200
    Event code 1 (ABS_Y)
      Value  36468
      Min        0
      Max    47752
      Resolution     200
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     8191
    Event code 26 (ABS_TILT_X)
      Value    -23
      Min      -60
      Max       60
      Resolution      57
    Event code 27 (ABS_TILT_Y)
      Value     36
      Min      -60
      Max       60
      Resolution      57
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
  Property type 0 (INPUT_PROP_POINTER)
Testing ... (interrupt to exit)
</pre>
I tried :
start gnome in newer : it don't see it
* changed version from newer to  older (install package, rmmod hid_uclogic; modprobe hid_uclogic)
gnome detect it,
* changed version from  older to newer (install package, rmmod hid_uclogic; modprobe hid_uclogic)
gnome doesn't detect it again (the evtest output)
* changed version from newer to  older (install package, rmmod hid_uclogic; modprobe hid_uclogic)
gnome detect it,(the other evtest output)

Popolon, 2020-10-16:
xsetwacom display  'stylus' in one case 'Pen stylus' in other case:

10.rc0.r1.g691dae8
<pre>
$ xsetwacom --list devices
Tablet Monitor Pad pad          	id: 15	type: PAD       
Tablet Monitor Dial pad         	id: 16	type: PAD       
Tablet Monitor Touch Strip pad  	id: 17	type: PAD       
Tablet Monitor Pen stylus       	id: 18	type: STYLUS    
</pre>

10.r0.gec5c16d 
<pre>
$ xsetwacom --list devices
Tablet Monitor Touch Strip pad  	id: 15	type: PAD       
Tablet Monitor Pad pad          	id: 16	type: PAD       
Tablet Monitor stylus           	id: 17	type: STYLUS    
Tablet Monitor Dial pad         	id: 18	type: PAD       
</pre>

dentex, 2020-12-01:
I have a Kamvas Pro 13 and the only thing that doesn't work is that touch strip.

I get a nice output with `evtest`, but besides that, I had no luck trying to mess with `xsetwacom` (see https://github.com/dwilches/huion-kamvas-pro-13/issues/1)
````
~$ sudo evtest 
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	Video Bus
/dev/input/event5:	Tablet Monitor
/dev/input/event6:	HP TrueVision HD Camera: HP Tru
/dev/input/event7:	HP WMI hotkeys
/dev/input/event8:	MOSART Semi. 2.4G Wireless Mouse
/dev/input/event9:	MOSART Semi. 2.4G Wireless Mouse
/dev/input/event10:	HP Wireless hotkeys
/dev/input/event11:	Tablet Monitor Pad
/dev/input/event12:	Tablet Monitor Touch Strip
/dev/input/event13:	Tablet Monitor Dial
/dev/input/event14:	ELAN0712:00 04F3:30FD Mouse
/dev/input/event15:	ELAN0712:00 04F3:30FD Touchpad
/dev/input/event16:	HDA Intel PCH Mic
/dev/input/event17:	HDA Intel PCH Headphone
/dev/input/event18:	HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event19:	HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event20:	HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event21:	HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event22:	HDA Intel PCH HDMI/DP,pcm=10
/dev/input/event23:	HDA Intel PCH HDMI/DP,pcm=11
/dev/input/event24:	HDA Intel PCH HDMI/DP,pcm=12
Select the device event number [0-24]: 12
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6d version 0x111
Input device name: "Tablet Monitor Touch Strip"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 256 (BTN_0)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max        1
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max        1
    Event code 8 (ABS_WHEEL)
      Value      0
      Min        0
      Max        7
      Resolution       3
    Event code 40 (ABS_MISC)
      Value      0
      Min        0
      Max      255
Properties:
Testing ... (interrupt to exit)
Event: time 1606839487.853348, type 3 (EV_ABS), code 40 (ABS_MISC), value 15
Event: time 1606839487.853348, -------------- SYN_REPORT ------------
Event: time 1606839488.001379, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 1
Event: time 1606839488.001379, -------------- SYN_REPORT ------------
Event: time 1606839488.149384, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 2
Event: time 1606839488.149384, -------------- SYN_REPORT ------------
Event: time 1606839488.309370, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 3
Event: time 1606839488.309370, -------------- SYN_REPORT ------------
Event: time 1606839488.507360, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 4
Event: time 1606839488.507360, -------------- SYN_REPORT ------------
[... 55 lines cut ...]
Event: time 1606839505.013615, -------------- SYN_REPORT ------------
Event: time 1606839505.211609, type 3 (EV_ABS), code 40 (ABS_MISC), value 0
Event: time 1606839505.211609, -------------- SYN_REPORT ------------
Event: time 1606839507.225635, type 3 (EV_ABS), code 40 (ABS_MISC), value 15
Event: time 1606839507.225635, -------------- SYN_REPORT ------------
Event: time 1606839507.385632, type 3 (EV_ABS), code 40 (ABS_MISC), value 0
Event: time 1606839507.385632, -------------- SYN_REPORT ------------
Event: time 1606839507.895637, type 3 (EV_ABS), code 40 (ABS_MISC), value 15
Event: time 1606839507.895637, -------------- SYN_REPORT ------------
Event: time 1606839508.113466, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 1
Event: time 1606839508.113466, -------------- SYN_REPORT ------------
Event: time 1606839508.301568, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 2
Event: time 1606839508.301568, -------------- SYN_REPORT ------------
Event: time 1606839508.499706, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 3
Event: time 1606839508.499706, -------------- SYN_REPORT ------------
Event: time 1606839508.697458, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 4
Event: time 1606839508.697458, -------------- SYN_REPORT ------------
Event: time 1606839508.885629, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 5
Event: time 1606839508.885629, -------------- SYN_REPORT ------------
Event: time 1606839509.033673, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 6
Event: time 1606839509.033673, -------------- SYN_REPORT ------------
Event: time 1606839509.353662, type 3 (EV_ABS), code 40 (ABS_MISC), value 0
Event: time 1606839509.353662, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 0
Event: time 1606839509.353662, -------------- SYN_REPORT ------------
```