DIGImend/digimend-kernel-drivers #423: Huion Q11K V2 pointer jumpy when using wireless mode

rorojasq, 2020-06-21:
When using the wireless mode, pressing pen buttons or putting it near the tablet also emits
ABS_{X,Y} events with a value of 0, causing the pointer to jump to the upper left corner of the screen.

OS tested:
- Ubuntu 20.04, using included driver and installing from repo with `dkms_install`
- Ubuntu 18.04, using included driver

Relevant evtest output:
```
Event: time 1592767143.841946, -------------- SYN_REPORT ------------
Event: time 1592767154.762301, type 1 (EV_KEY), code 320 (BTN_TOOL_PEN), value 1
Event: time 1592767154.762301, type 3 (EV_ABS), code 0 (ABS_X), value 0
Event: time 1592767154.762301, type 3 (EV_ABS), code 1 (ABS_Y), value 0
Event: time 1592767154.762301, type 3 (EV_ABS), code 26 (ABS_TILT_X), value 2
Event: time 1592767154.762301, -------------- SYN_REPORT ------------
Event: time 1592767154.764288, type 3 (EV_ABS), code 0 (ABS_X), value 34938
Event: time 1592767154.764288, type 3 (EV_ABS), code 1 (ABS_Y), value 24770
Event: time 1592767154.764288, type 3 (EV_ABS), code 26 (ABS_TILT_X), value 0
Event: time 1592767154.764288, -------------- SYN_REPORT ------------
...
Event: time 1592767092.288340, type 3 (EV_ABS), code 0 (ABS_X), value 38575
Event: time 1592767092.288340, type 3 (EV_ABS), code 1 (ABS_Y), value 22130
Event: time 1592767092.288340, -------------- SYN_REPORT ------------
Event: time 1592767092.416338, type 3 (EV_ABS), code 0 (ABS_X), value 0
Event: time 1592767092.416338, type 3 (EV_ABS), code 1 (ABS_Y), value 0
Event: time 1592767092.416338, -------------- SYN_REPORT ------------
Event: time 1592767092.424306, type 4 (EV_MSC), code 4 (MSC_SCAN), value d0044
Event: time 1592767092.424306, type 1 (EV_KEY), code 331 (BTN_STYLUS), value 1
Event: time 1592767092.424306, type 3 (EV_ABS), code 0 (ABS_X), value 38751
Event: time 1592767092.424306, type 3 (EV_ABS), code 1 (ABS_Y), value 22262
Event: time 1592767092.424306, -------------- SYN_REPORT ------------
...
Event: time 1592767155.864330, -------------- SYN_REPORT ------------
Event: time 1592767155.908327, type 4 (EV_MSC), code 4 (MSC_SCAN), value d0046
Event: time 1592767155.908327, type 1 (EV_KEY), code 332 (BTN_STYLUS2), value 0
Event: time 1592767155.908327, -------------- SYN_REPORT ------------
Event: time 1592767155.954333, type 3 (EV_ABS), code 0 (ABS_X), value 0
Event: time 1592767155.954333, type 3 (EV_ABS), code 1 (ABS_Y), value 0
Event: time 1592767155.954333, type 3 (EV_ABS), code 26 (ABS_TILT_X), value 6
Event: time 1592767155.954333, -------------- SYN_REPORT ------------
Event: time 1592767155.956325, type 3 (EV_ABS), code 0 (ABS_X), value 34848
Event: time 1592767155.956325, type 3 (EV_ABS), code 1 (ABS_Y), value 24623
Event: time 1592767155.956325, type 3 (EV_ABS), code 26 (ABS_TILT_X), value 5
Event: time 1592767155.956325, -------------- SYN_REPORT ------------
```

spbnick, 2020-06-22:
Thank you for the report, @rorojasq. Do you think you can also capture `sudo usbhid-dump -es` output for the same input?

rorojasq, 2020-06-22:
Sure.
[usbhid-dump.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4814252/usbhid-dump.txt)

spbnick, 2020-10-04:
Thank you, @rorojasq. I can see the zero-coordinate report in your usbhid-dump.txt. I cannot reproduce that with my tablet using wireless, though. I'm not sure what to do with this, aside from banning zero-coordinates altogether, which would be hard to target to exactly this tablet's wireless interface due to the fact that Huion chose to make all tablets (and even wireless interfaces) look essentially the same to the host. This is especially hard to justify because it might be just a hardware/firmware issue (which is more likely as my Q11Kv2 doesn't exhibit it).

Could you try updating the firmware on your tablet, and/or contacting Huion about your issue before I consider implementing such a change?

Thank you.
Moving this issue to the next release for now.

rorojasq, 2020-10-08:
I don't think it's the tablet because:
- The firmware is updated.
- works fine when using an usb cable.
- Works fine on windows, both wired and wireless.

Also, look at this log file: [evtest_log.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5351517/evtest_log.txt)
I moved the pen outside the drawing area, past the lower left corner. Normally it shouldn't produce `ABS_*` events. However, on wireless it produces motion events with a value of 0 followed by events with the max values.
This appears to happen randomly.

Some other issue that may be related, is that the current master, has crashed Xorg twice with a pen button event on wireless.
It goes something like this:
1. After some use, the tablet stops responding.
2. I remove and reattach the wireless receiver.
3. After more use, at the same moment I press a button, Xorg crashes.

spbnick, 2020-10-09:
Curiouser and curiouser. Haven't seen that behavior from my tablet, perhaps *my* firmware is old. However, X.org shouldn't crash no matter what the kernel driver sends and that should be reported to your distribution/X.org maintainers/X.org driver maintainers.

lukasz-szczur, 2021-03-19:
I have a similar problem with the Wi-Fi mode in Huion WH1409 V2.
Clicking on a stylus button (or initial detection of stylus) causes jumping of the cursor.
It is very frustrating because I use the button to move the view.
The problem does not occur on Windows 10 (tested using cable and Wi-Fi) and Debian (tested using cable).
It occurs on Debian even if I disable other Wi-Fi devices.

[command] cat /etc/debian_version
10.8

[command] uname -a
Linux luks 5.10.0-0.bpo.3-amd64 #1 SMP Debian 5.10.13-1~bpo10+1 (2021-02-11) x86_64 GNU/Linux

[command] sudo dkms status
digimend, 10, 5.10.0-0.bpo.3-amd64, x86_64: installed

[command] xinput --list
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ Logitech K830                             id=11   [slave  pointer  (2)]
⎜   ↳ Logitech MX Anywhere 2S                   id=13   [slave  pointer  (2)]
⎜   ↳ HUION Huion Tablet stylus                 id=8    [slave  pointer  (2)]
⎜   ↳ HUION Huion Tablet Pad pad                id=9    [slave  pointer  (2)]
⎜   ↳ HUION Huion Tablet Touch Strip pad        id=10   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Power Button                              id=7    [slave  keyboard (3)]
    ↳ Logitech K830                             id=12   [slave  keyboard (3)]
    ↳ Logitech MX Anywhere 2S                   id=14   [slave  keyboard (3)]

[command] xsetwacom --list devices
HUION Huion Tablet stylus               id: 8   type: STYLUS    
HUION Huion Tablet Pad pad              id: 9   type: PAD       
HUION Huion Tablet Touch Strip pad      id: 10  type: PAD

[command] xinput test 8
motion a[0]=0 a[1]=0 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=6542 a[1]=6380 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=13089 a[1]=12764 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26216 a[1]=25538 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26238 a[1]=25536 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26262 a[1]=25530 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26286 a[1]=25529 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26311 a[1]=25532 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26333 a[1]=25539 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26352 a[1]=25550 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26371 a[1]=25559 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26390 a[1]=25568 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26409 a[1]=25576 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26428 a[1]=25584 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26448 a[1]=25590 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26470 a[1]=25594 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26492 a[1]=25598 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26515 a[1]=25601 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26536 a[1]=25606 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26551 a[1]=25604 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26557 a[1]=25596 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=26556 a[1]=25579 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=0 a[1]=0 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7021 a[1]=6263 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=14056 a[1]=12527 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28216 a[1]=25061 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28272 a[1]=25068 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28331 a[1]=25081 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28389 a[1]=25100 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28447 a[1]=25119 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28503 a[1]=25127 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28555 a[1]=25118 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28604 a[1]=25095 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28644 a[1]=25068 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
[... 74 lines cut ...]
motion a[0]=29039 a[1]=25185 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=29023 a[1]=25201 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=29008 a[1]=25207 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28991 a[1]=25204 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28976 a[1]=25192 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28963 a[1]=25182 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28950 a[1]=25188 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=0 a[1]=0 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7162 a[1]=6408 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=27 a[1]=30 
motion a[0]=14329 a[1]=12799 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=21501 a[1]=19197 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28677 a[1]=25590 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28687 a[1]=25580 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28689 a[1]=25594 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28680 a[1]=25607 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28661 a[1]=25632 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28642 a[1]=25661 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28624 a[1]=25689 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=28605 a[1]=25715 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=0 a[1]=0 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=8000 a[1]=6174 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=27 a[1]=30 
motion a[0]=16008 a[1]=12346 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=24021 a[1]=18513 a[2]=0 a[3]=0 a[4]=0 a[5]=-900

mikesbytes, 2022-09-19:
This issue has not been active in some time, but I am having a similar issue with my WH1409 v2. However it has only recently started occurring. For some time, I was affected by the bug in #626. Prior to the bug, I did not experience the issue. After the fix, it is now happening. I can't say when exactly it started occurring, as I used the tablet in wired mode for a couple months to work around the aforementioned bug.