DIGImend/digimend-kernel-drivers #424: XP-PEN VID:PID List

megamuteki, 2020-06-22:
1  Deco 01 			28BD / 0902
	2  Deco 01 			28BD / 0042
	3  G640S  			28BD / 0906
	4  Star 03  		28BD / 0907
	5  Deco 02  		28BD / 0803
	6  Artist 12  		28BD / 080A
	7  Artist 13.3  	28BD / 000B
	8  Artist 15.6  	28BD / 000C
	9  Artist 15.6 Pro  28BD / 090D
	10 Star 05   		28BD / 0071
	11 Star 04   		28BD / 0074
	12 G430_G540   		28BD / 0075
	13 Star 06   		28BD / 0078
	14 Star 06C   		28BD / 0062
	15 Deco 03   		28BD / 0096
	16 Star 03   		28BD / 0081
	17 G640  			28BD / 0094
	18 G540 Pro   		28BD / 0061    
	19 Artist 22E Pro   28BD / 900B
	20 Artist 22E Pro   28BD / 090B
	21 Artist 22 Pro	28BD / 900C
	22 Artist 12 Pro	28BD / 091F
	23 Artist 13.3 Pro	28BD / 092B
 	24 Artist 22 Pro    28BD / 091D
	25 Artist 22 Pro    28BD / 090C
	26 Deco Pro Small	28BD / 0904
	27 Deco Pro Medium	28BD / 0909
	28 Artist 22R Pro	28BD / 091B
	29 Artist 22E Pro	28BD / 091E
	30 Artist 16 Pro	28BD / 900A
	31 Artist 16 Pro	28BD / 090A
	32 Deco 01 V2		28BD / 0905
	33 G960				28BD / 0920
	34 Artist15.6		28BD / 091A
	35 G960S			28BD / 0917
	36 G960S Pro		28BD / 0918
	37 G430S			28BD / 0913
	38 G640				28BD / 0914
	39 CT640W(-E)		28BD / 0921
	40 CT430-E			28BD / 0922
	41 Deco mini7		28BD / 0928	
	42 Deco mini4		28BD / 0929
	43 ID160F			28BD / 092C
	44 Artist 24 Pro	28BD / 092D

spbnick, 2020-06-22:
Thanks a lot, Masaru!

jbigler, 2021-01-01:
I found an updated list of Product IDs from XP-Pen:

PID_000B=Artist 13.3
PID_0071=Star 05
PID_0074=Star 04
PID_0075=Star G540 & G430
PID_0078=Star 06
PID_0096=Deco 03
PID_0081=Star 03
PID_0803=Deco 02
PID_080A=Artist 12
PID_0094=Star G640
PID_0902=Deco 01
PID_0903=Deco 02
PID_0904=Deco Pro Medium
PID_0905=Deco 01 V2
PID_0906=Star G640S
PID_0907=STAR 03 V2
PID_0908=Deco 03
PID_0909=Deco Pro Small
PID_090A=Artist 16Pro
PID_090B=Artist 22E Pro
PID_090D=Artist 15.6 Pro
PID_0913=Star G430S
PID_0914=Star G640
PID_0915=Star G540
PID_0917=Star G960S
PID_0918=Star G960S Plus
PID_091A=Artist 15.6
PID_091B=Artist 22R Pro
PID_091D=Artist 22 Pro
PID_091E=Artist 22E Pro
PID_091F=Artist 12 Pro
PID_0920=Star G960
PID_0921=Deco S&SW
PID_0922=Creator CT430
PID_0923=Deco L&LW
PID_0926=G640S
PID_0928=Deco mini7
PID_0929=Deco mini4
PID_092B=Artist 13.3 Pro
PID_092C=Innovator 16
PID_092D=Artist 24 Pro
PID_092E=Artist Pro 16TP 4K
PID_092F=Artist 22 (2nd)
PID_900A=Artist 16Pro
PID_900B=Artist 22E Pro
PID_900C=Artist 22 Pro
PID_090C=Artist 22 Pro

megamuteki, 2021-02-06:
As Of Feb06 List
New Model Artist22 Second 
PID_000B=Artist 13.3
PID_0071=Star 05
PID_0074=Star 04
PID_0075=Star G540 & G430
PID_0078=Star 06
PID_0096=Deco 03
PID_0081=Star 03
PID_0803=Deco 02
PID_080A=Artist 12
PID_0094=Star G640
PID_0902=Deco 01
PID_0903=Deco 02
PID_0904=Deco Pro Medium
PID_0905=Deco 01 V2
PID_0906=Star G640S
PID_0907=STAR 03 V2
PID_0908=Deco 03
PID_0909=Deco Pro Small
PID_090A=Artist 16Pro
PID_090B=Artist 22E Pro
PID_090D=Artist 15.6 Pro
PID_0913=Star G430S
PID_0914=Star G640
PID_0915=Star G540
PID_0917=Star G960S
PID_0918=Star G960S Plus
PID_091A=Artist 15.6
PID_091B=Artist 22R Pro
PID_091D=Artist 22 Pro
PID_091E=Artist 22E Pro
PID_091F=Artist 12 Pro
PID_0920=Star G960
PID_0921=Deco S&SW
PID_0922=Creator CT430
PID_0923=Deco L&LW
PID_0926=G640S
PID_0928=Deco mini7
PID_0929=Deco mini4
PID_092B=Artist 13.3 Pro
PID_092C=Innovator 16
PID_092D=Artist 24 Pro
PID_092E=Artist Pro 16TP 4K
PID_092F=Artist 22 (2nd)
PID_900A=Artist 16Pro
PID_900B=Artist 22E Pro
PID_900C=Artist 22 Pro
PID_090C=Artist 22 Pro

megamuteki, 2025-02-26:
PID_000B=Artist 13.3
PID_0071=Star 05
PID_0074=Star 04
PID_0075=Star G540 & G430
PID_0078=Star 06
PID_0081=Star 03
PID_0094=Star G640
PID_0096=Deco 03
PID_0201=AC19
PID_0802=VB960
PID_0803=Deco 02
PID_080A=Artist 12
PID_0902=Deco 01
PID_0903=Deco 02
PID_0904=Deco Pro Medium
PID_0905=Deco 01 V2
PID_0906=Star G640S
PID_0907=STAR 03 V2
PID_0908=Deco 03
PID_0909=Deco Pro Small
PID_090A=Artist 16Pro
PID_090B=Artist 22E Pro
PID_090D=Artist 15.6 Pro
PID_0913=Star G430S
PID_0914=Star G640
PID_0915=Star G540
PID_0917=Star G960S
PID_0918=Star G960S Plus
PID_091A=Artist 15.6
PID_091B=Artist 22R Pro
PID_091D=Artist 22 Pro
PID_091E=Artist 22E Pro
PID_091F=Artist 12 Pro
PID_0920=Star G960
PID_0921=Deco S&SW
PID_0922=Creator CT430
PID_0923=Deco L&LW
PID_0926=G640S
PID_0928=Deco mini7
PID_0929=Deco mini4
PID_092A=Artist 13.3 V2
PID_092B=Artist 13.3 Pro
PID_092C=Innovator 16
PID_092D=Artist 24 Pro
PID_092E=Artist Pro 16TP 4K
PID_092F=Artist 22 (2nd)
PID_900A=Artist 16Pro
PID_900B=Artist 22E Pro
PID_900C=Artist 22 Pro
PID_090C=Artist 22 Pro
PID_0930=Deco Fun XS
PID_0931=Deco Fun S
PID_0932=Deco Fun L
PID_0933=Deco Pro SW
PID_0934=Deco Pro MW
PID_093A=Artist 24