DIGImend/digimend-kernel-drivers #427: Huion H950p only left half of tablet working area is usable.

TheEKey, 2020-06-25:
Hello,

I found a strange behavior of my new H950p.

Only the left half (the half that is close to the buttons) is usable and mapped to the screen of resolution 1920 x 1080.
This half of the working area does map to the whole screen and detects the stylus alright. 

The right half is basically not usable as it's not mapped onto the screen.

When I booted to Windows 10 to test it out without proper driver, it had the same issue. After installing the official Huion driver, the tablet worked fine that the whole working area mapped precisely to the full 1920 x 1080.

1) This tablet has capability to work with Android. In Android mode, roughly only left half of the working area is mapped to the android screen. (like a vertical rectangle). Is it somehow a clue that the tablet is not configured properly and it is stuck in android mode even after connecting to Linux?

2) I may have missed out important information when I googled the issue. Most of the articles I could find about "Coordinate Transformation Matrix" assume that the tablet automatically utilize the whole working area and it's only a matter of how this area is mapped to the screen(s).

3) Is it still fixable with the "Coordinate Transformation Matrix"? Or has to do with some changes in the codes and install it from source? 

Please advise. Thanks.

AlanLane, 2020-07-10:
Same problem here with the same tablet.
Only half area of the tablet is working and also with a double speed. So it's impossible to draw a circle since it's faster on X axys. 
Anyway, i've tried to reinstall v9 driver twice and also altering the PID of **H950P** in **id-ids.h**. I think that the tablet is stucked in **android** mode once connected to a Unix System. 

I've tried of course the tablet with windows and android and it works like a charm. 

Would be nice to have a fix to this annoying problem. Thanks.

TheEKey, 2020-07-12:
Hello AlanLan,

After hours of googling and trial and error. Finally the driver is in place and my tablet is functioning.

Besides id-ids.h, there's still one more place to edit after the installation of driver. 
/usr/share/X11/xorg.conf.d/50-digimend.conf. 

In the Huion with Wacom input class
"256c:006e" is written there and this is what's stopping the Wacom driver to pick up the 006d device.

Hope this help you as well.

AlanLane, 2020-07-12:
Hi The EKey,

I've thoughts about editing that file to fix that PID device. And I can't remember if in the end i've done it or not.
Tomorrow i'll try the edit and let you know if it all works normally :)

Thanks for the advice ;)

AlanLane, 2020-07-13:
[image]
I've tried to edit the **50-digiment.conf** file and always half of the tablet is mapped on screen. I've tried to reinstall **v9** driver from source and this is what's inside the early lines. Still half of the tablet is mapped. Maybe you've installed v8 drivers? I'm still in trouble with this tablet on ubuntu.

jlj77, 2020-07-18:
Hi, I recently bought the H950P and thought I'd add my experience:
- lsusb returns Bus 001 Device 009: ID 256c:006d
- I have the full use of my tablet area, working at 1440x900
- evtest reports all buttons correctly

I'm running Linux Mint 19.3 and the v9 driver.

I'm still in the early stages of setting it up, and this is my first tablet, so I may yet run into issues. But, so far, so good.

SaulMoonves, 2020-07-19:
using the .deb I modified /usr/share/X11/xorg.conf.d/50-digimend.conf as needed and the tablet works, but only the left half of the tablet area.

SaulMoonves, 2020-07-21:
compiled and installed from git master and now I can use the whole tablet face, thanks for this great project.

vbelaia, 2020-09-08:
I'm having the same issue - only the left side of my Huion H1060P is working. I've updated the 50-digimend.conf and id-ids.h files to replace "006e" with "006d" and that doesn't fix it. 

Has anyone found a working solution on Ubuntu?

Duduz7, 2021-01-22:
> I'm having the same issue - only the left side of my Huion H1060P is working. I've updated the 50-digimend.conf and id-ids.h files to replace "006e" with "006d" and that doesn't fix it.
> 
> Has anyone found a working solution on Ubuntu?

Same here with my H1060P, the PID is 256c:006d instead of 256c:006e as the project page says, please help beacues even though the drivers are working, the stylus' pressure doesn't work

Xasin, 2021-01-27:
I am having a few problems that I feel are similar to this thread.

First things first:
This tablet works fine on my workhorse-PC, on which I have running Debian 10 (dist-upgraded from Debian 9, the digimend drivers were installed after the dist upgrade).
The tablet has full pressure and can be properly mapped via xsetwacom to fit my screens.

I also have a laptop, which used to run on Debian 9.
It also had the exact same issue reported by other users - only half the tablet was being used to map it onto the screen, making it pretty unusable. BUT it did show up as a listed tablet in the Debian settings tab!!

I then performed a dist-upgrade of the Laptop to Debian 10.
Now, the tablet's full area is being used properly, pressure works etc. etc. 
HOWEVER
It is not recognized by xsetwacom at all, meaning that on my laptop, I can not properly map it onto the screen the way I want to.

I already tried reinstalling (with purge) the digiment dkms drivers  (latest version from the release page, 10), which did not fix the issue of the tablet not being mappable via xsetwacom

Also, two mouse-cursors show up whenever I use the tablet - one for the stylus, and one for the mouse. Both can be used independently.

The tablet also shows up as 256c:006d under `lsusb` and other debug messages.

digimend-debug reports (not the full report, just what I feel is relevant, though I'd be happy to upload the full digimend-debug dmesg report if needed):

```
[  749.630425] usb 2-1.2: New USB device found, idVendor=256c, idProduct=006d, bcdDevice= 0.00
[  749.630433] usb 2-1.2: New USB device strings: Mfr=5, Product=6, SerialNumber=0
<parameter debugging here...>
[  749.642057] input: HID 256c:006d as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.0/0003:256C:006D.000C/input/input37
[  749.642559] input: HID 256c:006d Pad as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.0/0003:256C:006D.000C/input/input38
[  749.642909] input: HID 256c:006d Touch Strip as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.0/0003:256C:006D.000C/input/input39
[  749.643213] input: HID 256c:006d Dial as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.0/0003:256C:006D.000C/input/input40
[  749.646299] uclogic 0003:256C:006D.000C: input,hiddev0,hidraw1: USB HID v1.11 Keypad [HID 256c:006d] on usb-0000:00:1d.0-1.2/input0
<more parameter debug messages...>
[  749.651914] uclogic 0003:256C:006D.000D: No inputs registered, leaving
[  749.653892] uclogic 0003:256C:006D.000D: hidraw5: USB HID v1.11 Device [HID 256c:006d] on usb-0000:00:1d.0-1.2/input1
```

It almost seems like something is making the drivers register the tablet, but as generic HID.

jeffwilliams, 2022-02-09:
I had the same problem with the Huion 1060P with the v10 drivers: only half the area of the tablet is usable. It's recognized as HID 256c:006d and the v10 drivers properly use the wacom driver for it. 

What seemed to fix it are the steps in this article:

https://support.huion.com/en/support/solutions/articles/44001906540-how-to-manually-set-your-tablet-to-phone-mode

Holding down buttons K1 and K5 for three seconds to change to phone mode actually corrected the problem; the whole tablet area was usable, and it was still usable even after I mapped the area to only one of my monitors.