DIGImend/digimend-kernel-drivers #429: Cannot use CONFIG_STACK_VALIDATION during kernel update 5.4.0-39

Narann, 2020-06-26:
Hi,

First: Thanks for this driver and dedication to it. It work amazingly well.

I have updated my linux kernel and have an error:

```
Setting up linux-image-5.4.0-39-generic (5.4.0-39.43~18.04.1) ...
I: /vmlinuz.old is now a symlink to boot/vmlinuz-5.4.0-37-generic
I: /initrd.img.old is now a symlink to boot/initrd.img-5.4.0-37-generic
I: /vmlinuz is now a symlink to boot/vmlinuz-5.4.0-39-generic
I: /initrd.img is now a symlink to boot/initrd.img-5.4.0-39-generic
Setting up linux-modules-extra-5.4.0-39-generic (5.4.0-39.43~18.04.1) ...
Setting up linux-headers-5.4.0-39-generic (5.4.0-39.43~18.04.1) ...
/etc/kernel/header_postinst.d/dkms:
 * dkms: running auto installation service for kernel 5.4.0-39-generic

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.4.0-39-generic KVERSION=5.4.0-39-generic....(bad exit status: 2)
ERROR (dkms apport): kernel package linux-headers-5.4.0-39-generic is not supported
Error! Bad return status for module build on kernel: 5.4.0-39-generic (x86_64)
Consult /var/lib/dkms/digimend/9/build/make.log for more information.
   ...done.
```

And the make.log is:

```
DKMS make.log for digimend-9 for kernel 5.4.0-39-generic (x86_64)
Fri Jun 26 08:34:43 CEST 2020
make -C /lib/modules/5.4.0-39-generic/build SUBDIRS=/var/lib/dkms/digimend/9/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-39-generic'
warning: Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel
make[2]: *** No rule to make target 'arch/x86/tools/relocs_32.c', needed by 'arch/x86/tools/relocs_32.o'.  Stop.
arch/x86/Makefile:232: recipe for target 'archscripts' failed
make[1]: *** [archscripts] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-39-generic'
Makefile:22: recipe for target 'modules' failed
make: *** [modules] Error 2
```

It suggest to install libelf-dev, libelf-devel or elfutils-libelf-devel. I installed `libelf-dev` but as the process appear during kernel update, I can't reproduce.

As I don't use my tablet on this PC for now, I can't check if everything work.

This was just to share information, feel free to close this ticket.

Keep the good work and have a good day! Take care!

spbnick, 2020-06-28:
Thank you for the report @Narann. This looks like a package dependency issue. The `dkms` package should depend on all packages required to build kernel modules, and the drivers depend on `dkms` package. Please raise this issue with distribution support.

Narann, 2020-06-28:
Thanks, @spbnick, thanks, I will do that. Keep the good work!

closed 2020-06-28 by Narann