DIGImend/digimend-kernel-drivers #431: digimend-dkms_9_all.deb does not install on new Linux Mint 20 Ulyana

chlodevig, 2020-07-02:
I tried to install the digimend package with GDebi, which gives the following error:

[...]
Error! Bad return status for module build on kernel: 5.4.0-40-generic (x86_64)
Consult /var/lib/dkms/digimend/9/build/make.log for more information.
dpkg: error processing package digimend-dkms (--install):
 installed digimend-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 digimend-dkms

`$ cat /var/lib/dkms/digimend/9/build/make.log`
DKMS make.log for digimend-9 for kernel 5.4.0-40-generic (x86_64)
Do 2. Jul 16:38:17 CEST 2020
make -C /lib/modules/5.4.0-40-generic/build SUBDIRS=/var/lib/dkms/digimend/9/build modules
make[1]: Verzeichnis „/usr/src/linux-headers-5.4.0-40-generic“ wird betreten
  UPD     include/generated/utsrelease.h
make[2]: *** Keine Regel vorhanden, um das Ziel „arch/x86/tools/relocs_32.c“, 
  benötigt von „arch/x86/tools/relocs_32.o“, zu erstellen.  Schluss.
make[1]: *** [arch/x86/Makefile:232: archscripts] Fehler 2
make[1]: *** Auf noch nicht beendete Prozesse wird gewartet …
make[1]: Verzeichnis „/usr/src/linux-headers-5.4.0-40-generic“ wird verlassen
make: *** [Makefile:22: modules] Fehler 2

My tablet is the HUION 610 Pro V2, which works with the built-in drivers. However the buttons don't, so I wanted to give a try with the digimend-driver. If you need any further information, please let me know. TIA.

clay53, 2020-07-06:
I can't read this but I'm going to assume this is related to #387 right?

chlodevig, 2020-07-06:
It seems the issue has been reported. My bad, I didn't find it, when I first looked. Thanks for pointing me in the direction.