DIGImend/digimend-kernel-drivers #433: Gaomon M10K Pro

Miratu, 2020-07-06:
Hardware: Gaomon M10K Pro
Distro: Ubuntu 20.04 LTS 
Kernel: 5.4.0-40-lowlatency(x86_64)
Issue: Only the pen and its two buttons work. The configuration sees the presence of a tablet (when it is not connected it does not appear in the configuration) but reports that it does not detect any tablet or pen (although the pen works correctly).

Thanks for your help

Gaomon M10K Pro - https://www.gaomon.net/Pen_Tablet/M10KPRO.html

[touch_ring(directions).txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878286/touch_ring.directions.txt)
[touch_ring(clockwise).txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878287/touch_ring.clockwise.txt)
[touch_ring(center_key).txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878288/touch_ring.center_key.txt)
[touch_ring(anti-clockwise).txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878289/touch_ring.anti-clockwise.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878291/probe.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878292/pen_tilt.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878293/pen_pressure.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878295/pen_coords.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878296/pen_buttons.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878297/frame_buttons.txt)
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878298/descriptors.txt)

Kangaroux, 2020-07-30:
Would be great to have support for this.

mverouden, 2020-10-06:
@Miratu already uploaded a load of files, what is can the users provide to help advance the support for the Gaomon M10K Pro?

I am using it on macOS and its a joy to work on and hope I will be able to use on my linux systems as well.

Please let us know how we can help to get this tablet supported.

fidelbrea, 2020-11-17:
> Hardware: Gaomon M10K Pro
> lsusb: 256c:006d (like Huion HS610)
> Vendor: Gaomon
> Product: M10K Pro
> Status: Unsupported
> Specs: 10″×6.25″, 5080 LPI, 266 RPS , 8192 PL, tilt-sensitive, Frame controls 10 buttons, touch ring with a center button
> 
> Distro: Ubuntu 20.04 LTS
> Kernel: 5.4.0-40-lowlatency(x86_64) updated to 5.7.7-050707-generic x86_64
> Issue: Only the pen and its two buttons work. The configuration sees the presence of a tablet (when it is not connected it does not appear in the configuration) but reports that it does not detect any tablet or pen (although the pen works correctly).
> 
> Thanks for your help
> 
> Gaomon M10K Pro - https://www.gaomon.net/Pen_Tablet/M10KPRO.html
> 
> [touch_ring(directions).txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878286/touch_ring.directions.txt)
> [touch_ring(clockwise).txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878287/touch_ring.clockwise.txt)
> [touch_ring(center_key).txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878288/touch_ring.center_key.txt)
> [touch_ring(anti-clockwise).txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878289/touch_ring.anti-clockwise.txt)
> [probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878291/probe.txt)
> [pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878292/pen_tilt.txt)
> [pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878293/pen_pressure.txt)
> [pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878295/pen_coords.txt)
> [pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878296/pen_buttons.txt)
> [frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878297/frame_buttons.txt)
> [descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4878298/descriptors.txt)

Any new?

mverouden, 2020-11-29:
@Miratu @fidelbrea @Kangaroux  Check out https://github.com/DIGImend/digimend-kernel-drivers/issues/275

Support for the Huion HS610 is well on its way and part of it has been implemented in v10 already. Support for the touch ring has been set for v11 of the Digimend driver (see https://github.com/DIGImend/digimend-kernel-drivers/milestone/4).

Hope all of you find this informative,

Kind regards,

Miratu, 2020-12-01:
These are the steps that I have followed to use my tablet with Blender. The touch ring with a center button change value, so I haven't caught the handle on them yet. I still haven't been able to test the pen pressure.
I've also tried Inkscape and GIMP, but not enough.

sudo gedit /usr/share/X11/xorg.conf.d/70-wacom.conf  (in my case it appears as "70". In other cases like "50")

add the following at the end:

-----
#Gaomon m10k Pro
Section "InputClass"
        Identifier "GAOMON Gaomon Tablet_M10KPRO"
        MatchUSBID "256c:006d"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
EndSection

-----

---For the right hand---

sudo gedit /usr/local/bin/blenderRight

Since it did not exist, it will appear empty. Add the following:

-----
#!/bin/bash

echo assigning commands to graphics tablet buttons (pen in right hand)

echo key 3 has no function assigned at this time

xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 1 "key tab"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 2 "key +shift +d -d -shift"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 3 "key"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 8 "key +shift +ctrl +z -z -ctrl -shift"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 9 "key +ctrl +z -z -ctrl"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 10 "key g"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 11 "key r"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 12 "key s"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 13 "key shift"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 14 "key control"

cd /usr/bin

./blender

-----

---For lefties like me---

sudo gedit /usr/local/bin/blenderLeft

Since it did not exist, it will appear empty. Add the following:

-----
#!/bin/bash

echo don't recalibrate pen pressure when starting
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pen stylus' PressureRecalibration off

echo change orientation to use left hand
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pen stylus' Rotate half

echo assigning commands to graphics tablet buttons (pen in left hand)

echo key 12 has no function assigned at this time

xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 1 "key control"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 2 "key shift"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 3 "key s"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 8 "key r"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 9 "key g"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 10 "key +ctrl +z -z -ctrl"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 11 "key +shift +ctrl +z -z -ctrl -shift"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 12 "key"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 13 "key +shift +d -d -shift"
xsetwacom --set 'GAOMON Gaomon Tablet_M10KPRO Pad pad' Button 14 "key tab"

cd /usr/bin

./blender

-----

I hope you find it useful!

rodrigooeiras, 2021-06-04:
Hi, about the m10k pro, the buttons works now?

zakaria-chahboun, 2021-11-22:
I switched to OTD open table drivers 
It's a really great cross platform project with a beautiful GUI

fidelbrea, 2021-11-22:
That looks really interesting since it seems to make everything to work, but it is not a clean install proccess. I have tried it without success.

What I tried mainly is this script:
source: [https://opentabletdriver.net/Wiki/Install/Linux](https://opentabletdriver.net/Wiki/Install/Linux)

`# Add the Microsoft Packages repository, use the URL related to your distribution
sudo dpkg -i packages-microsoft-prod.deb

# Install the .NET runtime
sudo apt update
sudo apt install -y apt-transport-https
sudo apt update
sudo apt install -y dotnet-sdk-5.0

# This will install the package, assuming you are in the correct directory
sudo apt install ./OpenTabletDriver.deb

# Reload the systemd user unit daemon
systemctl --user daemon-reload

# Enable and start the user service
systemctl --user enable opentabletdriver --now`

I am able to launch opentabledriver as aplication and I open it, but it does not detect my Gaemon M10K Pro. Then I downloaded your json file and tried to load it as configuration... and it made the aplication to crash.

So, my feeling whith this project is a _dirty_ experience. I mean: I had to add the Microsoft packages repository, the https apt (that's OK), the dotNET SDK in order to install (finally) the opentabletdriver.

I hope sincerely to have been wrong and to read a comment saying that I have done it wrong from the beginning and that the steps to follow are ....

Any way @zakaria-chahboun , as I began this comment, that still looks interesting.

Regards!

zakaria-chahboun, 2021-11-23:
@fidelbrea You have to black list thw DIGImend drivers from the kernel,  and follow my steps in the link!
It's just works for me like a charm 
You don't have to import my JSON configuration, The tablet already supported by them.

zakaria-chahboun, 2021-12-10:
@fidelbrea  use this https://github.com/OpenTabletDriver/OpenTabletDriver/issues/1707