DIGImend/digimend-kernel-drivers #434: Huion Kamvas 20 (GT-191v3) dual monitor setting/calibration

fugata, 2020-07-09:
tl;dr:  The tip of the pen of GT-191v3 does not match the motion of the cursor on the screen under a dual-screen setup.

Details:   I hooked up GT-191v3 to my laptop running linux mint 20 xfce.

1) The GT-191v3 shows up correctly as a dual-monitor:   Laptop screen on the left (primary display) and GT-191v3 to the right.   The mouse pad moves the cursor correctly, and I can e.g. move xterm back-and-forth between the laptop screen and the GT-191v3. 

2) The resolution of both the laptop and GT-191v3 are both 1920x1080.

3) The pen moves the cursor; I didn't have to install any driver.  However, GT-191's pen movement is wrong:  The pen tip's movement is restricted to GT-191v3 but the corresponding cursor moves all the way from the far left edge of the laptop to the far right side of GT-191v3 --- effectively 3820x1080.  Specifically:

    a) If I place the pen tip at the upper left hand corner of GT-191v3, the cursor moves to the upper left hand corner of the laptop (which remember is the left screen).

    b) As the pen tip moves to the right along the top edge of GT191-v3, the cursors moves to the right and speeds up as it goes along.

    c) The cursor's speed reaches the highest point when the cursor reaches the upper right hand corner of the laptop's screen, at which point the pen tip is at the midpoint of the top edge of GT-191v3.

   d) As the pen continues to move to the right, the cursor continues to do so and gradually slows down.

   e) Finally, as the pen tip reaches the upper right hand corner of GT-191v, so does the cursor.

4) If I change the xfce4-display setting so that GT-191v3 overlaps with the laptop (or if I simply turn off the laptop screen), the pen tip moves in sync with the cursor -- no delay/gap/racing.

5) I tried this on two different brands of laptops and got the same result.  And I also tried xubuntu 20.04 and it makes no difference.

I really need to operate GT-191v3 as a dual screen.  Your help and advice is most appreciated!

attilaM68K, 2020-07-24:
You can map the tablet's input to its screen using this command:
`xinput map-to-output "[TABLET-NAME]" [VIDEO-OUTPUT] `
To find the tablet's name, just use this command:
`xinput --list`
The device ending in some kind of "...Pen" is the name you are looking for! In my case it was "Tablet Monitor Pen Pen (0)"
For the video output, you can use this command:
`xrandr --listmonitors`
It should look something like `HDMI-A-0`.

Alternatively, if you have your tablet set to the wacom driver, there are many GUI tools for mapping the tablet's input to the desired screen.

fugata, 2020-07-27:
> xinput map-to-output

fugata, 2020-07-27:
> You can map the tablet's input to its screen using this command:
> `xinput map-to-output "[TABLET-NAME]" [VIDEO-OUTPUT] `
> To find the tablet's name, just use this command:
> `xinput --list`
> The device [...]

That works.  Thanks!

closed 2020-07-27 by fugata