DIGImend/digimend-kernel-drivers #436: Tip: Huion HS611 no mouse cursor with Ubuntu 16.04 LTS

AstroLad47, 2020-07-10:
This is the solution I found, not an issue as such, in case someone comes here looking for help. The problem was the mouse cursor disappeared when the pen got within about 1 cm of the tablet surface. The cursor was moving, you just couldn't see it. I could also select an item or region with the pen, but pretty useless blind. Also, xsetwacom --list returned nothing. I had installed the driver from digimend-dkms_9_all.deb. That broke X11 because of the bad 50-digimend.conf file put in /usr/share/X11/xorg.conf.d. After some panic and frantic searching the web for a solution, I ran across a mention of the broken file. Fixing it restored X11, but the cursor still was not visible.

I then spent hours trying various solutions others had suggested on the web. Nothing worked. Okay, last resort: build the driver from source. Bingo! The master branch as of today (10 July 2020) fixed everything...so far. Mouse cursor is now visible, and xsetwacom sees the device. I have not yet tried using xsetwacom to change settings. 

Yeah! I don't have to boot Windows to use the new toy.

DebsankarBanerjee, 2020-10-18:
Built the driver from latest repo by:
make
sudo make install
The pen works fine, followed the steps to configure /etc/X11 and can see the tablet and pen in xsetwacom BUT the cursor is still invisible in all the drawing pages. device - h610 pro in ubuntu 16.04. Anyone fixed this issue?

DebsankarBanerjee, 2020-10-18:
It worked finally. The answers in web are not very well organized or explained. I am writing down my experience here to help in this particular problem.
Device: Huion h610 pro V2
OS: Ubuntu 16.04 (Kernel: 4.15.0-120-generic)
Problem: **Disappearing or invisible cursor while drawing in tablet.**
------------------------------------------- steps I followed --------------------------------------------
1. cloned the main branch on this day (17 Oct 2020) and manually installed driver using 
    make 
    sudo make install
2. Setup the Xorg in /etc/X11
    This is where the error happened (in my opinion).
    I used this [solution](https://askubuntu.com/questions/500141/huion-h610-tablet). This solution is very nice and well explained. 
    I could see my tablet and pen in `xsetwacom --list` but the cursor would not appear while using the pen. This becomes difficult
    from here as I could not find any well explained fix and I am still not sure what is actually causing it. But at least it has to do with 
    how one creates the xorg file "52-tablet.conf". Finally I used the slightly different code to write the file as described in the 
    master repo [here](https://github.com/DIGImend/digimend-kernel-drivers). **Notice the difference from the previous solution in 
    .conf file - the second case has the device id specified in the code** ` MatchUSBID "<VID>:<PID>"` 
This made all the difference! Now I can see the cursor and use the stylus properly. Hope this will help someone with this problem. All the thanks to [Nikolai Kondrashov](https://github.com/spbnick).

coreleon, 2021-04-09:
I did what every one said BUT still the cursor disappears. I tried to open a session with Openbox (instead of Unity) and the cursor WAS OK!
It seems that it has something to do with Unity/Compiz.
Is there any idea what is the cause of this behavior?