DIGImend/digimend-kernel-drivers #438: digimend-dkms problems in Ubuntu 20.04

misterybodon, 2020-07-18:
Hello guys, 

So I bought a tablet days ago, and followed all the instructions you give in the readme file; the drawing tablet is recognized by the computer -so thanks for that- but when upgrading my system I get an error (not a warning). 

I've pasted the output straight into https://paste.ubuntu.com/p/383xfzrfk6/ 

Any ideas on this?

I attach the file in case you prefer that to paste.ubuntu.

[file1.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4942834/file1.txt)

docjmwright, 2020-07-19:
Hi misterybodon,
I am no expert at digimend, but it looks like you don't have the programs flex and bison installed.
Try:   apt install flex bison
and try again.

misterybodon, 2020-07-19:
@docjmwright hey, thanks for that, 
I've noticed the same after reading the log file, installed everything but the problem persists. 
I've read many posts in stackoverflow and readdit and there is no clear solution, Some of them require to change the linux-header-generic but I don't want to change something I barely understand.
There is also another issue raised by a user that I should have checked before but it was too late.

will-code-for-pizza, 2020-08-30:
[https://github.com/DIGImend/digimend-kernel-drivers/issues/387](url)

twiggy, 2020-09-13:
I've gotten this to work in Mint 20 which is effectively Ubuntu 20, but I had to compile from source. A new .deb release would be nice. If I have time I might build one.

protospheric, 2021-02-01:
> [https://github.com/DIGImend/digimend-kernel-drivers/issues/387](url)

nothing there

will-code-for-pizza, 2021-02-27:
Sorry, somebody has deleted my solution. Do not know, why... :-(