DIGImend/digimend-kernel-drivers #444: Cannot change settings of Gaomon S620 tablet

Dreamurrrr, 2020-08-09:
I have Linux Mint and I installed the drivers. When I reboot the PC, the system still does not detect the tablet:
[image]

Also, when I execute `lsusb`, the model of the tablet does not pop up (its `Bus 001 Device 012: ID 256c:006d`):
```Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 1038:12aa SteelSeries ApS 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 08ff:2691 AuthenTec, Inc. 
Bus 001 Device 010: ID 04d8:00df Microchip Technology, Inc. 
Bus 001 Device 008: ID 8087:0a2b Intel Corp. 
Bus 001 Device 006: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 005: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
Bus 001 Device 004: ID 1d57:0005 Xenta Wireless Receiver (Keyboard and Mouse)
Bus 001 Device 012: ID 256c:006d  
Bus 001 Device 003: ID 1038:1614 SteelSeries ApS 
Bus 001 Device 009: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 002: ID 1038:172b SteelSeries ApS 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Dreamurrrr, 2020-08-09:
I compiled from source and I can now change the settings of the tablet! My recommendation to fix this issue is to provide a new release.
[image]

GAZ082, 2020-08-29:
Hey. I compiled from source with DKMS and still cannot see the tablet. Did you do anything else besides installing? I tried to enable the wacom drivers with the config file and no lucl.

Dreamurrrr, 2020-08-29:
> Hey. I compiled from source with DKMS and still cannot see the tablet. Did you do anything else besides installing? I tried to enable the wacom drivers with the config file and no luck.
> 
> Mine is working, the digitizer and the buttons on the pen are ok but the frame buttons are not.

Make sure you did `make install` and reboot your PC. I made a fourm post on how to setup everything for the tablet here if you are still having issues: https://osu.ppy.sh/community/forums/topics/1122479

Dreamurrrr, 2020-08-29:
Read step 2 fully for troubleshooting @GAZ082

GAZ082, 2020-09-10:
Hey. As I wanted to use Wayland, used libwacom. The maintainer of that lib released an update that adds support to S620 and got it running in no time. Thanks @Dreamurrrr for your support tho!

Mierscheid, 2021-01-17:
Hi GAZ082,

I have the same tablet and the same problem. The output of xsetwacom is empty, but the tablet appears under xinput --list: 
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Paten    USB Gaming Mouse                    	id=11	[slave  pointer  (2)]
⎜   ↳ Contour Design ShuttleXpress            	id=12	[slave  pointer  (2)]
⎜   ↳ HID 0566:3013 Consumer Control          	id=15	[slave  pointer  (2)]
⎜   ↳ GAOMON Gaomon Tablet Pen Pen (0)        	id=22	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ GAOMON Gaomon Tablet Pen                	id=10	[slave  keyboard (3)]
    ↳ Contour Design ShuttleXpress            	id=20	[slave  keyboard (3)]
    ↳ HID 0566:3013 System Control            	id=16	[slave  keyboard (3)]
    ↳ HID 0566:3013                           	id=13	[slave  keyboard (3)]
    ↳ HID 0566:3013 Consumer Control          	id=21	[slave  keyboard (3)]
    ↳ HID 0566:3013                           	id=14	[slave  keyboard (3)]
    
    Therefore, I can not configure the Tablet. This is stupid, because I have a second monitor. :-(
    I have installed libwacom and get the following output:
    
libwacom-list-local-devices
# Device node: /dev/input/event258
[Device]
Name=GAOMON S620
ModelName=
DeviceMatch=usb:256c:006d:GAOMON Gaomon Tablet Pen;usb:256c:006d:GAOMON Gaomon Tablet Pad;
Class=Bamboo
Width=6
Height=4
IntegratedIn=
Layout=gaomon-s620.svg
Styli=0xffffd;

[Features]
Reversible=true
Stylus=true
Ring=false
Ring2=false
Touch=false
TouchSwitch=false
# StatusLEDs=
NumStrips=0
Buttons=4

[Buttons]
Left=A;B;C;D;
# Right=
# Top=
# Bottom=
# Touchstrip=
# Touchstrip2=
# OLEDs=
# Ring=
# Ring2=
EvdevCodes=0x100;0x101;0x102;0x103;
RingNumModes=0
Ring2NumModes=0
StripsNumModes=0

---------------------------------------------------------------
# Device node: /dev/input/event257
[Device]
Name=GAOMON S620
ModelName=
DeviceMatch=usb:256c:006d:GAOMON Gaomon Tablet Pen;usb:256c:006d:GAOMON Gaomon Tablet Pad;
Class=Bamboo
Width=6
Height=4
IntegratedIn=
Layout=gaomon-s620.svg
Styli=0xffffd;

[Features]
Reversible=true
Stylus=true
Ring=false
Ring2=false
Touch=false
TouchSwitch=false
# StatusLEDs=
NumStrips=0
Buttons=4

[Buttons]
Left=A;B;C;D;
# Right=
# Top=
# Bottom=
# Touchstrip=
# Touchstrip2=
# OLEDs=
# Ring=
# Ring2=
EvdevCodes=0x100;0x101;0x102;0x103;
RingNumModes=0
Ring2NumModes=0
StripsNumModes=0

Yes, I know this is getting OT, but I'm already totally desperate because no clue has helped anything.

TIA
Cheers
Mierscheid

GAZ082, 2021-01-17:
Hey. As I wrote, I ended up using libwacom an works. Make sure you use the latest version.

Mierscheid, 2021-01-17:
Hmm, how did you configure it? Via system settings or via Bash?

GAZ082, 2021-01-17:
In Gnome system settigs, "wacom tablet". (you wont see anything in this screenshot because currently I do not have plugged the tablet).
[image]

nummer9, 2021-03-01:
Hey, I recently installed the drivers via .deb, after a reboot I could see the tablet in the graphics tablet menu. I could set another action to the keys on the pen, but not to the keys on the tablet. Anyone know how to customize tablet keys?
Thanks in advance

bigbigmdm, 2021-10-01:
Please see  my post in GitHub: [https://github.com/bigbigmdm/GAOMON-S620-in-LINUX](url)