DIGImend/digimend-kernel-drivers #448: Can't install on fedora 32

coryburt, 2020-08-16:
dkms add .

Creating symlink /var/lib/dkms/digimend/9/source ->
                 /usr/src/digimend-9

DKMS: add completed.
dkms build digimend/9

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.7.14-200.fc32.x86_64 KVERSION=5.7.14-200.fc32.x86_64...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.7.14-200.fc32.x86_64 (x86_64)
Consult /var/lib/dkms/digimend/9/build/make.log for more information.
make: *** [Makefile:74: dkms_modules_install] Error 10

Is this simply a no-go for Fedora 32?

cbm755, 2020-08-16:
What does it say in the `make.log` file it mentions?

I'm new to this project but I just built it on Fedora 32, and it seems to work for me:
```
...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.7.14-200.fc32.x86_64 KVERSION=5.7.14-200.fc32.x86_64...
cleaning build area...

DKMS: build completed.
dkms install digimend/10

...
```

coryburt, 2020-08-16:
Thanks for looking at this.  The snippet I included IS the make.log.  The
make.log says to "check the make.log" for more info.  Go figure.

Clearly, I am doing something wrong because this works for most, and there
is little indication that anyone (who bothers asking online) is having the
same troubles.  Common debugging sense says that I have something installed
that is messing up the dkms environment or some other devel condition, but,
I'm just guessing now.

cbm755, 2020-08-16:
I guess some dependency is missing.

Do you have any other DKMS modules?  I previously installed `v4l2loopback-dkms` [from here](https://copr.fedorainfracloud.org/coprs/sentry/v4l2loopback/).  Maybe try that, see if it installs, then come back and try here again?

coryburt, 2020-08-17:
Well, I may, indeed, try it.  However, it seems I may have taken the
Internet too literally in the need to even install this driver.  My nice
new Huion tablet works fairly well as a plug-and-go tablet on my Fedora 32
laptop.  I assumed the worst and jumped right into attempting to install
the digimend driver before even trying the tablet itself. I don't know if
the buttons on the tablet are supported OOTB by apps under Fedora, but the
pressure-sensitive pen varies the line width, as expected, in my Snapd
install of Krita.  It might not be everything, but, at least it's not
RPM-hell.  In the famous words of Gilda Radner, "Never mind!"  Thanks for
taking a look, anyway!

cwells, 2020-08-25:
The release didn't work for me, but I was able to build from master on F32.