DIGImend/digimend-kernel-drivers #453: make fails on make[1]

Jaz-Jackson-1225, 2020-09-02:
I found this driver while looking for something that would support my Huion H420 tablet, since libinput shows /dev/input/event15 as a device with pressure and location (exactly what I need) to draw on a project. 
downloaded the source .tar.gz file in the releases section, unzipped it in a directory I put drivers inside of, typed in `make`, and got this error:
```
[jaz@Jaz-Jackson ~/prog/digimend-kernel-drivers-9]$ make
make -C /lib/modules/5.8.3-arch1-1/build SUBDIRS=/home/jaz/prog/digimend-kernel-drivers-9 modules
make[1]: *** /lib/modules/5.8.3-arch1-1/build: No such file or directory.  Stop.
make: *** [Makefile:22: modules] Error 2
```
what am I missing that is preventing the installation?
I am using Arch linux kernel version 5.8.3-arch1-1
ran pacman -Syyu just to make sure I was behind on something important for the install and re-ran the make in the source directory, but it gave the same output. running sudo make did the same exact output.

any help would be greatly appreciated

nic3-14159, 2020-09-04:
Based on the error message, it seems that you are missing the "build" directory in the kernel module directory. The build directory contains header files needed to build kernel modules such as digimend. On Arch, it is provided by the _linux-headers_ package. Try installing that package and then try running make again.

nic3-14159, 2020-09-05:
It seems like digimend version 9 from the releases no longer builds properly on Arch. I tried that release on my machine, and I got the same error. Looking at changes to the Makefile, it seems that the line
`$(MAKE) -C $(KDIR) SUBDIRS=$(PWD) $@`
in version 9 has been changed to 
`$(MAKE) -C $(KDIR) M=$(PWD) $@`
in the git version. Version 9 seems to build and work with my tablet after changing that line.

Also, are you sure your tablet doesn't work out of the box without having to build digimend? According to the [wiki](https://digimend.github.io/tablets/Huion_H420/), the H420 has been supported natively by the kernel since 3.17.

Jaz-Jackson-1225, 2020-09-05:
I thought the H420 would work as well!
it worked flawlessly on my Kubuntu install, and I managed to use it on everything from Blender drawing to Krita painting. I expected it to be a plug-and-play scenario with my Arch install but all I got was a load of errors from libinput when I tried to debug it. it said there were missing things that make the tablet considered a tablet, so I went searching for a solution and found DIGImend. nothing will ever be easy for me until the day I die. that is a proven fact. I will paste my libinput report in a sec

Jaz-Jackson-1225, 2020-09-05:
```[jaz@Jaz-Jackson ~]$ sudo !!
sudo libinput debug-events
[sudo] password for jaz:
-event2   DEVICE_ADDED     Power Button                      seat0 default group1  cap:k
-event5   DEVICE_ADDED     Video Bus                         seat0 default group2  cap:k
-event0   DEVICE_ADDED     Lid Switch                        seat0 default group3  cap:S
-event1   DEVICE_ADDED     Sleep Button                      seat0 default group4  cap:k
-event7   DEVICE_ADDED     USB2.0 0.3M UVC WebCam: USB2.0    seat0 default group5  cap:k
-event11  DEVICE_ADDED     Sylvania SP481 Consumer Control   seat0 default group6  cap:kp scroll-nat
-event4   DEVICE_ADDED     Asus Laptop extra buttons         seat0 default group7  cap:k
-event3   DEVICE_ADDED     AT Translated Set 2 keyboard      seat0 default group8  cap:k
-event6   DEVICE_ADDED     ETPS/2 Elantech Touchpad          seat0 default group9  cap:p  size 41x23mm tap(dl off) left scroll-nat scroll-2fg-edge dwt-on
-event12  DEVICE_ADDED     Sylvania SP481 (AVRCP)            seat0 default group6  cap:k
Failed to open /dev/input/event19 (No such file or directory)
-event18  DEVICE_ADDED     HUION H420 System Control         seat0 default group10 cap:k
-event17  DEVICE_ADDED     HUION H420 Consumer Control       seat0 default group10 cap:kp scroll-nat
event16 - HUION H420 Keyboard: libinput bug: missing tablet capabilities: xy pen btn-stylus resolution. Ignoring this device.
-event14  DEVICE_ADDED     HUION H420 Pad                    seat0 default group10 cap:P left buttons:12 strips:0 rings:0 mode groups:1
-event13  DEVICE_ADDED     HUION H420 Pen                    seat0 default group10 cap:T  size 106x59mm left calib
event15 - HUION H420 Mouse: libinput bug: missing tablet capabilities: xy pen btn-stylus resolution. Ignoring this device.
```

Jaz-Jackson-1225, 2020-09-05:
what really confuses me is the "bug" where it ignores the device with this output in every lib-input instance that scans the devices:
```
event15 - HUION H420 Mouse: libinput bug: missing tablet capabilities: xy pen btn-stylus resolution. Ignoring this device.
event16 - HUION H420 Keyboard: libinput bug: missing tablet capabilities: xy pen btn-stylus resolution. Ignoring this device.
```

nic3-14159, 2020-09-05:
Have you tried the [digimend AUR package](https://aur.archlinux.org/packages/digimend-drivers-git-dkms/)? Building digimend with `make` might fix your issues, but you'd have to rerun make for every kernel update. DKMS rebuilds the driver for every update, but I think the dkms make target (`sudo make dkms_install`) is broken on digimend version 9 on distros with newer dkms/kernel versions, like Arch. You could clone the digimend git repo and build from that to get the fixed makefile, or use the AUR package, which basically does the same thing.

Jaz-Jackson-1225, 2020-09-05:
I have yet tried the AUR package. I will try it immediately

Jaz-Jackson-1225, 2020-09-05:
this looks promising. the aur didn't give an error code and exited the installer.....

do I have to invoke anything or is it just a drop-in and "it just works" kind of install?

nic3-14159, 2020-09-05:
I believe it should just work. You might also want to install the xf86-input-wacom package, which offers a few more features over the default xorg-driver. Digimend is configured to allow tablets to work with the wacom package if is installed

Jaz-Jackson-1225, 2020-09-05:
OMG IT WORKS!!!!!!!
THANK YOU!
Immediately after typing in yay -S xf86-input-wacom, it installed that package and the tablet flashed twice. i stuck the pen to it and it WORKED! it made the cursor move!

Jaz-Jackson-1225, 2020-09-05:
thank you so so much @nic3-14159 you are awesome!

closed 2020-09-05 by Jaz-Jackson-1225

nic3-14159, 2020-09-05:
Glad I could help!

Jaz-Jackson-1225, 2020-09-05:
[image]
proof that your guidance worked beautifully