DIGImend/digimend-kernel-drivers #454: Help Installing on Linux - Ubuntu 18.04.4 LTS

RylanSchaeffer, 2020-09-04:
Hi! I'm struggling to get my HUION H640P tablet to work on Ubuntu 18.04.4 LTS. I first tried download and installing the .deb file (from here https://github.com/DIGImend/digimend-kernel-drivers/releases), but after rebooting, the tablet doesn't seem to be recognized. I then uninstalled the deb file with `sudo apt remove digimend-dkms`, downloaded `digimend-kernel-drivers-9.tar.gz`, extracted all, changed into the extracted directory and ran `sudo make dkms_install` which appears to have succeeded:

```
$ sudo make dkms_install
dkms add .

Creating symlink /var/lib/dkms/digimend/9/source ->
                 /usr/src/digimend-9

DKMS: add completed.
dkms build digimend/9

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=4.15.0-112-generic KVERSION=4.15.0-112-generic...
Signing module:
 - /var/lib/dkms/digimend/9/4.15.0-112-generic/x86_64/module/hid-polostar.ko
 - /var/lib/dkms/digimend/9/4.15.0-112-generic/x86_64/module/hid-uclogic.ko
 - /var/lib/dkms/digimend/9/4.15.0-112-generic/x86_64/module/hid-viewsonic.ko
 - /var/lib/dkms/digimend/9/4.15.0-112-generic/x86_64/module/hid-kye.ko
Nothing to do.
cleaning build area...

DKMS: build completed.
dkms install digimend/9

hid-kye:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.15.0-112-generic/updates/dkms/

hid-uclogic.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.15.0-112-generic/updates/dkms/

hid-polostar.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.15.0-112-generic/updates/dkms/

hid-viewsonic.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.15.0-112-generic/updates/dkms/

depmod...

DKMS: install completed.
install -D -m 0755 hid-rebind /lib/udev/hid-rebind
install -D -m 0644 udev.rules /lib/udev/rules.d/90-digimend.rules
install -D -m 0644 xorg.conf /usr/share/X11/xorg.conf.d/50-digimend.conf
udevadm control --reload
```

I then run `sudo modprobe -r hid-kye hid-uclogic hid-polostar hid-viewsonic` and reconnect the tablet. But it still doesn't work.

Another website said to try `xsetwacom --list devices` to see if the tablet was enabled, which I did. There's no output.

I'm wondering if the issue is that I have secure boot enabled. Can anyone tell me how to get this tablet to work?

nic3-14159, 2020-09-06:
What is the output of `modinfo hid-uclogic` after you installed digimend and ran `sudo modprobe -r hid-kye hid-uclogic hid-polostar hid-viewsonic`?

RylanSchaeffer, 2020-09-06:
```
$ modinfo hid-uclogic
filename:       /lib/modules/4.15.0-115-generic/updates/dkms/hid-uclogic.ko
version:        9
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     26978585E76A6B0F0DACC57
alias:          hid:b0003g*v000028BDp00000042
alias:          hid:b0003g*v000028BDp00000094
alias:          hid:b0003g*v000028BDp00000075
alias:          hid:b0003g*v000028BDp00000071
alias:          hid:b0003g*v000028BDp00000074
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000047
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid
retpoline:      Y
name:           hid_uclogic
vermagic:       4.15.0-115-generic SMP mod_unload 
signat:         PKCS#7
signer:         
sig_key:        
sig_hashalgo:   md4

```

nic3-14159, 2020-09-06:
It looks like it installed correctly, but I guess your system isn't loading the module properly or something. I was just making sure it wasn't an issue with the Makefile in older versions of digimend that caused the kernel to load the in-kernel version instead of the one you downloaded and built, which I have seen on my system.

I haven't looked much at secure boot, but based on some quick research, it seems that secure boot can prevent unsigned modules from being loaded. Is there any message in dmesg after you run `sudo modprobe hid-uclogic` that says something about unsigned modules?

RylanSchaeffer, 2020-09-06:
`sudo modprobe hid-uclogic` produces no output:

```
$sudo modprobe hid-uclogic
$
```

nic3-14159, 2020-09-06:
Anything in dmesg?

RylanSchaeffer, 2020-09-06:
I'm not too familiar with drivers. What is dmesg?

nic3-14159, 2020-09-06:
It's a command that shows kernel and driver messages. Just run `dmesg`, and kernel messages should show up. After running `sudo modprobe hid-uclogic`, run `dmesg`, and see if there is anything in the last dozen lines or so that says something about "unsigned modules".

RylanSchaeffer, 2020-09-06:
Ah thank you! Output:

```
[70750.323662] [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
[70750.710505] [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
[70751.432356] wlp5s0: authenticate with 86:bb:69:d8:54:83
[70751.441038] wlp5s0: send auth to 86:bb:69:d8:54:83 (try 1/3)
[70751.583651] wlp5s0: send auth to 86:bb:69:d8:54:83 (try 2/3)
[70751.691415] wlp5s0: send auth to 86:bb:69:d8:54:83 (try 3/3)
[70751.795649] wlp5s0: authentication with 86:bb:69:d8:54:83 timed out
[70751.966379] wlp5s0: authenticate with 84:bb:69:d8:54:80
[70751.969637] wlp5s0: send auth to 84:bb:69:d8:54:80 (try 1/3)
[70752.010874] wlp5s0: authenticated
[70752.011585] wlp5s0: associate with 84:bb:69:d8:54:80 (try 1/3)
[70752.015722] wlp5s0: RX AssocResp from 84:bb:69:d8:54:80 (capab=0x1411 status=0 aid=1)
[70752.020716] wlp5s0: associated
[70752.039291] IPv6: ADDRCONF(NETDEV_CHANGE): wlp5s0: link becomes ready
[70754.244093] ACPI Exception: AE_AML_PACKAGE_LIMIT, Index (0x0000000FF) is beyond end of object (length 0xF) (20170831/exoparg2-430)
[70754.244112] No Local Variables are initialized for Method [GINF]
[70754.244113] Initialized Arguments for Method [GINF]:  (2 arguments defined for method invocation)
[70754.244114]   Arg0:   00000000049f71da <Obj>           Integer 00000000000000FF
[70754.244117]   Arg1:   00000000534dfd84 <Obj>           Integer 0000000000000000
[70754.244122] ACPI Error: Method parse/execution failed \_SB.GINF, AE_AML_PACKAGE_LIMIT (20170831/psparse-550)
[70754.244126] ACPI Error: Method parse/execution failed \_SB.GADR, AE_AML_PACKAGE_LIMIT (20170831/psparse-550)
[70754.244130] ACPI Error: Method parse/execution failed \_SB.SGOV, AE_AML_PACKAGE_LIMIT (20170831/psparse-550)
[70754.244133] ACPI Error: Method parse/execution failed \_SB.CGWR, AE_AML_PACKAGE_LIMIT (20170831/psparse-550)
[70754.244137] ACPI Error: Method parse/execution failed \_SB.TBFP, AE_AML_PACKAGE_LIMIT (20170831/psparse-550)
[70754.244140] ACPI Error: Method parse/execution failed \_SB.WMTF.WMTF, AE_AML_PACKAGE_LIMIT (20170831/psparse-550)
[70925.330293] [drm] Reducing the compressed framebuffer size. This may lead to less power savings than a non-reduced-size. Try to increase stolen memory size if available in BIOS.
[71548.548461] audit: type=1400 audit(1599413012.101:139): apparmor="ALLOWED" operation="open" profile="libreoffice-soffice" name="/usr/share/drirc.d/00-mesa-defaults.conf" pid=9188 comm="soffice.bin" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[71548.550438] audit: type=1400 audit(1599413012.105:140): apparmor="ALLOWED" operation="open" profile="libreoffice-soffice" name="/usr/share/drirc.d/00-mesa-defaults.conf" pid=9188 comm="soffice.bin" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[71548.554136] audit: type=1400 audit(1599413012.105:141): apparmor="ALLOWED" operation="open" profile="libreoffice-soffice" name="/usr/share/drirc.d/00-mesa-defaults.conf" pid=9188 comm="soffice.bin" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[71548.555146] audit: type=1400 audit(1599413012.109:142): apparmor="ALLOWED" operation="open" profile="libreoffice-soffice" name="/usr/share/drirc.d/00-mesa-defaults.conf" pid=9188 comm="soffice.bin" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[71548.564533] audit: type=1400 audit(1599413012.117:143): apparmor="ALLOWED" operation="open" profile="libreoffice-soffice" name="/home/rylan/.cache/mesa_shader_cache/index" pid=9188 comm="soffice.bin" requested_mask="wrc" denied_mask="wrc" fsuid=1000 ouid=1000
[71548.570366] audit: type=1400 audit(1599413012.125:144): apparmor="ALLOWED" operation="open" profile="libreoffice-soffice" name="/usr/share/drirc.d/00-mesa-defaults.conf" pid=9188 comm="soffice.bin" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[71555.814086] audit: type=1400 audit(1599413019.365:145): apparmor="ALLOWED" operation="open" profile="libreoffice-soffice" name="/home/rylan/.thunderbird/profiles.ini" pid=9186 comm="soffice.bin" requested_mask="r" denied_mask="r" fsuid=1000 ouid=1000
[71555.820146] audit: type=1400 audit(1599413019.373:146): apparmor="ALLOWED" operation="open" profile="libreoffice-soffice" name="/home/rylan/.thunderbird/yxyjfdme.default/secmod.db" pid=9186 comm="soffice.bin" requested_mask="r" denied_mask="r" fsuid=1000 ouid=1000
[71555.823446] audit: type=1400 audit(1599413019.377:147): apparmor="ALLOWED" operation="open" profile="libreoffice-soffice" name="/home/rylan/.thunderbird/yxyjfdme.default/cert8.db" pid=9186 comm="soffice.bin" requested_mask="wr" denied_mask="wr" fsuid=1000 ouid=1000
[71555.823887] audit: type=1400 audit(1599413019.377:148): apparmor="ALLOWED" operation="open" profile="libreoffice-soffice" name="/home/rylan/.thunderbird/yxyjfdme.default/key3.db" pid=9186 comm="soffice.bin" requested_mask="wr" denied_mask="wr" fsuid=1000 ouid=1000
[71842.757411] input: 00:00:73:00:03:60 as /devices/virtual/input/input26
[75457.573909] hid_uclogic: loading out-of-tree module taints kernel.
```

nic3-14159, 2020-09-06:
Hmm, I don't see anything that stands out to me, and I don't believe that last line indicates any issues. What does dmesg say after you plug in your tablet?

spbnick, 2020-09-13:
@RylanSchaeffer, it seems your installation went fine, and it looks like your modules were even signed (!), judging from the output you posted. Let's check if you get the right module loaded. Please plug in the tablet and post the output of `sudo cat /sys/module/hid_uclogic/version`. Please also post the output of `sudo evdev` for your tablet.