DIGImend/digimend-kernel-drivers #455: Unable to configure HS64's buttons

augustoolucas, 2020-09-12:
Hello guys,

I tried to configure the buttons of my HS64 using the [instructions on readme](https://github.com/DIGImend/digimend-kernel-drivers#configuring-wacom-xorg-driver), but the command `xsetwacom "device_id" button x key any_key` outputs: 

>Property 'Wacom Button Actions' does not exist on device.

Here's the output of `xsetwacom list`:

>HUION Huion Tablet Touch Strip pad	id: 13	type: PAD       
>HUION Huion Tablet Dial pad     	id: 14	type: PAD 

`xinput list` shows this:

>⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
>⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
>⎜   ↳ HUION Huion Tablet Pen                  	id=11	[slave  pointer  (2)]
>⎜   ↳ HUION Huion Tablet Pad                  	id=12	[slave  pointer  (2)]
>⎜   ↳ HUION Huion Tablet Touch Strip pad      	id=13	[slave  pointer  (2)]
>⎜   ↳ HUION Huion Tablet Dial pad             	id=14	[slave  pointer  (2)]

I was able to switch the display by using `xsetwacom set "device_id" MapToOutput HDMI-X`, but the `device_id` that worked was 11, not 13 or 14 as listed by `xsetwacom list`. Any of the ids I tried to set the buttons outputs that "does not exist on device" message.

I've installed the drivers using these instructions: https://github.com/DIGImend/digimend-kernel-drivers/issues/354#issuecomment-594785170 

After that I did these steps: https://github.com/DIGImend/digimend-kernel-drivers#enabling-wacom-xorg-driver

And then rebooted the system. Did I do something wrong? Do I need to provide any more info?

Thank you.

Popolon, 2020-10-09:
As far I undestand, you need to set again the keys (with xsetwacom) at each X11 start, and probably after the tablet detection ?

omnitrix9991, 2021-07-05:
[huion-HS64.tablet.tar.gz](https://github.com/DIGImend/digimend-kernel-drivers/files/6762762/huion-HS64.tablet.tar.gz)
<p> the tablet works fully, even the GUI configurator, you need digimend v10 drivers, now copy the huion-HS64.tablet file to /usr/share/libwacom </p>
<p>thanks to juansegaes for making the .tablet file , just had to edit the name according to xsetwacom devices</p>
[here](https://github.com/juansegaes/HS64)
</p> maybe you can use the file as it is, if you edit files as per this [commit](https://github.com/Huion-Linux/DIGImend-kernel-drivers-for-Huion/commit/66605efcffdb00232439a209acf2cd3fda714980) </p>

omnitrix9991, 2021-07-05:
[huion-HS64.tablet.tar.gz](https://github.com/DIGImend/digimend-kernel-drivers/files/6762843/huion-HS64.tablet.tar.gz)

the tablet works fully, even the GUI configurator, you need digimend v10 drivers, now copy the huion-HS64.tablet file to /usr/share/libwacom

thanks to juansegaes for making the .tablet file , just had to edit the name according to xsetwacom devices for v10 drivers
[juansegaes huion-HS64.tablet file](https://github.com/juansegaes/HS64)
maybe you can use the file as it is, if you edit files in v10 driver , as per this [commit](https://github.com/Huion-Linux/DIGImend-kernel-drivers-for-Huion/commit/66605efcffdb00232439a209acf2cd3fda714980)