DIGImend/digimend-kernel-drivers #456: Huion Q11K appears as a "screen"/"display" in Linux Mint/Cinnamon 20 "graphics tablet" config panel but wasn't in 18.3

twiggy, 2020-09-13:
Got a Q11K a bit ago and was running it in Linux Mint 18.3 and had some really good luck with it.

I upgraded ( reinstalled ) Mint to get up to the latest v20. After compiling from source the drivers loaded up great and I was able to use the tablet and it showed up in the panel/config. 

I noticed though that it looked different than it did in 18.3. The option for left handed ( I am ) was gone..and Map to monitor was gone.

After digging through the Mint/Cinnamon code https://github.com/linuxmint/cinnamon-control-center/tree/master/panels/wacom it appears that it thinks its a screen now. Which is why the " adjust display resolution" is showing.

[image]

I've been digging through my 50-digimend.conf in my new and old install file seeing if I could force it to Pad from screen, but I'm not sure I fully understand the syntax etc beyond matching the USB ID etc.

I can fix the map to monitor via xsetcom and I'm guessing left hand is just some kind of rotate. If someone could point me at where the code might be determining screen vs pad I'm happy to keep digging as it'd be nice to be able to use the Mint config panel vs the xsetwacom stuff.

twiggy, 2020-09-13:
This turns out to be more of a libwacom issue. I added a custom file to /usr/share/libwacom based on a copy of huion-420.tablet

From there i set 

Name=whateveryouwant
DeviceMatch=usb:256c:006e:TABLET Pen Tablet Pen;

Now libwacom-list-local-devices shows my device and cinnamon shows its correctly ( for the most part ). This should work on Ubuntu/Gnome as well I think, but have not tested

Read https://github.com/linuxwacom/libwacom towards the bottom for more info. https://github.com/linuxwacom/libwacom/blob/master/data/wacom.example also helpful.

I'll keep working on this .tablet file and see if I can't get it supporting button mapping.

spbnick, 2020-09-13:
Thank you for the report, @twiggy. I think this might be related to #442.

spbnick, 2020-10-05:
Should be fixed in the v10 release.

closed 2020-10-05 by spbnick