DIGImend/digimend-kernel-drivers #458: Installation not possible with 5.8.10 kernel : dkms error

OlivierV78, 2020-09-18:
Hi,

Installation is not possible with 5.8.10 kernel : there is a dkms error.

`meloli@Asus-A17:~/Téléchargements$ sudo gdebi digimend-dkms_9_all.deb 
[sudo] Mot de passe de meloli : 
Reading package lists... Done
Building dependency tree        
Reading state information... Done
Reading state information... Done

Collection of graphics tablet drivers by DIGImend project
 This is a collection of graphics tablet drivers for the Linux kernel, produced
 and maintained by the DIGImend project.
 .
 See http://digimend.github.io/drivers/digimend/tablets/ for a list of
 supported tablet models.
Voulez-vous installer le paquet logiciel ? [o/N] :o
/usr/bin/gdebi:113: FutureWarning: Possible nested set at position 1
  c = findall("[[(](\S+)/\S+[])]", msg)[0].lower()
Sélection du paquet digimend-dkms précédemment désélectionné.
(Lecture de la base de données... 312798 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de digimend-dkms_9_all.deb ...
Dépaquetage de digimend-dkms (9) ...
Paramétrage de digimend-dkms (9) ...
Loading new digimend-9 DKMS files...
Building for 5.8.10-050810-generic
Building initial module for 5.8.10-050810-generic
ERROR (dkms apport): kernel package linux-headers-5.8.10-050810-generic is not supported
Error! Bad return status for module build on kernel: 5.8.10-050810-generic (x86_64)
Consult /var/lib/dkms/digimend/9/build/make.log for more information.
dpkg: erreur de traitement du paquet digimend-dkms (--install) :
 installed digimend-dkms package post-installation script subprocess returned error exit status 10
Des erreurs ont été rencontrées pendant l'exécution :
 digimend-dkms
`

LittleWhole, 2020-09-21:
I am getting the exact same error. I can confirm this.

cesarkero, 2021-04-25:
This is what i'm getting too: 
```
Error! Bad return status for module build on kernel: 5.8.0-50-generic (x86_64)
Consult /var/lib/dkms/digimend/10/build/make.log for more information.
dpkg: error al procesar el paquete digimend-dkms (--configure):
 el subproceso instalado paquete digimend-dkms script post-installation devolvió
 el código de salida de error 10
Se encontraron errores al procesar:
 digimend-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

VerLeth, 2025-04-24:
I'm also getting this same error message, have any of you guys managed to find a fix ?