DIGImend/digimend-kernel-drivers #459: [Support request]  Bosto 16HDT

chiboreache, 2020-09-18:
awesome analog of WACOM Cintiq Pro 16, but 7 times cheaper

please make it work..

chiboreache, 2020-09-18:
Manufacturer: Nuvoton

nic3-14159, 2020-10-04:
Hi! Please post tablet diagnostics as described [here](https://digimend.github.io/support/howto/trbl/diagnostics/), which would be useful for anyone working on support for this tablet.

chiboreache, 2020-10-18:
There is an issue with page [Collecting tablet diagnostics](http://digimend.github.io/support/howto/trbl/diagnostics/):

[finger_touch_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5396943/finger_touch_coords.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5396944/probe.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5396945/hid_report_descriptors.txt)
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5396946/descriptors.txt)

nic3-14159, 2020-10-18:
What exactly seems to be the issue?

chiboreache, 2020-10-18:
pen not working at all

touch working with cursor offset, when it plugged in as second display

dancingfoot, 2020-12-26:
> pen not working at all
> 
> touch working with cursor offset, when it plugged in as second display

The touch worked after you install the DIGImend drivers? Did you install anything else?
Did you install with deb pakage? Or compiled?

chiboreache, 2020-12-26:
> > pen not working at all
> > touch working with cursor offset, when it plugged in as second display
> 
> The touch worked after you install the DIGImend drivers? Did you install anything else?
> Did you install with deb pakage? Or compiled?

No. No. No. No.

chiboreache, 2020-12-26:
@dancingfoot 
do you think it's doable? 

we have holidays soon (and sort of black friday) so i thinking about sell it

and buy wacom later, since it already have awesome Linux support

dancingfoot, 2021-04-16:
The pen is working with Wacom drivers.
Link to post: https://github.com/DIGImend/digimend-kernel-drivers/issues/518