DIGImend/digimend-kernel-drivers #463: Huion Kamvas Pro 20 report

Popolon, 2020-09-28:
I have since today a kamvas pro 20, USB ID 256c:006e. The pen works perfectly (at least for coordinates and pressure, didn't tested tilt, and don't know which tool I could use for it.

https://www.huion.com/pen_display/KamvasPro/Kamvas-Pro-20(2019).html

There has 8 buttons and a 'slider' on each side of the tablet, the 'slider' have strange results in evdev, it looks like the two are well detected (2 axis), but moving fingers on it is displayed as pressure on other buttons.

Here is the report of evdev on event22 (buttons, 21 is the pen)
<pre>
/dev/input/event21:	Tablet Monitor Pen
/dev/input/event22:	Tablet Monitor Pad
Select the device event number [0-22]: 22
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
Input device name: "Tablet Monitor Pad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 256 (BTN_0)
    Event code 257 (BTN_1)
    Event code 258 (BTN_2)
    Event code 259 (BTN_3)
    Event code 260 (BTN_4)
    Event code 261 (BTN_5)
    Event code 262 (BTN_6)
    Event code 263 (BTN_7)
    Event code 264 (BTN_8)
    Event code 265 (BTN_9)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 331 (BTN_STYLUS)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max        1
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max        1
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
</pre>

 I tried to follow http://digimend.github.io/support/howto/trbl/diagnostics/ as someone suggested me on the IRC channel.

The first part, "frame_wheel_srolling.txt" wasn't very clear for me, so I made two, one for the left slider (frame_left_wheel_srolling.txt), and one for the right one (frame_right_wheel_srolling.txt). starting from bottom, going to top, and coming back to button. I followed what was asked for other part.

The same way I tried to follow the movement of the Dials, turning around the dials, on what I call sliders, even if they are only vertical one, I don't know if there is left-right sensitivity.

Popolon, 2020-09-28:
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5294815/descriptors.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5294821/probe.txt)
[frame_left_wheel_srolling.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5294823/frame_left_wheel_srolling.txt)
[frame_right_wheel_srolling.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5294824/frame_right_wheel_srolling.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5294826/pen_coords.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5294828/pen_tilt.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5294831/pen_pressure.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5294833/pen_buttons.txt)
[frame_dials.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5294837/frame_dials.txt)
[frame_left_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5294839/frame_left_buttons.txt)
[frame_right_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5294840/frame_right_buttons.txt)

Popolon, 2020-09-28:
I was not sure for the tilt, but it looks like it isn't currently managed. I mean, the stylus has tilt management, but the driver doesn't manage it. Here is what display evtest, only coordinates on screen are displayed:
<pre>
Select the device event number [0-22]: 21
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
Input device name: "Tablet Monitor Pen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 320 (BTN_TOOL_PEN)
    Event code 330 (BTN_TOUCH)
    Event code 331 (BTN_STYLUS)
    Event code 332 (BTN_STYLUS2)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value  40216
      Min        0
      Max    86970
      Resolution     200
    Event code 1 (ABS_Y)
      Value  29976
      Min        0
      Max    47752
      Resolution     200
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     8191
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
Event: time 1601327770.279002, type 1 (EV_KEY), code 320 (BTN_TOOL_PEN), value 1
Event: time 1601327770.279002, type 3 (EV_ABS), code 0 (ABS_X), value 46051
Event: time 1601327770.279002, type 3 (EV_ABS), code 1 (ABS_Y), value 19303
Event: time 1601327770.279002, -------------- SYN_REPORT ------------
Event: time 1601327770.321018, type 3 (EV_ABS), code 0 (ABS_X), value 46096
Event: time 1601327770.321018, type 3 (EV_ABS), code 1 (ABS_Y), value 19306
Event: time 1601327770.321018, -------------- SYN_REPORT ------------
</pre>

Popolon, 2020-09-28:
I can confirm that the tilt isn't managed, by, under Krita, by changing the option of a brush, for size:
* disabling, pressure
* enabling tilt instead.

The size doesn't change when I tilt the pen.

Popolon, 2020-09-29:
Yesterday I installed the Arch Linux package (AUR) digimend-kernel-drivers-dkms-git but didn't rebooted the system. 

After booting today, most things seems to be well detected by evtest and work (pen tilt, 3 buttons are still missing, and, I don't know if the 2 strips (that I called sliders), can be managed as 2 entries, they use the same event currently). Here are the informations I obtain :

<pre>
/dev/input/event22:	Tablet Monitor Pen
/dev/input/event23:	Tablet Monitor Pad
/dev/input/event24:	Tablet Monitor Touch Strip
/dev/input/event25:	Tablet Monitor Dial
</pre>
Pen tilt is now managed by Code 26 and 27, I'm not sure, but the values are probably from -60 to 60, as, the pen manage 60° angles :
* Tilt X : type 3 (EV_ABS), code 26 (ABS_TILT_X), value 54
* Tilt Y : type 3 (EV_ABS), code 27 (ABS_TILT_Y), value -40

The  Dial don't seems to be used, what I called "slide", is managed by "Touch Strip".

Both sides use the same event (I don't know if hardware allow to detect them as separate) :
* type 3 (EV_ABS), code 40 (ABS_MISC)  15 for touch down, and 0 for touch up
* type 3 (EV_ABS), code 8 (ABS_WHEEL), value 0 to 6 (0 top, 6 bottom), for strips.
<pre>
Event: time 1601379017.615285, type 3 (EV_ABS), code 40 (ABS_MISC), value 15
Event: time 1601379017.615285, -------------- SYN_REPORT ------------
Event: time 1601379017.689285, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 1
Event: time 1601379017.689285, -------------- SYN_REPORT ------------
Event: time 1601379017.789285, type 3 (EV_ABS), code 8 (ABS_WHEEL), value 2
</pre>

Buttons at left are mapped from 0 to 7 (Code 256 to 263)
Only 5 of the 8 buttons are mapped at right (8, 9, SOUTH, EAST,  C, code from 264 to 306), 3 are still missing at the bottom.
<pre>
Event: time 1601379797.204047, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90008
Event: time 1601379797.204047, type 1 (EV_KEY), code 263 (BTN_7), value 0
</pre>

Popolon, 2020-10-09:
After applying the Wacom mapping in <code>/etc/X11/xorg.conf.d/50-tablet.conf<code>
Section "InputClass"
    Identifier "Tablet"
    Driver "wacom"
    MatchDevicePath "/dev/input/event*"
    MatchUSBID "256c:006e"
EndSection

libwacom shows:
<pre>
$ libwacom-list-local-devices
# Device node: /dev/input/event22
[Device]
Name=GAOMON S56K
ModelName=
DeviceMatch=usb:256c:006e:Tablet Monitor Pen;usb:256c:006e:Tablet Monitor Pad;
Class=Bamboo
Width=6
Height=5
IntegratedIn=
Styli=0xffffd;

[Features]
Reversible=false
Stylus=true
Ring=false
Ring2=false
Touch=false
TouchSwitch=false
# StatusLEDs=
NumStrips=0
Buttons=0
</pre>

I can set with xsetwacom the 3 first upper left buttons 1/2/3
xsetwacom --set "Tablet Monitor Pad pad"  Button 1 "Key i"
xsetwacom --set "Tablet Monitor Pad pad"  Button 2 "Key n"
xsetwacom --set "Tablet Monitor Pad pad"  Button 3 "Key b"

But for fourth button:
<pre>
xsetwacom --set "Tablet Monitor Pad pad"  Button 4 "Key p"
Unsupported offset into 'Wacom Button Actions' property.
</pre>

Popolon, 2020-10-09:
I seen on libwacom wiki there is a toolset called evemu, this return interesting information, about the stripe.

I used <code>evemu-record</code>, that display the events (didn't write them here, it detect as evtest strip events, but there is other informations about the device):
<pre>
/dev/input/event21:	Tablet Monitor
/dev/input/event22:	Tablet Monitor Pad
/dev/input/event23:	Tablet Monitor Touch Strip
/dev/input/event24:	Tablet Monitor Dial
Select the device event number [0-24]: 23
# EVEMU 1.3
# Kernel: 5.8.14-arch1-1
# DMI: dmi:bvnAmericanMegatrendsInc.:bvr0401:bd07/15/2011:br4.6:svnSystemmanufacturer:pnSystemProductName:pvrSystemVersion:rvnASUSTeKComputerINC.:rnP8Z68-MPRO:rvrRev1.xx:cvnChassisManufacture:ct3:cvrChassisVersion:
# Input device name: "Tablet Monitor Touch Strip"
# Input device ID: bus 0x03 vendor 0x256c product 0x6e version 0x111
# Supported events:
#   Event type 0 (EV_SYN)
#     Event code 0 (SYN_REPORT)
#     Event code 1 (SYN_CONFIG)
#     Event code 2 (SYN_MT_REPORT)
#     Event code 3 (SYN_DROPPED)
#     Event code 4 ((null))
#     Event code 5 ((null))
#     Event code 6 ((null))
#     Event code 7 ((null))
#     Event code 8 ((null))
#     Event code 9 ((null))
#     Event code 10 ((null))
#     Event code 11 ((null))
#     Event code 12 ((null))
#     Event code 13 ((null))
#     Event code 14 ((null))
#     Event code 15 (SYN_MAX)
#   Event type 1 (EV_KEY)
#     Event code 256 (BTN_0)
#   Event type 3 (EV_ABS)
#     Event code 0 (ABS_X)
#       Value        0
#       Min          0
#       Max          1
#       Fuzz         0
#       Flat         0
#       Resolution   0
#     Event code 1 (ABS_Y)
#       Value        0
#       Min          0
#       Max          1
#       Fuzz         0
#       Flat         0
#       Resolution   0
#     Event code 8 (ABS_WHEEL)
#       Value        0
#       Min          0
#       Max          7
#       Fuzz         0
#       Flat         0
#       Resolution   3
#     Event code 40 (ABS_MISC)
#       Value        0
#       Min          0
#       Max        255
#       Fuzz         0
#       Flat         0
#       Resolution   0
# Properties:
N: Tablet Monitor Touch Strip
I: 0003 256c 006e 0111
P: 00 00 00 00 00 00 00 00
B: 00 0b 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 01 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 02 00 00 00 00 00 00 00 00
B: 03 03 01 00 00 00 01 00 00
B: 04 00 00 00 00 00 00 00 00
B: 05 00 00 00 00 00 00 00 00
B: 11 00 00 00 00 00 00 00 00
B: 12 00 00 00 00 00 00 00 00
B: 14 00 00 00 00 00 00 00 00
B: 15 00 00 00 00 00 00 00 00
B: 15 00 00 00 00 00 00 00 00
A: 00 0 1 0 0 0
A: 01 0 1 0 0 0
A: 08 0 7 0 0 3
A: 28 0 255 0 0 0
</pre>

Popolon, 2020-10-11:
with code : 08 E0 01 01, the 2 following bytes are divided in 4 bits by button ranges

<pre>
08 E0 01 01 00 00 00 00 00 00 00 00
            12 34
</pre>
* 1: bottom left
* 2: top left
* 3: bottom right
* 4: top right.
For each of this groups, the buttons goes from top to button with 1 bit for each so
* button1: 1
* button2: 2
* button3: 4
* button4: 8
The 16 buttons are reported by usbhid-dump, but 3 last one are myssing in what hid-uclogic driver report to input/event. I didn't find in the code how to fix that, as far I understand, the parameters are self detected ? 

I added to  hid-ids.h:
<pre>
#define USB_DEVICE_ID_HUION_TABLET3	0x006c
</pre>
Then, in hid-uclogic-core.c, in static const struct hid_device_id uclogic_devices
        { HID_USB_DEVICE(USB_VENDOR_ID_HUION,
                                USB_DEVICE_ID_HUION_TABLET3) },
In uclogic_devices.c, as other HUION_TABLET*:
        case VID_PID(USB_VENDOR_ID_HUION,
                     USB_DEVICE_ID_HUION_TABLET3):
So it can go to uclogic_params_huion_init(&p, hdev);
unloaded/reloaded the divers, but it still doesn't send all the button to evdev (using wacom driver)

But it looks like uclogic-probe has already some problems to read Params block #2:
<pre>
    |                Product: Tablet Monitor
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 86970
    |                          Max Y: 47752
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |       Firmware version: HUION_M193_190606
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd
</pre>

erjiang, 2020-10-30:
The 22-inch version that I have has the same USB product ID. I hacked together [some code to custom configure the buttons](https://github.com/erjiang/huion-keys), but it would be better to have the driver properly support it. I dumped the data from hidraw for the frame buttons in case it would be helpful: https://github.com/erjiang/huion-keys/blob/main/huion_dump.txt

Popolon, 2020-10-31:
I tried on ArchLinux with, for installing packages:
sudo pacman -S base-devel xdo python-cffi

My version of:
* xdo is  0.5.7
* python 3.8.6
* python-cffi 1.14.3

doing
<pre>
 python xdo_build.py
</pre>

 I have the following errors:
<pre>
_xdo_cffi.c: In function '_cffi_checkfld__xdo_t':
_xdo_cffi.c:5106:32: error: 'xdo_t' {aka 'struct xdo'} has no member named 'modmap'
 5106 |   { XModifierKeymap * *tmp = &p->modmap; (void)tmp; }
      |                                ^~
_xdo_cffi.c:5107:30: error: 'xdo_t' {aka 'struct xdo'} has no member named 'keymap'
 5107 |   { unsigned long * *tmp = &p->keymap; (void)tmp; }
      |                              ^~
In file included from _xdo_cffi.c:61:
_xdo_cffi.c: At top level:
_xdo_cffi.c:5269:15: error: 'xdo_t' {aka 'struct xdo'} has no member named 'modmap'
 5269 |   { "modmap", offsetof(xdo_t, modmap),
      |               ^~~~~~~~
_xdo_cffi.c:5270:34: error: 'xdo_t' {aka 'struct xdo'} has no member named 'modmap'
 5270 |               sizeof(((xdo_t *)0)->modmap),
      |                                  ^~
In file included from _xdo_cffi.c:61:
_xdo_cffi.c:5272:15: error: 'xdo_t' {aka 'struct xdo'} has no member named 'keymap'
 5272 |   { "keymap", offsetof(xdo_t, keymap),
      |               ^~~~~~~~
_xdo_cffi.c:5273:34: error: 'xdo_t' {aka 'struct xdo'} has no member named 'keymap'
 5273 |               sizeof(((xdo_t *)0)->keymap),
      |                                  ^~
</pre>

The xdo sturcture in /usr/include/xdo.h:
<pre>
/**
 * The main context.
 */
typedef struct xdo {

  /** The Display for Xlib */
  Display *xdpy;

  /** The display name, if any. NULL if not specified. */
  char *display_name;

  /** @internal Array of known keys/characters */
  charcodemap_t *charcodes;

  /** @internal Length of charcodes array */
  int charcodes_len;

  /** @internal highest keycode value */
  int keycode_high; /* highest and lowest keycodes */

  /** @internal lowest keycode value */
  int keycode_low;  /* used by this X server */
  
  /** @internal number of keysyms per keycode */
  int keysyms_per_keycode;

  /** Should we close the display when calling xdo_free? */
  int close_display_when_freed;

  /** Be extra quiet? (omits some error/message output) */
  int quiet;

  /** Enable debug output? */
  int debug;

  /** Feature flags, such as XDO_FEATURE_XTEST, etc... */
  int features_mask;

} xdo_t;
</pre>

erjiang, 2020-11-01:
@Popolon Let's move this conversation to https://github.com/erjiang/huion-keys/issues/1 since it's not about the digimend code.

Popolon, 2020-11-01:
It works now, as wrote in the ticket. I wrote this her for other Huion kamvas pro20/22 (and 24?) users.

Popolon, 2020-11-08:
About pen-screen calibration
<pre>
top-left       08 81 02 00 02 00 FF 1F 00 00 00 00
bottom-right   08 81 EC 52 63 B8 3D 1A 01 00 0C FF
h              v     x3 x2 y2 y1       x1
</pre>
Pen coordinates are x1x2x3, y1,y2

<pre>
hexa  deci     hexa deci
000002         0002 <= min border (top left, looks like the display border too:w)
152EC 86764    B863 47203 <= display border (bottom right)
15423 87075    BA88 47752 <= coordinates max magnetic field (bottom right)
</pre>
Defaut xsetwacom coordinates :
<pre>
xsetwacom set 'Tablet Monitor Pen stylus' "Area" "0 0 86970 47752"
<pre>

What I set, that's not perfect but already far better, I will update/replace this if I get better results
<pre>
xsetwacom set 'Tablet Monitor Pen stylus' "Area" "0 -200  86764 47203"
</pre>

Popolon, 2020-11-11:
Not related to Digimend driver but related to this tablet, I used [Monica](https://sources.archlinux.org/other/monica/) (there is an AUR package for ArchLinux) to calibrate display colors, that's still not perfect in my point of view, but that's vert easy to setup and already work well.