DIGImend/digimend-kernel-drivers #467: Support for Gaomon PD1161

PushkinKachroo, 2020-10-13:
I just bought Gaomon PD1161, which seems like a popular and affordable drawing tablet.  However, it seems to not have support in digimend-kernel-drivers.  I installed it on Ubuntu 18.04LTS, but it is having problems for this device.

Can the driver for Gaomon PD1161 be included PLEASE?

Popolon, 2020-10-13:
I believe this tablet only existe since spring 2018, so after Ubuntu 18.04. What is the usb ID (you can have it with lsusb. What work and don't work ?

You can help developers by reporting all the following diagnostic of your tablet: http://digimend.github.io/support/howto/trbl/diagnostics/

PushkinKachroo, 2020-10-13:
I am getting peculiar behaviors.  When I just connected Gaomon PD1161, the screen came on and the cursor which would move with the stylus.  However, there was a shifted cursor, i.e. the cursor would not be at the point of contact of the stylus with the tablet.  So, it needs configuration.  After, I installed digimend driver, the stylus would stop working.  I tried using the deb file as well as building ot from source with the same outcomes.  Now, I uninstalled the driver, and once again, I have the screen, but now the cursor is on my main laptop screen and will stay there when I use the stylus, but moves normally with my laptop mouse.

Moreover, the screen (using HDMI) shows when I connect via an adapter that uses the USB-C connector of the laptop, but not when I use the HDMI on the laptop directly.  The lusb output is below:

pushkin@pushkin-Inspiron-7573:~$ lsusb
Bus 002 Device 016: ID 05e3:0749 Genesys Logic, Inc. 
Bus 002 Device 015: ID 0bc2:331a Seagate RSS LLC 
Bus 002 Device 014: ID 05e3:0626 Genesys Logic, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 8087:0a2a Intel Corp. 
Bus 001 Device 004: ID 0bda:5693 Realtek Semiconductor Corp. 
Bus 001 Device 017: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 016: ID 256c:006d  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pushkin@pushkin-Inspiron-7573:~$

PushkinKachroo, 2020-10-13:
Do you think upgrading to ubuntu 20.04 LTS could resolve this issue, as this model came in 2018?

Popolon, 2020-10-17:
For the screen, that's not a probelm with the driver, but with the desktop

If you are under GNOME, you should first set your tablet as Wacom in x.org (looks like it doesn't work with last git version of the driver)
On other there are probably integrated conf too.

I use XFCE and PTXconf, but this need to be set again at each start.

So I put in a shell script, as my tablet screen is HDMI-3:
<pre>
stylus=`xsetwacom --list devices | grep STYLUS | sed  's/\s*id:.*//'`
# set the screen
xsetwacom set "$stylus" MapToOutput HDMI-3
</pre>

soulslicer, 2020-10-17:
Did you managed to get it to work?

PushkinKachroo, 2020-10-22:
No, it is still not working and has the exact same behavior as before.

As I stated before, I have removed the digimend driver, and now I can see the screen on the pad, but the curson never goes to the pad but stays on the main laptop screen.

Can you PLEASE give me step by step instructions what I should try?

PushkinKachroo, 2020-10-22:
> Did you managed to get it to work?
No, it is still not working and has the exact same behavior as before. :(

Popolon, 2020-10-22:
Which desktop environement do you use ? GNOME, KDE, XFCE?

PushkinKachroo, 2020-10-22:
> Which desktop environement do you use ? GNOME, KDE, XFCE?

Gnome 3.28.2

Popolon, 2020-10-22:
Did you tried the link http://digimend.github.io/support/howto/drivers/wacom/ else could you read what I wrote first. If you follow this, gnome settings will automatically detect it as a wacom, and you will be able to set screen associated  with your tablet.

Popolon, 2020-10-22:
https://help.gnome.org/users/gnome-help/stable/wacom-multi-monitor.html.en

PushkinKachroo, 2020-10-22:
> https://help.gnome.org/users/gnome-help/stable/wacom-multi-monitor.html.en

No wacom tablet shows up.  If I install the digimend driver, then the wacom settings show up if I use my bamboo tablet BUT NOT when I connect the GAOMON PD1161.

PushkinKachroo, 2020-10-22:
Using xinput list gives me:

⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ DELL07EC:00 06CB:7E92 Mouse             	id=17	[slave  pointer  (2)]
⎜   ↳ DELL07EC:00 06CB:7E92 Touchpad          	id=18	[slave  pointer  (2)]
⎜   ↳ SynPS/2 Synaptics TouchPad              	id=25	[slave  pointer  (2)]
⎜   ↳ ELAN2097:00 04F3:2504                   	id=19	[slave  pointer  (2)]
⎜   ↳ Logitech USB Keyboard                   	id=12	[slave  pointer  (2)]
⎜   ↳ Logitech USB Optical Mouse              	id=14	[slave  pointer  (2)]
⎜   ↳ Tablet Monitor Pen Pen (0)              	id=26	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ Integrated_Webcam_HD: Integrate         	id=15	[slave  keyboard (3)]
    ↳ Integrated_Webcam_HD: Integrate         	id=16	[slave  keyboard (3)]
    ↳ ELAN2097:00 04F3:2504                   	id=20	[slave  keyboard (3)]
    ↳ Intel Virtual Button driver             	id=21	[slave  keyboard (3)]
    ↳ Intel HID events                        	id=22	[slave  keyboard (3)]
    ↳ Dell WMI hotkeys                        	id=23	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=24	[slave  keyboard (3)]
    ↳ Tablet Monitor Pen                      	id=10	[slave  keyboard (3)]
    ↳ Logitech USB Keyboard                   	id=11	[slave  keyboard (3)]
    ↳ Logitech USB Keyboard                   	id=13	[slave  keyboard (3)]

PushkinKachroo, 2020-10-22:
Remember, I have uninstalled the digimend driver.  I have the Gaomon PD1161 connected for the xinput and lusb commands

lusb output is the following:
Bus 002 Device 010: ID 05e3:0749 Genesys Logic, Inc. 
Bus 002 Device 009: ID 0bc2:331a Seagate RSS LLC 
Bus 002 Device 008: ID 05e3:0626 Genesys Logic, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 8087:0a2a Intel Corp. 
Bus 001 Device 004: ID 0bda:5693 Realtek Semiconductor Corp. 
Bus 001 Device 012: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 011: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 010: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 009: ID 256c:006d  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Edelweissss, 2023-07-17:
Did you ever find a fix to get this tablet model to work?

Popolon, 2023-12-15:
If it is still not managed need at least to know what is it's USB ID. what is the line that appear when you connect the tablet and disappear when disconnecting it? Is the one with : 256c:006d ?