DIGImend/digimend-kernel-drivers #469: UGEE M708 V2 no one cares anymore?.. Anyone can help with this issue?

Saizzou, 2020-10-20:
Well Hi everyone,
I know its been told that ugee m708 V2 **is not compatible** but I just tried some configuration and well it almost fully works.. I installed the digimend-kernel-drivers as mentioned in the installation. Then I have done some configuration:

1)The Working Path:

Changed the /X11/xorg.conf.d/70-wacom.conf as I added:
```
`# UGEE 10 inch PenTablet
Section "InputClass"
        Identifier "10 inch PenTablet"
        MatchProduct "28bd:0907"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
EndSection
`
```
at the end of the line. I restart my pc and well my stylus and Tablet is now detected and I can see how my stylus is moving my cursor and clicking on the Tablet is also make a left click. (Holding the Pen on the Tablet is also working as: Holding Left Mouse Button)

Here the output:
```
`$xsetwacom list
10 inch PenTablet Mouse touch   	id: 8	type: TOUCH     
10 inch PenTablet stylus        	id: 9	type: STYLUS `
```

2) The 1st Failure:

Now after I saw that the Buttons are not working I tried to configure my Pen Buttons (This Model have 2 on the Stylus Pen):
```
`$xsetwacom set "10 inch PenTablet stylus" Button 1 1
$xsetwacom set "10 inch PenTablet stylus" Button 3 2
$xsetwacom set "10 inch PenTablet stylus" Button 2 3`
```

PS: Front button (Button 3) as Right Click and other one as Middle Click. 

Here started my first error. The Pen Buttons 1-2 are OK. (Pen Button 1 is the Touching the Tablet function) So I decided to check the overall configuration. Here the output:
```
`$xsetwacom -s get "10 inch PenTablet stylus" all
xsetwacom set "10 inch PenTablet stylus" "Area" "0 0 32767 32767"
xsetwacom set "10 inch PenTablet stylus" "Button" "1" "button +1 "
xsetwacom set "10 inch PenTablet stylus" "Button" "2" "button +2 "
xsetwacom set "10 inch PenTablet stylus" "Button" "3" "button +3 "
xsetwacom set "10 inch PenTablet stylus" "Button" "8" "button +8 "
xsetwacom set "10 inch PenTablet stylus" "ToolDebugLevel" "0"
xsetwacom set "10 inch PenTablet stylus" "TabletDebugLevel" "0"
xsetwacom set "10 inch PenTablet stylus" "Suppress" "2"
xsetwacom set "10 inch PenTablet stylus" "RawSample" "4"
xsetwacom set "10 inch PenTablet stylus" "PressureCurve" "0 0 100 100"
xsetwacom set "10 inch PenTablet stylus" "Mode" "Absolute"
xsetwacom set "10 inch PenTablet stylus" "TabletPCButton" "off"
xsetwacom set "10 inch PenTablet stylus" "Touch" "off"
Property 'Wacom Hardware Touch Switch' does not exist on device.
xsetwacom set "10 inch PenTablet stylus" "Gesture" "off"
xsetwacom set "10 inch PenTablet stylus" "ZoomDistance" "0"
xsetwacom set "10 inch PenTablet stylus" "ScrollDistance" "0"
xsetwacom set "10 inch PenTablet stylus" "TapTime" "250"
xsetwacom set "10 inch PenTablet stylus" "CursorProximity" "30"
xsetwacom set "10 inch PenTablet stylus" "Rotate" "none"
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
xsetwacom set "10 inch PenTablet stylus" "Threshold" "26"
xsetwacom set "10 inch PenTablet stylus" "BindToSerial" "0"
xsetwacom set "10 inch PenTablet stylus" "PressureRecalibration" "on"
xsetwacom set "10 inch PenTablet stylus" "PanScrollThreshold" "2795"
`
```
Well from here I can see that there is actually a Button 8 configure so I tried to configure it too but without any success.

3) 3rd Failure The Tablet:

Just to cut it short the turning or MapToOutput function is all ok and configurable. Just on the Tablet I have 8 buttons which I couldn't manage to configure. Here again a output for get -s output:
```
`xsetwacom -s get "10 inch PenTablet Mouse touch" all
xsetwacom set "10 inch PenTablet Mouse touch" "Area" "0 0 32767 32767"
xsetwacom set "10 inch PenTablet Mouse touch" "Button" "1" "button +1 "
xsetwacom set "10 inch PenTablet Mouse touch" "Button" "2" "button +2 "
xsetwacom set "10 inch PenTablet Mouse touch" "Button" "3" "button +3 "
xsetwacom set "10 inch PenTablet Mouse touch" "Button" "8" "button +8 "
xsetwacom set "10 inch PenTablet Mouse touch" "ToolDebugLevel" "0"
xsetwacom set "10 inch PenTablet Mouse touch" "TabletDebugLevel" "0"
xsetwacom set "10 inch PenTablet Mouse touch" "Suppress" "2"
xsetwacom set "10 inch PenTablet Mouse touch" "RawSample" "4"
xsetwacom set "10 inch PenTablet Mouse touch" "PressureCurve" "0 0 100 100"
xsetwacom set "10 inch PenTablet Mouse touch" "Mode" "Absolute"
Property 'Wacom Hover Click' does not exist on device.
xsetwacom set "10 inch PenTablet Mouse touch" "Touch" "on"
Property 'Wacom Hardware Touch Switch' does not exist on device.
xsetwacom set "10 inch PenTablet Mouse touch" "Gesture" "off"
xsetwacom set "10 inch PenTablet Mouse touch" "ZoomDistance" "0"
xsetwacom set "10 inch PenTablet Mouse touch" "ScrollDistance" "0"
xsetwacom set "10 inch PenTablet Mouse touch" "TapTime" "250"
Property 'Wacom Proximity Threshold' does not exist on device.
xsetwacom set "10 inch PenTablet Mouse touch" "Rotate" "none"
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
xsetwacom set "10 inch PenTablet Mouse touch" "Threshold" "0"
xsetwacom set "10 inch PenTablet Mouse touch" "BindToSerial" "0"
Property 'Wacom Pressure Recalibration' does not exist on device.
xsetwacom set "10 inch PenTablet Mouse touch" "PanScrollThreshold" "13"
`
```
Here I found it confusing that the driver is seeing it as an Mouse Touch. Shouldn't it be Tablet? And the line Property Wacom Strip Buttons does not exist on device. Guess this is the part where the **"Not Compatible"** part comes in. Well I tried to configure "X11/xorg.conf.d/70-wacom.conf" with the :
`Section "InputClass"
        Identifier "10 inch PenTablet"
        MatchProduct "28bd:0907"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
        MatchIsTablet "True"
EndSection`

Still nothing changed. If anyone has another Idea pls write down. I'll try some other configurations maybe it'll work. Would be nice if I could use those button tho but anyway Nice Work I appreciate it. :+1:

Saizzou, 2020-10-20:
UPDATE:
I wrote a small bash to look if Linux even detect them. So here is the script:

```
#!/bin/bash

STATE1=$(xinput --query-state "10 inch PenTablet stylus" | grep 'button\[' | sort)
while true; do
    sleep 0.2
    STATE2=$(xinput --query-state "10 inch PenTablet stylus" | grep 'button\[' | sort)
    comm -13 <(echo "$STATE1") <(echo "$STATE2")
    STATE1=$STATE2
done
```
And now I can tell after I used the Script for "10 inch PenTablet Mouse touch" where I saw I have an output for the 2 Buttons mentioned before I dont get the other buttons. 

Well I guess this is the best compatible way as now. Still if you have ideas I would love to try them to. 

And be free to use the Bash if someone thinks its funny or wanna check his/her button function :)

danielytics, 2021-09-24:
he [official linux support](https://www.ugee.com/download-436.html) seems to work well for me for the M708 V2.

Without it, using digimend, I couldn't get all of the buttons to work (I'm using Wayland so xsetwacom and such don't work for me and I haven't been able to figure out the wayland alternative), but with the official drivers, their GUI config tool works perfectly, allowing me to reconfigure all buttons (including the stylus buttons).

teras, 2021-09-27:
@danielytics  Indeed the official drivers work, but my other input methods are destroyed. My mouse is too fast and not accelerated and the screen areas are completely broken (I am having a dual-screen KDE system with 150% HiDPI settings).

The original drivers indeed don't fully work (i.e unable to set custom keys) but maybe it's better the sanity of my mouse instead of some random closed source drivers that don't even provide correct packages but only debian-based. :cry:

nic3-14159, 2021-09-27:
@teras @danielytics @Saizzou Hello all! Could someone run the following script and post the output, and then collect and post raw input samples? The M708 seems to have the same USB ids as the XP-Pen Star 03, and newer XP-Pen tablets seem to require a new initialization procedure to work properly. The script attempts to perform that procedure on a tablet. Essentially, follow the steps on [this page](https://digimend.github.io/support/howto/trbl/diagnostics/), but run my script instead of uclogic-probe.

https://github.com/nic3-14159/uclogic-new-init/blob/base/uclogic_new_init.py
This requires the pyusb module to be installed

danielytics, 2021-09-27:
@teras hmm, seems to work for me on a screen with 3x scaling. But you're right, there are shortcomings (being closed source, having to have the utility running, no commandline way to change config so no way to setup a key binding to switch config for different programs, not being available for all distros...)

@nic3-14159:
```
❯ lsusb  | grep Pen
Bus 003 Device 014: ID 28bd:0924 XP-Pen 
❯ sudo python3 uclogic_new_init.py 28bd:0924
Please hold the pen on the tablet surface while running this program.
Press Enter to continue
Trying init packet 0x02B00200000000000000:
Received 02A1BE945D4A2D120C1E0000
Invalid key!

Trying init packet 0x02B00400000000000000:
Received 02B1040000000000101D0000
Key is 0x02B00400000000000000, interface 2, endpoint 0x03
```

nic3-14159, 2021-09-27:
@danielytics Thanks for posting those. It looks like this tablet does respond to the new initialization procedure, and the dumps appear to be from the tablet in its fully initialized mode. It should be possible now to add proper support for this tablet.

That USB ID is a bit strange, it seems like the M708 V2 should be 28bd:0907? If you disconnect and reconnect your tablet, does it still have that ID? Does the ID change if you run uclogic-probe? I've heard of some cases where uclogic tablets change their USB IDs after being probed or unplugged/replugged.

danielytics, 2021-09-27:
I don’t have it with me now, I’ll check when I get back home on Wednesday if nobody else does it before then.

teras, 2021-09-28:
I can't test it thoroughly right now either, but I will. I did some preliminary tests and indeed the USB events come and go :)
So it seems that this is more a configuration issue instead of support.

teras, 2021-10-01:
@nic3-14159 
Here it is (run with sudo)
Followed by the dump.
Also, the last 16 events, are when clicking on the 8 buttons, from top to bottom.

```
# lsusb | grep Pen
Bus 001 Device 019: ID 28bd:0907 XP-Pen 10 inch PenTablet

# ./uclogic_new_init.py 28bd:0907
Please hold the pen on the tablet surface while running this program.
Press Enter to continue
Trying init packet 0x02B00200000000000000:
Received 02B10200000000000000
Invalid key!

Trying init packet 0x02B00400000000000000:
Received 02B10400000000000000
Key is 0x02B00400000000000000, interface 2, endpoint 0x03

# sudo uclogic-probe 1 19
M
P 31 00 30 00 20 00 69 00 6E 00 63 00 68 00 20 00 50 00 65 00 6E 00 54 00 61 00 62 00 6C 00 65 00 74 00
S 64 0C 03 6A C6 0A 77 08 00 FF 1F D8 13
S 65 04 03 01 67
S 6E 07 03 00 10 00 01 08
S 79 10 03 20 20 20 20 20 20 20 20 20 20 20 20 20 20
S 7A 08 03 00 00 00 00 00 00
S 7B 0C 03 48 00 4B 00 20 00 4F 00 6E 00
Failed to get string descriptor 0xC8: Input/Output Error
```

[dump.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/7270451/dump.txt)

itsKia2, 2021-12-29:
Have been using the driver from the following link: https://github.com/ezhang-dev/hid_m708
Have used it for more than a full year. Still does not give any errors and works perfectly fine. Highly recommended to look into this. 
I use the following script (with an alias) to load the module and map my keys everytime I reconnect the tablet:
```
#! bin/bash

sudo modprobe -r hid_m708
sudo modprobe hid_m708
sleep 1
sh /home/kia/.scripts/tabletmaps.sh
echo "Tablet reconnected"
```

MikeWalrus, 2022-03-29:
> @danielytics Thanks for posting those. It looks like this tablet does respond to the new initialization procedure, and the dumps appear to be from the tablet in its fully initialized mode. It should [...]

@nic3-14159 
My tablet also shows up as 28bd:0924, and UGEE still calls it M708. Maybe this is a newer version of the product?