DIGImend/digimend-kernel-drivers #470: Dial on Q620M doesn't work

ticotaco72, 2020-10-22:
I installed Digimend 10 release on Fedora KDE 32; Kernel 5.8.15;
Dial worked on master branch some three-four months ago, but now it doesn't work in any program. I don't know how to check if system gets info about dial movement. I'm happy to provide anything that's needed.

nzrt-pm, 2020-11-13:
same thing, dial doesn't work for me as well

nic3-14159, 2020-11-15:
Hello! Could you try following the instructions on [this](https://digimend.github.io/support/howto/trbl/locating_failure/) page? That page should contain steps that can help debug what is wrong with the dial. The steps under "Tablet Hardware" should tell you if the hardware is working at all (although with two separate similar failures, I'd think it's more of a Linux issue than a hardware issue). The "Debugfs interface" and "Evtest tool" sections would probably be more useful, as those steps would tell you if Linux actually recognizes the dial events properly.

nzrt-pm, 2020-11-16:
update. have tested it with a couple of distros and only kubuntu works for me via sh script(but I can't change the wheel controls it's fixed as a mouse scroll wheel), also I had to reinstall my endeavouros root. the wheel works but it doesn't output it correctly it my mouse input (left mid and right mouse button doesn't work whenever I try to use the tablet's wheel and I have to restart the pc for it to return to normal) tablet works perfectly with windows.

kxr, 2020-12-13:
Hello, I just got this tablet and its a pretty slick tablet. I plugged it in my Fedora workstation (5.9.13-100.fc32.x86_64) and the basic functionality seems to work out of the box. My use case is just basic writing. What don't seem to work is the dial and the pressure sensor. I would love to have it fully supported in Linux. How can I contribute to get the missing feature working?

nzrt-pm, 2020-12-14:
update, after fiddling with it, I managed to make the buttons work, except for the dial wheel(dial buttons also work) testing the wheel with 

sudo usbhid-dump -es -m 256c:006d

 works, but nothing that I've tried setting it up in xinput worked.

here's the script I've made 

``xsetwacom --set 'HUION Huion tablet_Q620M Pad pad' Button 13 "key ctrl +q"
xsetwacom --set 'HUION Huion tablet_Q620M Pad pad' Button 1 "key ctrl +s"
xsetwacom --set 'HUION Huion tablet_Q620M Pad pad' Button 3 "key P" 
xsetwacom --set 'HUION Huion tablet_Q620M Pad pad' Button 2 "key B"
xsetwacom --set 'HUION Huion tablet_Q620M Pad pad' Button 8 "key ctrl ["
xsetwacom --set 'HUION Huion tablet_Q620M Pad pad' Button 9 "key ctrl ]"
xsetwacom --set 'HUION Huion tablet_Q620M Pad pad' Button 10 "key e"
xsetwacom --set 'HUION Huion tablet_Q620M Pad pad' Button 11 "key m"
xsetwacom --set 'HUION Huion tablet_Q620M Pad pad' Button 12 "key ctrl +z"
``

thezealousfool, 2020-12-26:
I got this tablet as well.

Pressure sensitivity works on Krita but does not recognize as a tablet/stylus in GIMP or Firefox (tested on https://patrickhlauke.github.io/touch/pen-tracker/).

I'm assuming somehow Qt apps work? Can people running KDE test GIMP? Does the tablet show up in the KDE settings?

Happy to contribute but am a little clueless how to proceed.

NeoTheFox, 2021-02-11:
For everyone tracking this issue - Q620M dial and buttons work with [huion-keys](https://github.com/erjiang/huion-keys).

ticotaco72, 2021-02-14:
I'm sorry that I've been so inactive after reporting this issue. All tesing done on Fedora 33 KDE

1. In `evtest` I have 5 devices:
```
/dev/input/event20:     HUION Huion Tablet_Q620M                       -- here I saw movement and pressure and tilt of pen and pen buttons
/dev/input/event21:     HUION Huion Tablet_Q620M Pad                 --here I saw events from frame buttons and inner button in dial
/dev/input/event22:     HUION Huion Tablet_Q620M Touch Strip      -- here is nothing
/dev/input/event23:     HUION Huion Tablet_Q620M Dial                  -- here I saw events after I moved dial and values seemed simply left/right
/dev/input/event24:     HUION Huion Tablet_Q620M Keyboard        --here is nothing, it shows that it has all event codes which keyboard normally has??
```
2. Left frame button have strange bindings. They're mostly binded to left mouse button, right mouse button, space button and these bindings are repeqated(two buttons corresponding e.g. to space button)
3. In Krita: pressure works, tilt doesn't work, for stylus buutons to work as espected I need to tick the "listen to mouse feed" setting
4. KDE settings detect tablet only as a game pad where I see only frame buttons as working

ticotaco72, 2021-02-14:
5. GIMP detects it only as a mouse, no tablet specific things work

NeoTheFox, 2021-02-16:
@ticotaco72 please use `libinput debug-tablet` to confirm if features of the tablet itself work. If you see it reacting to tilt, pressure, button presses via libinput debug - the "not working" part is in the software you are using. For example, for GIMP - check if you have the pressure enabled in input device settings like this.
[image]

I suggest you file a bug report with the respective projects if you think that features are missing.

ticotaco72, 2023-11-23:
libinput in version 1.23.0 on kernel version 6.5.11 Fedora 38 reports:
- that there is Huion Touch Strip, Dial, Pen, Pad and Keyboard
- Touch Strip and Dial are ignored as 
```
event19 - HUION Huion tablet_Q620M Touch Strip: libinput bug: missing tablet capabilities: pen btn-stylus resolution. Ignoring this device.
event20 - HUION Huion tablet_Q620M Dial: libinput bug: missing tablet capabilities: pen btn-stylus resolution. Ignoring this device.
```
Dial emits events in evtest and Touch Strip doesn't

- Keyboard device is always silent (also in evtest) and there are some buttons more than needed in Pad
- libinput debug-tablet always reports 0.00 on dist, rotation and slider, while in evdev section 0.00 is always reported for ABS_Z and ABS_DISTANCE

I suppose a big deal of the issues (at least the ignoring devices issue?) could be fixed at libinput side?

ticotaco72, 2023-11-24:
for reference:
https://gitlab.freedesktop.org/libinput/libinput/-/issues/600