DIGImend/digimend-kernel-drivers #471: Problem with Huion 1060P and xournal++

ajgausa, 2020-10-25:
Hello everyone,

I recently purchased a Huion 1060P and I'm experiencing a strange issue. I would really appreciate any help.

I'm using Linux Mint 20, kernel 5.4.0-52-generic.

The tablet works out of the box without installing the digimend driver. In particular, it works very well with both inkscape and xournal++. However, it is not recognized by xsetwacom, so I cannot configure any of the buttons in the pad. For this reason, after rrmoding the hid-uclogic, I installed the digimend-dkms_10_all.deb package (for some reason, the v9 and v8 versions do not compile, the sys.h header is not found).

After installing the digimend driver and rebooting, the tablet works as a mouse in the desktop and it is detected by xsetwatcom, The output of xsetwatcom --list is

HID 256c:006d Touch Strip pad   	id: 13	type: PAD       
HID 256c:006d stylus            	        id: 17	type: STYLUS    
HID 256c:006d Pad pad           	id: 18	type: PAD 

The problem is that when I try to use xournal++, whenever I touch the pad with the pen with the pen tool selected, the whole document is dragged around, rather that getting anything drawn. Inkscape, however, works exactly as before, so it is very likely that this is not an issue with the driver.

The output of  xsetwacom --get 'HID 256c:006d stylus' Button 1 is button +1, but I don't really know what that means.

As I said, any help would be really appreciated.

Best regards,

Angel

ajgausa, 2020-10-27:
I'll answer myself: it was indeed a problem with Xournal++. It detected the stylus as a mouse and acted accordingly. Setting the stylus to 'pen' in the preferences menu makes the tablet works beautifully.

juliapacula, 2020-12-05:
Hello, I have the same issue as you do with stylus. I tried setting stylus to work as a Pen, but unfortunately there are no results. What were your steps to achieve desired results?

pbettens, 2021-01-21:
some problem with H950P. 

```
>$ xsetwacom list
HUION Huion Tablet_H950P Touch Strip pad        id: 10  type: PAD       
HUION Huion Tablet_H950P stylus         id: 11  type: STYLUS    
HUION Huion Tablet_H950P Pad pad        id: 12  type: PAD  
```

with xournal++ document is dragged around. I have tried to modifx preferences mouse → pen → mouse but without results. the tablet work fine with inkscape and gimp… and xournal ;-) 

best regards.

auralius, 2021-03-04:
I have the same issue as pbettens.  I use H950 P as well. Now I return to Xournal.

emnik, 2021-04-10:
Try setting the input as the screenshot. This has worked for me!

[image]

al-jshen, 2021-04-26:
Had the same issue, and changing from "mouse" to "pen" did not solve the issue for me. Enabling the experimental input system did. Now it works with either "mouse" or "pen" for me.

wyatt8740, 2023-01-07:
I should note, this is actually a GTK+3 issue; the stylus name as seen by xinput doesn't contain the word 'pen' or 'wacom', so GDK (part of GTK) assumes that the stylus is a touch screen device.

This impacts all GTK3 programs that don't have special workarounds like Xournal++ does.

[This patch](https://github.com/wyatt8740/gtk3-classic/blob/master/fixes__stylus-input-devices-are-pens.patch) can be applied to gtk3's source code to fix it at its source.