DIGImend/digimend-kernel-drivers #474: XP-Pen Innovator 16 diagnostics - ID160F

razcore-art, 2020-11-02:
Recently I got an Innovator 16 so I'm attaching the diagnostics I got by following [the instructions](http://digimend.github.io/support/howto/trbl/diagnostics/). Hope it helps, if I can do anything else let me know.

Hope I collected correct data.

[diagnostics.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/5478254/diagnostics.zip)

[XP-Pen Innovator 16 presentation](https://www.xp-pen.com/product/638.html)
[XP-Pen Innovator 16 drivers](https://www.xp-pen.com/download-523.html)

---

For some reason `lsusb` returns XP-Pen with my headphones name (Jabra Revo) but I assume that doesn't matter too much.

pivui, 2020-11-16:
If my experience is of any help:

I have just received this tablet and it seems to work fine with the latest digimend release (I am on xubuntu 20.04): pressure, tilt, buttons and wheels.

It was not directly detected by `xsetwacom --list`, but adding a configuration file as 
```
Section "InputClass"
       Identifier "UGTABLET 15.6 inch PenDisplay"
       MatchUSBID "28bd:092c"
       MatchDevicePath "/dev/input/event*"
       Driver "wacom"
EndSection
```
in `/etc/X11/xorg.conf.d/` worked for me.

However, I have trouble mapping the buttons the way I want with the wacom driver so I ended up switching back to evdev driver and with a custom configuration file in `/etc/udev/hwdb.d/` I am mapping the scan-codes to some key-codes I want. Here is my configuration file:

```
evdev:input:b0003v28BDp092Ce0100*
 KEYBOARD_KEY_70005=a
 KEYBOARD_KEY_70008=b
 KEYBOARD_KEY_700e2=c
 KEYBOARD_KEY_7002c=d 
 KEYBOARD_KEY_70019=e # Buttons after wheel
 KEYBOARD_KEY_70016=f
 KEYBOARD_KEY_7001d=g
 KEYBOARD_KEY_70011=h # the last button send 3 scan-code, one of them beeing 700e2, as above..
 KEYBOARD_KEY_700e0=unknown # second button starting from bottom send two scancodes, disabling the first one
 KEYBOARD_KEY_D0044=space # Pen tip button
 KEYBOARD_KEY_D0045=0x111 # Pen other button

```
The only issue I have is with the last button that seems to send several scan-codes at once, one of them being the same as the scan-code sent by the third button...

razcore-art, 2020-11-16:
Thanks for the info. I'm on Ubuntu 20.10 and the tablet seemed to be working with digimend drivers like you said. I didn't know how to configure the buttons and that's why I ended up using the proprietary drivers script, but I'll try your suggestion.

razcore-art, 2020-11-17:
Yes this works. I don't care about the frame buttons, just the stylus so I have this:

```
evdev:input:b0003v28BDp092Ce0100*
 KEYBOARD_KEY_D0044=btn_middle # Pen 1st button
 KEYBOARD_KEY_D0045=btn_right  # Pen 2nd button
```

in `/etc/udev/hwdb.d/99-xp-pen-id160f.hwdb`

For anyone looking, note that this needs to be compiled and reset/triggered with:

```bash
systemd-hwdb update
udevadm control --reload
udevadm trigger
```

pivui, 2020-11-26:
For those who are still interested in remapping the buttons:

I could not find a way to remap the buttons the way I wanted with only a .hwdb config file as above due to some buttons launching multiple scancodes + the need to use something else (e.g. xbindkeys) to remap to some action such as ctrl + s. 

So I have written a python script that filters out the events coming of the buttons/inner wheel (considered as a separate device, a mouse) and replaces them by (user-defined) events sent by a virtual keyboard. 

The code is available [here](https://github.com/pivui/ButtonMapping) but beware that you may need to tweak things to your needs as it is not extremely flexible. 
A better solution would probably to work with existing packages such as [this one](https://github.com/philipl/evdevremapkeys).

Note that as the buttons on the pen are associated with the display device, this does not allow to remap the buttons on the pen (but the other method mentioned above does). I have tried to grab all the devices related to the tablet and re-throw anything that is not a button but it induced some lag with the tilt of the pen so I've given up on that.