DIGImend/digimend-kernel-drivers #476: xp-pen artist 22 pro

Dr-Kraven, 2020-11-04:
I'm not very acquainted with how these drivers work, as I've only just begun programming, but I've downloaded and installed the ubuntu version on my Linux Mint 19.3 Desktop and my display still doesn't respond to my pens. Is there a way to boot up a gui? I have a second display, so I'll have to map the pen-touch to the correct monitor once it does work, if I can get it to work.

Thank you for your time,
Dr. K

ebarranco, 2020-11-05:
hi!  Artist 22 Pro has oficial linux drivers on the oficial xp-pen website. Digimend driver uses the same GUI use for wacom tablets.

linuxgirl22, 2020-11-06:
Hi, just to suggest ptxconf, for mapping to 2nd display. I'm linking to the AUR's version, only because it has a click-through link and also shows dependencies needed. I don't use Mint, so am not sure about installation of dependencies and ptxconf itself, or if it'll ultimately run, but, if you ask on the Mint forum, people should be able to help you there. Hope it works for you. 

https://aur.archlinux.org/packages/ptxconf-git/

nic3-14159, 2020-11-15:
Hello! As far as I can tell, Digimend does not yet support the Artist 22 Pro. Would you be able to post diagnostics for your tablet as outlined on [this page?](https://digimend.github.io/support/howto/trbl/diagnostics/) That would be helpful for that tablet to be supported by Digimend.