DIGImend/digimend-kernel-drivers #480: Debian install instructions (dkms dependency?)

uvtc, 2020-11-11:
I just got a Huion H610 Pro v2. Using it with Debian Testing. I think it's working, but am not sure I took the correct steps --- please see below.

My Debian system:

~~~
uname -a
Linux tarragon 5.9.0-1-amd64 #1 SMP Debian 5.9.1-1 (2020-10-17) x86_64 GNU/Linux
~~~

The install instructions here, in the readme, say to just install the digimend-dkms_10_all.deb package, but it turned out that I also needed to install the dkms package.

When I first tried to just install the .deb, I got:

~~~
# dpkg -i digimend-dkms_10_all.deb 
Selecting previously unselected package digimend-dkms.
(Reading database ... 195192 files and directories currently installed.)
Preparing to unpack digimend-dkms_10_all.deb ...
Unpacking digimend-dkms (10) ...
dpkg: dependency problems prevent configuration of digimend-dkms:
 digimend-dkms depends on dkms (>= 2.1.0.0); however:
  Package dkms is not installed.

dpkg: error processing package digimend-dkms (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 digimend-dkms
~~~

Wasn't sure how to proceed, and did not want to install from source, but look in the readme regarding installing from source and noticed the dkms package mentioned. Searched for that, took a chance and installed it:

~~~
# apt install dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Suggested packages:
  menu
Recommended packages:
  linux-headers-686-pae | linux-headers-amd64 | linux-headers-generic | linux-headers
The following NEW packages will be installed:
  dkms
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 77.3 kB of archives.
After this operation, 298 kB of additional disk space will be used.
Get:1 http://ftp.us.debian.org/debian testing/main amd64 dkms all 2.8.3-4 [77.3 kB]
Fetched 77.3 kB in 0s (546 kB/s)
Selecting previously unselected package dkms.
(Reading database ... 195215 files and directories currently installed.)
Preparing to unpack .../archives/dkms_2.8.3-4_all.deb ...
Unpacking dkms (2.8.3-4) ...
Setting up dkms (2.8.3-4) ...
Setting up digimend-dkms (10) ...
Loading new digimend-10 DKMS files...
Building for 5.9.0-1-amd64
Module build for kernel 5.9.0-1-amd64 was skipped since the
kernel headers for this kernel does not seem to be installed.
Processing triggers for man-db (2.9.3-2) ...
~~~

Now I see that both packages appear to be installed:

~~~
$ dpkg -l | grep dkms
ii  digimend-dkms                         10                              all          Collection of graphics tablet drivers by DIGImend project
ii  dkms                                  2.8.3-4                         all          Dynamic Kernel Module Support Framework
~~~

and when I plug in the tablet, it seems to work! (Woot!)

Am I missing part of the install? Was I supposed to install some kernel headers? I update/upgrade my system periodically and it upgrades the kernel every so often.

Should the readme here be updated to say something like, "on Debian, make sure you have the dkms package installed before installing the digimend-dkms package"?

Incidentally, my desktop is xfce, and the "Settings --> Mouse and Touchpad" lets me select the Huion device to see its settings.

uvtc, 2020-11-11:
Interestingly, on another Debian Testing system which I thought was identical, I get an installation which pulls in those kernel headers:

~~~
# apt install dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  linux-compiler-gcc-10-x86 linux-headers-5.9.0-1-amd64 linux-headers-5.9.0-1-common linux-headers-amd64 linux-kbuild-5.9
Suggested packages:
  menu
The following NEW packages will be installed:
  dkms linux-compiler-gcc-10-x86 linux-headers-5.9.0-1-amd64 linux-headers-5.9.0-1-common linux-headers-amd64 linux-kbuild-5.9
0 upgraded, 6 newly installed, 0 to remove and 0 not upgraded.
Need to get 10.0 MB of archives.
After this operation, 59.6 MB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://ftp.us.debian.org/debian testing/main amd64 dkms all 2.8.3-4 [77.3 kB]
Get:2 http://ftp.us.debian.org/debian testing/main amd64 linux-compiler-gcc-10-x86 amd64 5.9.1-1 [181 kB]
Get:3 http://ftp.us.debian.org/debian testing/main amd64 linux-headers-5.9.0-1-common all 5.9.1-1 [8,646 kB]
Get:4 http://ftp.us.debian.org/debian testing/main amd64 linux-kbuild-5.9 amd64 5.9.1-1 [424 kB]
Get:5 http://ftp.us.debian.org/debian testing/main amd64 linux-headers-5.9.0-1-amd64 amd64 5.9.1-1 [687 kB]
Get:6 http://ftp.us.debian.org/debian testing/main amd64 linux-headers-amd64 amd64 5.9.1-1 [1,176 B]
Fetched 10.0 MB in 3s (3,946 kB/s)              
Selecting previously unselected package dkms.
(Reading database ... 169170 files and directories currently installed.)
Preparing to unpack .../0-dkms_2.8.3-4_all.deb ...
Unpacking dkms (2.8.3-4) ...
Selecting previously unselected package linux-compiler-gcc-10-x86.
Preparing to unpack .../1-linux-compiler-gcc-10-x86_5.9.1-1_amd64.deb ...
Unpacking linux-compiler-gcc-10-x86 (5.9.1-1) ...
Selecting previously unselected package linux-headers-5.9.0-1-common.
Preparing to unpack .../2-linux-headers-5.9.0-1-common_5.9.1-1_all.deb ...
Unpacking linux-headers-5.9.0-1-common (5.9.1-1) ...
Selecting previously unselected package linux-kbuild-5.9.
Preparing to unpack .../3-linux-kbuild-5.9_5.9.1-1_amd64.deb ...
Unpacking linux-kbuild-5.9 (5.9.1-1) ...
Selecting previously unselected package linux-headers-5.9.0-1-amd64.
Preparing to unpack .../4-linux-headers-5.9.0-1-amd64_5.9.1-1_amd64.deb ...
Unpacking linux-headers-5.9.0-1-amd64 (5.9.1-1) ...
Selecting previously unselected package linux-headers-amd64.
Preparing to unpack .../5-linux-headers-amd64_5.9.1-1_amd64.deb ...
Unpacking linux-headers-amd64 (5.9.1-1) ...
Setting up linux-compiler-gcc-10-x86 (5.9.1-1) ...
Setting up linux-kbuild-5.9 (5.9.1-1) ...
Setting up dkms (2.8.3-4) ...
Setting up linux-headers-5.9.0-1-common (5.9.1-1) ...
Setting up linux-headers-5.9.0-1-amd64 (5.9.1-1) ...
/etc/kernel/header_postinst.d/dkms:
dkms: running auto installation service for kernel 5.9.0-1-amd64:.
Setting up linux-headers-amd64 (5.9.1-1) ...
Processing triggers for man-db (2.9.3-2) ...

# dpkg -i digimend-dkms_10_all.deb 
Selecting previously unselected package digimend-dkms.
(Reading database ... 187375 files and directories currently installed.)
Preparing to unpack digimend-dkms_10_all.deb ...
Unpacking digimend-dkms (10) ...
Setting up digimend-dkms (10) ...
Loading new digimend-10 DKMS files...
Building for 5.9.0-1-amd64
Building initial module for 5.9.0-1-amd64
Done.

hid-kye.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/5.9.0-1-amd64/updates/dkms/

hid-uclogic.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/5.9.0-1-amd64/updates/dkms/

hid-polostar.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/5.9.0-1-amd64/updates/dkms/

hid-viewsonic.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/5.9.0-1-amd64/updates/dkms/

depmod.........

DKMS: install completed.
~~~

nic3-14159, 2020-11-12:
I'm fairly certain you need the kernel headers package in order to install digimend. I suspect that your tablet shows up in the settings because it is recognized by the digimend code that is merged into the kernel, and not the debian package you installed. In other words, I think it did not actually install the digimend deb package properly.

What does the command `modinfo hid-uclogic` output? That would indicate which version of the driver is being used when you plug in your tablet.

uvtc, 2020-11-17:
On machine-1 (on machine from my first post):

~~~
filename:       /lib/modules/5.9.0-1-amd64/kernel/drivers/hid/hid-uclogic.ko
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
alias:          {snip} {snip} {snip} ...
depends:        hid,usbcore,usbhid
retpoline:      Y
intree:         Y
name:           hid_uclogic
vermagic:       5.9.0-1-amd64 SMP mod_unload modversions 
sig_id:         PKCS#7
signer:         Debian Secure Boot CA
sig_key:        B5:5E:B3:B9
sig_hashalgo:   sha256
signature:      {snip}
~~~

Same output whether the tablet is plugged in or not.

uvtc, 2020-11-17:
From machine-1, from `lsusb`

~~~
Bus 002 Device 007: ID 256c:006d HUION Huion Tablet
~~~

BTW, sorry for the delay in replying here.

uvtc, 2020-11-17:
On machine-2, from `/sbin/modinfo hid-uclogic` I get:

~~~
filename:       /lib/modules/5.9.0-1-amd64/updates/dkms/hid-uclogic.ko
version:        10
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     57AA0864931B0963BD47BF3
alias:          {snip} {snip} {snip}...
depends:        hid,usbcore,usbhid
retpoline:      Y
name:           hid_uclogic
vermagic:       5.9.0-1-amd64 SMP mod_unload modversions 
~~~

and lsusb shows:

~~~
Bus 002 Device 003: ID 256c:006d HUION Huion Tablet
~~~

nic3-14159, 2020-11-17:
Based on the output for the modprobe command, your 1st machine is using the driver in the kernel, which means Digimend did not install properly. If you install the kernel headers, then uninstall and reinstall the deb package, it should be properly installed on your first machine. Also make sure to run `modprobe -r hid-uclogic` after installing to make sure linux loads the installed version. The output from your 2nd machine does look like the digimend package installed correctly.

I think the readme does not mention dkms or kernel headers for the deb package because the metadata inside the package indicates that it depends on dkms, which causes dpkg to raise an error if dkms is not found. That's what the "dependency problems" error indicates. The linux headers dependency is probably intended to be handled by the dkms package, which strangely wasn't installed when you installed dkms.

uvtc, 2020-11-18:
Thank you, Nicholas. I did as you said, and the output of the modprobe command on machine-1 now looks like it does on machine-2.

I think it would be useful to users if, in the readme, under "Installing --> Installing Debian Package", it also said:

"Prerequisites: If you don't already have them, apt install the `dkms` and `linux-headers-amd64` packages."