DIGImend/digimend-kernel-drivers #482: Artisul D16 Pro Diagnostics

WanderingFeral, 2020-11-18:
Tablet manufacturer: Artisul (looks like a UC-Logic re-brand)
Model name: D15 Pro
Purchased from: https://www.artisul.com/store/55-26-D16-Pro.html#/29-warehouse-eu

Diagnostics:

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5560784/descriptors.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5560785/frame_buttons.txt)
[frame_dials.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5560786/frame_dials.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5560787/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5560788/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5560789/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5560791/pen_pressure.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5560792/pen_tilt.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5560793/probe.txt)

The stylus works (just have to map it to the built-in screen) but the buttons don't. Is there any way to get them working temporarily?

WanderingFeral, 2020-11-21:
I have been doing some digging around and have found the following:

/sys/kernel/debug/hid/0003:256C:006D.000E/events (Dial and button) returns:

```
report (size 9) (numbered) =  10 00 32 00 00 00 00 00 00
Button.0001 = 0
Digitizers.Touch = 0
GenericDesktop.Dial = 50
GenericDesktop.X = 0
GenericDesktop.Y = 0
Digitizers.Width = 0

report (size 9) (numbered) =  10 00 32 00 00 00 00 00 00
Button.0001 = 0
Digitizers.Touch = 0
GenericDesktop.Dial = 50
GenericDesktop.X = 0
GenericDesktop.Y = 0
Digitizers.Width = 0

report (size 9) (numbered) =  10 00 32 00 00 00 00 00 00
Button.0001 = 0
Digitizers.Touch = 0
GenericDesktop.Dial = 50
GenericDesktop.X = 0
GenericDesktop.Y = 0
Digitizers.Width = 0

report (size 9) (numbered) =  10 00 32 00 00 00 00 00 00
Button.0001 = 0
Digitizers.Touch = 0
GenericDesktop.Dial = 50
GenericDesktop.X = 0
GenericDesktop.Y = 0
Digitizers.Width = 0

report (size 9) (numbered) =  10 00 32 00 00 00 00 00 00
Button.0001 = 0
Digitizers.Touch = 0
GenericDesktop.Dial = 50
GenericDesktop.X = 0
GenericDesktop.Y = 0
Digitizers.Width = 0

report (size 9) (numbered) =  10 00 32 00 00 00 00 00 00
Button.0001 = 0
Digitizers.Touch = 0
GenericDesktop.Dial = 50
GenericDesktop.X = 0
GenericDesktop.Y = 0
Digitizers.Width = 0

report (size 9) (numbered) =  10 00 32 00 00 00 00 00 00
Button.0001 = 0
Digitizers.Touch = 0
GenericDesktop.Dial = 50
GenericDesktop.X = 0
GenericDesktop.Y = 0
Digitizers.Width = 0

report (size 9) (numbered) =  10 00 32 00 00 00 00 00 00
Button.0001 = 0
Digitizers.Touch = 0
GenericDesktop.Dial = 50
GenericDesktop.X = 0
GenericDesktop.Y = 0
Digitizers.Width = 0

report (size 9) (numbered) =  10 00 ce ff 00 00 00 00 00
[... 448 lines cut ...]
Button.0007 = 1
Button.0008 = 0
Button.0009 = 0
Button.000a = 0
Button.0001 = 0
Button.0002 = 0

report (size 12) (numbered) =  08 e0 01 01 00 00 00 00 00 00 00 00
Digitizers.BarrelSwitch = 0
GenericDesktop.X = 0
GenericDesktop.Y = 0
Button.0001 = 0
Button.0002 = 0
Button.0003 = 0
Button.0004 = 0
Button.0005 = 0
Button.0006 = 0
Button.0007 = 0
Button.0008 = 0
Button.0009 = 0
Button.000a = 0
Button.0001 = 0
Button.0002 = 0

```

WanderingFeral, 2020-11-21:
libinput record (/dev/input/event16:	UC-Logic ARTISUL D16Pro Pad) returns:

```
Select the device event number: 16
Recording to 'stdout'.
version: 1
ndevices: 1
libinput:
  version: "1.16.3"
  git: "unknown"
system:
  kernel: "5.9.9-arch1-1"
  dmi: "dmi:bvnInsydeCorp.:bvrV1.12:bd08/02/2016:br0.0:efr2.10:svnAcer:pnAspireE5-575:pvrV1.12:rvnAcer:rnIronman_SK:rvrV1.12:cvnChassisManufacturer:ct10:cvrChassisVersion:"
devices:
- node: /dev/input/event16
  evdev:
    # Name: UC-Logic ARTISUL D16Pro Pad
    # ID: bus 0x3 vendor 0x256c product 0x6d version 0x111
    # Size in mm: unknown, missing resolution
    # Supported Events:
    # Event type 0 (EV_SYN)
    # Event type 1 (EV_KEY)
    #   Event code 256 (BTN_0)
    #   Event code 257 (BTN_1)
    #   Event code 258 (BTN_2)
    #   Event code 259 (BTN_3)
    #   Event code 260 (BTN_4)
    #   Event code 261 (BTN_5)
    #   Event code 262 (BTN_6)
    #   Event code 263 (BTN_7)
    #   Event code 264 (BTN_8)
    #   Event code 265 (BTN_9)
    #   Event code 304 (BTN_SOUTH)
    #   Event code 305 (BTN_EAST)
    #   Event code 331 (BTN_STYLUS)
    # Event type 3 (EV_ABS)
    #   Event code 0 (ABS_X)
    #       Value           0
    #       Min             0
    #       Max             1
    #       Fuzz            0
    #       Flat            0
    #       Resolution      0
    #   Event code 1 (ABS_Y)
    #       Value           0
    #       Min             0
    #       Max             1
    #       Fuzz            0
    #       Flat            0
    #       Resolution      0
    # Event type 4 (EV_MSC)
    #   Event code 4 (MSC_SCAN)
    # Properties:
    name: "UC-Logic ARTISUL D16Pro Pad"
    id: [3, 9580, 109, 273]
    codes:
      0: [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15] # EV_SYN
      1: [256, 257, 258, 259, 260, 261, 262, 263, 264, 265, 304, 305, 331] # EV_KEY
      3: [0, 1] # EV_ABS
      4: [4] # EV_MSC
    absinfo:
      0: [0, 1, 0, 0, 0]
      1: [0, 1, 0, 0, 0]
    properties: []
  hid: [5, 13, 9, 2, 161, 1, 133, 8, 9, 32, 160, 20, 37, 1, 117, 1, 9, 66, 9, 68, 9, 70, 149, 3, 129, 2, 149, 3, 129, 3, 9, 50, 149, 1, 129, 2, 149, 1, 129, 3, 149, 1, 164, 5, 1, 101, 19, 85, 253, 117, 24, 52, 9, 48, 39, 226, 12, 1, 0, 71, 238, 52, 0, 0, 129, 2, 9, 49, 39, 58, 151, 0, 0, 71, 196, 29, 0, 0, 129, 2, 180, 9, 48, 117, 16, 39, 255, 31, 0, 0, 129, 2, 129, 3, 192, 192, 5, 1, 9, 7, 161, 1, 133, 247, 20, 37, 1, 117, 1, 5, 13, 9, 57, 160, 9, 68, 149, 1, 129, 2, 5, 1, 9, 48, 9, 49, 149, 2, 129, 2, 149, 21, 129, 1, 5, 9, 25, 1, 41, 10, 149, 10, 129, 2, 192, 5, 1, 9, 5, 160, 5, 9, 25, 1, 41, 2, 149, 2, 129, 2, 149, 52, 129, 1, 192, 192 ]
  udev:
    properties:
    - ID_INPUT=1
    - ID_INPUT_TABLET=1
    - ID_INPUT_TABLET_PAD=1
[... 110 lines cut ...]
  - evdev:
    - [  3, 311788,   2,   7,     -50] # EV_REL / REL_DIAL                -50
    - [  3, 311788,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +1644ms
  - evdev:
    - [  3, 347868,   2,   7,     -50] # EV_REL / REL_DIAL                -50
    - [  3, 347868,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +36ms
  - evdev:
    - [  3, 471979,   2,   7,     -50] # EV_REL / REL_DIAL                -50
    - [  3, 471979,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +124ms
  - evdev:
    - [  3, 535955,   2,   7,     -50] # EV_REL / REL_DIAL                -50
    - [  3, 535955,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +64ms
  - evdev:
    - [  3, 704004,   2,   7,     -50] # EV_REL / REL_DIAL                -50
    - [  3, 704004,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +169ms
  - evdev:
    - [  4, 159842,   4,   4,      30] # EV_MSC / MSC_SCAN                 30 (obfuscated)
    - [  4, 159842,   1, 256,       1] # EV_KEY / BTN_0                     1
    - [  4, 159842,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +455ms
  - evdev:
    - [  4, 299953,   4,   4,      30] # EV_MSC / MSC_SCAN                 30 (obfuscated)
    - [  4, 299953,   1, 256,       0] # EV_KEY / BTN_0                     0
    - [  4, 299953,   0,   0,       0] # ------------ SYN_REPORT (0) ---------- +140ms

```

phelipefox, 2021-03-12:
Hi! I don't know how much you understand programming, but maybe this driver [Fork](https://github.com/beyondoblivion/digimend-kernel-drivers) might be useful for you. This Artisul D16 Pro has 1 more shortcut button than the Artisul D16. Maybe you can get something!