DIGImend/digimend-kernel-drivers #484: XP-Pen Artist 22R Pro

Draghi, 2020-11-23:
Saw an issue for the 22 [here](https://github.com/DIGImend/digimend-kernel-drivers/issues/476) but I wasn't sure which version of the display that was about (I imagine that's the cause of the multiple IDs in this [issue](https://github.com/DIGImend/digimend-kernel-drivers/issues/424)).

I've attached the diagnostics for the 22R here. First time running them through and the screen is mildly slick, so let me know if I've made a blunder anywhere and I'll work through them or if you need extra information.

Product Page: https://www.xp-pen.com/product/541.html

Just some misc info that may/may-not be useful:
- The pen is recognized as a mouse by linux (Arch / Ubuntu Focal Fossa). Pressing registers a left click, lower button acts like middle mouse (but only in gimp, not other apps) and the upper button (right click) doesn't appear to work. The pen maps the coordinates over multiple and pressure/tilt both don't work.
- A "[driver](https://www.xp-pen.com/download-483.html)" exists, but it's an extremely inconvenient GUI app that can't be closed or run as a daemon. It contains some XML files with mapping information, if that's of use. It contains information for most of XP-Pen's line.
 - I can't seem to tell which corresponds to the 22R, but 22 is mentioned in a couple spots and the "driver" does work with linux, though I could never get the tilt working through it. Supposedly the display supports 60 degrees of tilt.
- Various keys send default key combos (the right-dial sends "[" and "]", the left one acts like a scrollwheel with ctrl held etc.)

[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5583231/probe.txt)

[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5583232/pen_tilt.txt)

[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5583233/pen_pressure.txt)

[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5583235/pen_coords.txt)

[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5583236/pen_buttons.txt)

[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5583237/hid_report_descriptors.txt)

[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5583240/frame_buttons.txt)

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5583241/descriptors.txt)

There are two dial files since there are two dials on the screen, which are independent from one-another, "_l" is the left dial and "_r" is the right dial. They have no modes.

[frame_dials_l.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5583239/frame_dials_l.txt)

[frame_dials_r.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5583238/frame_dials_r.txt)

jbigler, 2021-01-01:
I also just purchased this tablet and am looking to get it working with Linux. 
I installed the latest driver and configured a simple xorg.conf file to identify it as a wacom tablet.
After doing so and using xsetwacom to map it to my XP-Pen Artist 22R Pro screen coordinates, I was able to use the pen with pressure and tilt(!) support as well as the lower button on the pen. Nothing else seems to be currently mappable using xsetwacom.

I will add my diagnostic results as well. There is a problem with the descriptors.txt file provided by @Draghi because it was made after the probe. That causes the USB ID to be set to "ID 28bd:1227 XP-Pen TABLET" which is not correct. Unplugging power from the unit is the only way to return to the correct result which is: "ID 28bd:091b XP-Pen 21.5 inch PenDisplay".

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5759050/descriptors.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5759051/probe.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5759052/pen_tilt.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5759053/pen_pressure.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5759054/pen_coords.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5759055/pen_buttons.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5759056/hid_report_descriptors.txt)
[frame_dial_right.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5759057/frame_dial_right.txt)
[frame_dial_left.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5759058/frame_dial_left.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5759059/frame_buttons.txt)

jbigler, 2021-02-09:
After trying to use the tablet using libinput vs the wacom driver in X11, I've determined that the wacom driver provides the best experience, especially in regards to the smoothness of its tilt support. However, the DIGImend kernel driver does not allow the wacom driver to see the frame/pad, which makes it very difficult to remap the buttons to be more usable. I tried creating custom HID descriptors and recompiling the drivers, but I don't have enough experience with it. My attempts did nothing but cause the while tablet to disappear.
After researching through the various issues and pull request that have been submitted, I hacked together the following solution to use until there is official support added to DIGImend.

First I had to create the following file: `/etc/X11/xorg.conf.d/30-xppen.conf`
```
Section "InputClass"
        Identifier "XP-Pen Frame"
        MatchUSBID "28bd:091b"
        MatchIsKeyboard "on"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
EndSection

Section "InputClass"
        Identifier "XP-Pen Tablet"
        MatchIsTablet "on"
        Driver "wacom"
        MatchUSBID "28bd:091b"
        MatchDevicePath "/dev/input/event*"
EndSection
```
This provides the best stylus support via the wacom driver while still allowing the frame buttons to produce their default actions by using libinput.

The stylus has the problem of no response to the upper half of the rocker switch. To fix that problem, I created an entry in the hwdb by creating the following file: `/etc/udev/hwdb.d/90-xppen.hwdb`
```
evdev:input:b0003v28BDp091Be0100-e0*
 KEYBOARD_KEY_d0045=0x14c
```
This allows the upper button to act as a the default 2nd stylus button, usually equivalent to a right-click.

Unfortunately, using the hwdb to remap the buttons is difficult because many of the buttons submit multiple scancodes per press, which hwdb doesn't support. This [comment](https://github.com/DIGImend/digimend-kernel-drivers/issues/474#issuecomment-734438415) on another tablet with a similar problem has provided an interesting solution that I may implement if necessary. 

My primary art program is Krita which has very configurable keyboard shortcuts, so just knowing what the buttons are sending to the system makes it easier to map them, especially since Krita allows setting up alternate shortcuts, so the defaults don't have to be lost.

Here is a list of the scancodes and keycodes sent by the frame buttons.
```
Left side:
01. b             70005 / code 48 (KEY_B)
02. e             70008 / code 18 (KEY_E)
03. alt           700e2 / code 56 (KEY_LEFTALT)
04. space         7002c / code 57 (KEY_SPACE)
05. ctrl+S        700e0 / code 29 (KEY_LEFTCTRL) + 70016 / code 31 (KEY_S)

LDL: ctrl+-       700e0 / code 29 (KEY_LEFTCTRL) + 70056 / code 74 (KEY_KPMINUS)
LDR: ctrl++       700e0 / code 29 (KEY_LEFTCTRL) + 70057 / code 78 (KEY_KPPLUS)

06. ctrl+z        700e0 / code 29 (KEY_LEFTCTRL) + 7001d / code 44 (KEY_Z)
07. ctrl+alt+z    700e0 / code 29 (KEY_LEFTCTRL) + 700e2 / code 56 (KEY_LEFTALT) + 7001d / code 44 (KEY_Z)
08. ctrl+shift+z  700e0 / code 29 (KEY_LEFTCTRL) + 700e1 / code 42 (KEY_LEFTSHIFT) + 7001d / code 44 (KEY_Z)
09. v             70019 / code 47 (KEY_V)
10. l             7000f / code 38 (KEY_L)

Right side:
11. ctrl+o        700e0 / code 29 (KEY_LEFTCTRL) + 70012 / code 24 (KEY_O)
12. ctrl+n        700e0 / code 29 (KEY_LEFTCTRL) + 70011 / code 49 (KEY_N)
13. ctrl+shift+n  700e0 / code 29 (KEY_LEFTCTRL) + 700e1 / code 42 (KEY_LEFTSHIFT) + 70011 (KEY_N)
14. ctrl+e        700e0 / code 29 (KEY_LEFTCTRL) + 70008 / code 18 (KEY_E)
15. f             70009 / code 33 (KEY_F)

RDL: [            7002f / code 26 (KEY_LEFTBRACE)
RDR: ]            70030 / code 27 (KEY_RIGHTBRACE)

16. d             70007 / code 32 (KEY_D)
17. x             7001b / code 45 (KEY_X)
18. ctrl+delete   700e0 / code 29 (KEY_LEFTCTRL) + 7004c / code 111 (KEY_DELETE)
19. ctrl+c        700e0 / code 29 (KEY_LEFTCTRL) + 70006 / code 46 (KEY_C)
20. ctrl+v        700e0 / code 29 (KEY_LEFTCTRL) + 70019 / code 47 (KEY_V)
```

kurikaesu, 2021-08-22:
Try the latest code I pushed in my PR: https://github.com/DIGImend/digimend-kernel-drivers/pull/557
I have the 22R Pro and all the controls are enabled however the right dial is currently unmappable.

Left side buttons can be handled by the wacom driver but the right side buttons will need custom mapping. Read through the PR correspondence to see how I mapped mine.

redstrate, 2022-12-24:
I've decided to pick back up the work on this tablet, since I own one and I would like to use it without the help of [userspace-tablet-daemon](https://github.com/kurikaesu/userspace-tablet-driver-daemon) (even though it's an nice project!)

While `hid_generic` handles the tablet wonderfully, `uclogic` is still needed in order for the pen to fully function. The PR linked is unfortunately incomplete and potentially dropped? (it's missing some weird defs, like one missing enum and couple of externs) None of the buttons on the tablet itself work and one of the dials is non-functional, but it's a start.

redstrate, 2022-12-26:
I cleaned up the driver by @kurikaesu and submitted it upstream! You can find it on the list here: https://lore.kernel.org/linux-input/2068502.VLH7GnMWUR@adrastea/T/#u