DIGImend/digimend-kernel-drivers #486: Huion H610: detected but not working

zenny, 2020-11-29:
Hi, 

I am on voidlinux 5.4 LTS stable kernel. I built kernel modules from this master repo (also tried with  https://github.com/Huion-Linux/DIGImend-kernel-drivers-for-Huion which lacks some updates), the H610 gets detected, but says `interface is invalid` and simply fails to work:

```
# dmesg | grep :256C
[    4.635184] input: HUION H610 as /devices/pci0000:00/0000:00:12.2/usb1/1-2/1-2.4/1-2.4.3/1-2.4.3:1.0/0003:256C:006E.0007/input/input13
[    4.636633] input: HUION H610 Mouse as /devices/pci0000:00/0000:00:12.2/usb1/1-2/1-2.4/1-2.4.3/1-2.4.3:1.0/0003:256C:006E.0007/input/input14
[    4.638148] hid-generic 0003:256C:006E.0007: input,hidraw6: USB HID v1.11 Mouse [HUION H610] on usb-0000:00:12.2-2.4.3/input0
[    4.642188] input: HUION H610 Mouse as /devices/pci0000:00/0000:00:12.2/usb1/1-2/1-2.4/1-2.4.3/1-2.4.3:1.1/0003:256C:006E.0008/input/input15
[    4.643666] input: HUION H610 as /devices/pci0000:00/0000:00:12.2/usb1/1-2/1-2.4/1-2.4.3/1-2.4.3:1.1/0003:256C:006E.0008/input/input16
[    4.645188] hid-generic 0003:256C:006E.0008: input,hiddev1,hidraw7: USB HID v1.11 Mouse [HUION H610] on usb-0000:00:12.2-2.4.3/input1
[    4.647855] input: HUION H610 Keyboard as /devices/pci0000:00/0000:00:12.2/usb1/1-2/1-2.4/1-2.4.3/1-2.4.3:1.2/0003:256C:006E.0009/input/input17
[    4.700585] input: HUION H610 Consumer Control as /devices/pci0000:00/0000:00:12.2/usb1/1-2/1-2.4/1-2.4.3/1-2.4.3:1.2/0003:256C:006E.0009/input/input18
[    4.702104] input: HUION H610 System Control as /devices/pci0000:00/0000:00:12.2/usb1/1-2/1-2.4/1-2.4.3/1-2.4.3:1.2/0003:256C:006E.0009/input/input19
[    4.703735] hid-generic 0003:256C:006E.0009: input,hidraw8: USB HID v1.11 Keyboard
[HUION H610] on usb-0000:00:12.2-2.4.3/input2
[    4.725633] input: HUION H610 as /devices/pci0000:00/0000:00:12.2/usb1/1-2/1-2.4/1-2.4.3/1-2.4.3:1.0/0003:256C:006E.0007/input/input20
[    4.727258] input: HUION H610 Pad as /devices/pci0000:00/0000:00:12.2/usb1/1-2/1-2.4/1-2.4.3/1-2.4.3:1.0/0003:256C:006E.0007/input/input21
[    4.728917] uclogic 0003:256C:006E.0007: input,hiddev2,hidraw6: USB HID v1.11 Keypad [HUION H610] on usb-0000:00:12.2-2.4.3/input0
[    4.745795] input: HUION H610 Mouse as /devices/pci0000:00/0000:00:12.2/usb1/1-2/1-2.4/1-2.4.3/1-2.4.3:1.1/0003:256C:006E.0008/input/input22
[    4.747521] uclogic 0003:256C:006E.0008: input,hiddev1,hidraw7: USB HID v1.11 Mouse [HUION H610] on usb-0000:00:12.2-2.4.3/input1
[    4.788623] uclogic 0003:256C:006E.0009: interface is invalid, ignoring
[   10.501233] uclogic 0003:256C:006E.0009: interface is invalid, ignoring

```

Any input? 

Cheers and stay safe,

tcharlss, 2020-12-04:
Same error for me on Ubuntu 19.10, Digimend installed from source with DKMS.

```
[    1.394778] input: HUION PenTablet as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb5/5-1/5-1:1.0/0003:256C:006E.0001/input/input2
[    1.394821] input: HUION PenTablet Mouse as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb5/5-1/5-1:1.0/0003:256C:006E.0001/input/input3
[    1.394857] hid-generic 0003:256C:006E.0001: input,hidraw0: USB HID v1.11 Mouse [HUION PenTablet] on usb-0000:08:00.3-1/input0
[    1.398389] input: HUION PenTablet Mouse as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb5/5-1/5-1:1.1/0003:256C:006E.0002/input/input4
[    1.398436] input: HUION PenTablet as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb5/5-1/5-1:1.1/0003:256C:006E.0002/input/input5
[    1.398502] hid-generic 0003:256C:006E.0002: input,hiddev0,hidraw1: USB HID v1.11 Mouse [HUION PenTablet] on usb-0000:08:00.3-1/input1
[    1.398598] input: HUION PenTablet Keyboard as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb5/5-1/5-1:1.2/0003:256C:006E.0003/input/input6
[    1.461290] input: HUION PenTablet Consumer Control as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb5/5-1/5-1:1.2/0003:256C:006E.0003/input/input7
[    1.461319] input: HUION PenTablet System Control as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb5/5-1/5-1:1.2/0003:256C:006E.0003/input/input8
[    1.461353] hid-generic 0003:256C:006E.0003: input,hidraw2: USB HID v1.11 Keyboard [HUION PenTablet] on usb-0000:08:00.3-1/input2
[    4.828463] input: HUION PenTablet as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb5/5-1/5-1:1.0/0003:256C:006E.0001/input/input21
[    4.828521] input: HUION PenTablet Pad as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb5/5-1/5-1:1.0/0003:256C:006E.0001/input/input22
[    4.828576] uclogic 0003:256C:006E.0001: input,hiddev2,hidraw0: USB HID v1.11 Keypad [HUION PenTablet] on usb-0000:08:00.3-1/input0
[    4.971388] input: HUION PenTablet Mouse as /devices/pci0000:00/0000:00:08.1/0000:08:00.3/usb5/5-1/5-1:1.1/0003:256C:006E.0002/input/input23
[    4.971455] uclogic 0003:256C:006E.0002: input,hiddev0,hidraw1: USB HID v1.11 Mouse [HUION PenTablet] on usb-0000:08:00.3-1/input1
[    5.375206] uclogic 0003:256C:006E.0003: interface is invalid, ignoring
[    5.468017] uclogic 0003:256C:006E.0003: interface is invalid, ignoring
```

It used to work previously on the same machine on Ubuntu 19.04.

who8mypnuts, 2020-12-17:
I have the H610 Pro V2 and I need to run the .deb file after every restart to make my tablet connect.
Prior to running the deb, it shows up with lsusb, but doesn't function.

After running the .deb, it instantly connects and works smooth as silk.
Running Kubuntu 20.10.

JoseExposito, 2024-05-03:
Hi!

Your issue should be fixed by [DIGImend v13](https://github.com/DIGImend/digimend-kernel-drivers/releases/tag/v13). Please, if you have a chance, give it a try and report any problems found.

Thanks!

closed 2024-05-03 by JoseExposito