DIGImend/digimend-kernel-drivers #488: XP-Pen Star 06C

Fe-Ti, 2020-12-04:
Recently I've got a [XP-Pen Star 06C](https://www.xp-pen.com/download-53.html). It's not mentioned in the list of known tablets, so there are some diagnostics attached to the issue (to be correct they were uploaded to Google Drive).
**Collected diagnostics: [Star06C_V2_Diagnostics.zip](https://drive.google.com/file/d/19hf9n8BIQWDQx_SMYcHWzkVijajp-wRk/view?usp=sharing)**

P.S.
The archive contains several repetitions for pen_coords,pen_buttons and pen_pressure.

P.P.S.
I know that tablet does not work in Linux and Windows (except its dial and 6 buttons), but somehow it does work in Haiku and (at least) in one UEFI BIOS. But when I was collecting data, I found some strange magic. After running `uclogic-probe` with root privileges I discovered that the tablet had begun operating properly.
(Note: without running uclogic-probe I can see that Debian shows three readable /dev/hidraw* files)

olekhov, 2020-12-12:
Hello, I've recently got a XP-Pen Star 06 tablet. The only difference is that it is wireless whereas yours is cabled.
I think the behavior is identical.
There is an [issue](https://github.com/DIGImend/digimend-kernel-drivers/pull/419) with PR that adds support for this tablet.
I wonder, why it does not work. 
I have the intention to make this tablet "supported".

Fe-Ti, 2020-12-12:
Hi!
Thanks for the reply!
Yes, I've read the PR and tried to make similar changes. But, unfortunately, whatever I changed, Star 06C V2 was working incorrectly (it was detected as a touchscreen). So after two or three weeks I gave up on trying to change the source code by myself.
Currently I'm using udev for running a shell script, which in turn runs uclogic-probe. It's a kludge, but it works tolerably.
(Also I thought to write a user space program for reading input from /dev/hidraw*)

josephbburg, 2021-06-22:
Hello. I too am trying to get the Star 6 Wireless to work with DIGImend. I have tried to build the driver using DKMS on Manjaro - I checked out the tag v10 and then cherry-picked the PR adding support for Star 6, then built this following the instructions in the readme. There were no errors, as far as I could tell. But the Star wasn't detected by `xsetwacom --list`no matter what I did in the configuration settings. I am going to try the xppen drivers for now, but I want to help add support for the Star 6 Wireless however I can.

hendursaga, 2023-06-02:
I have an XP-Pen Star 06. I'm presuming it's the C version, since it's USB. I'll see how far I can get this to work.

Fe-Ti, 2023-06-02:
@hendursaga, I've tried to make it work. The easy part is getting the surface and pen buttons to function properly (just add several lines of code like I did in PR). The problem was in getting other buttons and a ring to work. I've tried to poke around, but due to the lack of knowledge I've failed.
So I wish you'll succeed better :)