DIGImend/digimend-kernel-drivers #492: Is Gaomon M10K supported?

tuket, 2020-12-10:
Hello, I was just wondering if the [Gaomon M10K](https://www.gaomon.net/Pen_Tablet/M10K_2018.html) is currently supported. I'm planning to buy one.
I haven't seen it in the [list of suported tablets](http://digimend.github.io/drivers/digimend/tablets/) but the S620, for example, is not listed and [seems to be supported](https://github.com/DIGImend/digimend-kernel-drivers/issues/376).
Thanks

ferresb, 2021-01-04:
Hi,

I just got the Gaomon M10K 2018, and managed to use it using digimend kernels.

It is even compatible with Wacom X.Org drivers to customize button use and more.

Hope this helps,

tuket, 2021-01-04:
Thanks for your comment!
I ended up buying the S620 because I was afraid it wouldn't work.
Anyways, I'm pretty sure this will be helpful to other people :)

closed 2021-01-04 by tuket

User-green, 2025-01-27:
> Dear sir, Sorry to disturb you. Would you mind sharing about VID:PID of Gaomon M10K 2018 or M10K? Now I have the similar product Gaomon M6 with 256c:0064, but it is not workable. I still try to find an available method to drive it......Thanks a lot.

I have M10K 2018, what exactly is needed?