DIGImend/digimend-kernel-drivers #493: Hovering issue with Huion 950p

papadako, 2020-12-16:
Hello,

I have a hovering problem with my Huion 950p, using the latest git code of the digimend-kernel-drivers on 5.10 and Ubuntu 20.10.
I had the same problem with older kernels on 20.04.

dmesg output

```
[ 2797.084810] usb 3-10.3.1: new full-speed USB device number 16 using xhci_hcd
[ 2797.189534] usb 3-10.3.1: New USB device found, idVendor=0416, idProduct=3f00, bcdDevice= 0.00
[ 2797.189538] usb 3-10.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2797.189541] usb 3-10.3.1: Product: WPM USB
[ 2797.189543] usb 3-10.3.1: Manufacturer: Nuvoton
[ 2797.194619] input: Nuvoton WPM USB as /devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10.3/3-10.3.1/3-10.3.1:1.0/0003:0416:3F00.0011/input/input58
[ 2797.253006] hid-generic 0003:0416:3F00.0011: input,hidraw3: USB HID v1.10 Device [Nuvoton WPM USB] on usb-0000:00:14.0-10.3.1/input0
```
My tablet is recognized as a Nuvoton. Is this correct?

modinfo-hidlogic output
`filename:       /lib/modules/5.10.1-051001-generic/extra/hid-uclogic.ko
version:        10
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     57AA0864931B0963BD47BF3
alias:          hid:b0003g*v000028BDp00000078
alias:          hid:b0003g*v000028BDp00000042
alias:          hid:b0003g*v000028BDp00000094
alias:          hid:b0003g*v000028BDp00000075
alias:          hid:b0003g*v000028BDp00000055
alias:          hid:b0003g*v000028BDp00000071
alias:          hid:b0003g*v000028BDp00000074
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000047
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006D
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid,usbhid
retpoline:      Y
name:           hid_uclogic
vermagic:       5.10.1-051001-generic SMP mod_unload `

Best

nic3-14159, 2020-12-16:
Hello! I don't think it is supposed to be recognized as Nuvoton. Huion tablets usually have an idVendor of 256c, but that device has an idVendor of 0416, which corresponds to Nuvoton. If you unplug the tablet, does dmesg indicate that a USB device was disconnected? And when you plug it back in, does it show up again as the Nuvoton device? Also, does the tablet show up in the output of `lsusb`?

papadako, 2020-12-17:
Hello,

when I disconnect the device it is disconnected and when I connect it again it appears as a Nuvoton.

lsusb is a bit confusing. In my first plug-unplug processes it reported 

`Bus 003 Device 020: ID 0416:3f00 Winbond Electronics Corp. WPM USB`

but for all the rest plug-uplugs I did it reports

`Bus 003 Device 023: ID 256c:006d
`

Notice that now dmesg reports (one line before the end):

`[81748.035293] usb 3-10.3.1: new full-speed USB device number 27 using xhci_hcd
[81748.167422] usb 3-10.3.1: New USB device found, idVendor=256c, idProduct=006d, bcdDevice= 0.00
[81748.167427] usb 3-10.3.1: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[81748.182470] input: HID 256c:006d as /devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10.3/3-10.3.1/3-10.3.1:1.0/0003:256C:006D.0021/input/input89
[81748.182690] input: HID 256c:006d Pad as /devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10.3/3-10.3.1/3-10.3.1:1.0/0003:256C:006D.0021/input/input90
[81748.182832] input: HID 256c:006d Touch Strip as /devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10.3/3-10.3.1/3-10.3.1:1.0/0003:256C:006D.0021/input/input91
[81748.182968] input: HID 256c:006d Dial as /devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10.3/3-10.3.1/3-10.3.1:1.0/0003:256C:006D.0021/input/input92
[81748.183204] uclogic 0003:256C:006D.0021: input,hiddev1,hidraw3: USB HID v1.11 Keypad [HID 256c:006d] on usb-0000:00:14.0-10.3.1/input0
**[81748.187911] uclogic 0003:256C:006D.0022: No inputs registered, leaving**
[81748.187994] uclogic 0003:256C:006D.0022: hidraw4: USB HID v1.11 Device [HID 256c:006d] on usb-0000:00:14.0-10.3.1/input1
`

papadako, 2020-12-23:
Any suggestions for fixing this?

papadako, 2020-12-30:
Just adding to the issue it seems that the device is initially recognized as a Nuvoton, then it disconnects and connects as a 256c vendor device.

`[1211418.295101] usb 3-4: new full-speed USB device number 45 using xhci_hcd
[1211418.443716] usb 3-4: New USB device found, idVendor=0416, idProduct=3f00, bcdDevice= 0.00
[1211418.443722] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[1211418.443724] usb 3-4: Product: WPM USB
[1211418.443726] usb 3-4: Manufacturer: Nuvoton
[1211418.444617] input: Nuvoton WPM USB as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:0416:3F00.003B/input/input140
[1211418.503436] hid-generic 0003:0416:3F00.003B: input,hidraw3: USB HID v1.10 Device [Nuvoton WPM USB] on usb-0000:00:14.0-4/input0
[1211420.429931] usb 3-4: USB disconnect, device number 45
[1211420.795134] usb 3-4: new full-speed USB device number 46 using xhci_hcd
[1211420.970497] usb 3-4: New USB device found, idVendor=256c, idProduct=006d, bcdDevice= 0.00
[1211420.970502] usb 3-4: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[1211420.976421] hid-generic 0003:256C:006D.003C: hiddev1,hidraw3: USB HID v1.11 Device [HID 256c:006d] on usb-0000:00:14.0-4/input0
[1211420.980670] input: HID 256c:006d as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.1/0003:256C:006D.003D/input/input141
[1211420.980833] hid-generic 0003:256C:006D.003D: input,hidraw4: USB HID v1.11 Device [HID 256c:006d] on usb-0000:00:14.0-4/input1
[1211421.017677] input: HID 256c:006d as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:256C:006D.003C/input/input142
[1211421.017780] input: HID 256c:006d Pad as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:256C:006D.003C/input/input143
[1211421.017857] input: HID 256c:006d Touch Strip as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:256C:006D.003C/input/input144
[1211421.017942] input: HID 256c:006d Dial as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:256C:006D.003C/input/input145
[1211421.018104] uclogic 0003:256C:006D.003C: input,hiddev1,hidraw3: USB HID v1.11 Keypad [HID 256c:006d] on usb-0000:00:14.0-4/input0
[1211421.103696] uclogic 0003:256C:006D.003D: No inputs registered, leaving
[1211421.103860] uclogic 0003:256C:006D.003D: hidraw4: USB HID v1.11 Device [HID 256c:006d] on usb-0000:00:14.0-4/input1`

papadako, 2021-02-24:
Can anyone provide any hint/comment/suggestion on how to resolve this?

The pen works as expected in windows so it is not a hardware problem.

papadako, 2021-03-03:
Dear all,

I am closing this. The hovering problem I had was with the kwin mouse mark desktop effect. But in the openboard the pen works as expected.

closed 2021-03-03 by papadako