DIGImend/digimend-kernel-drivers #495: UGEE m708 working "out of the box" but ignoring all xsetwacom commands

dabrowskiw, 2020-12-19:
Hi everyone!

I have a problem with my UGEE m708 on Ubuntu 20.10 under KDE. It works out of the box as in: Pressure sensitivity works and the buttons send commands (e.g. zoom, undo). However, I need to be able to configure it, at minimum to limit it to one screen, preferably also to customize what the buttons do. And here's where the problem lies: It ignores everything I throw at it with xsetwacom. 

In lsusb, it is recognized as ```Bus 001 Device 002: ID 056a:51bf Wacom Co., Ltd Pen and multitouch sensor```. ```xsetwacom --list devices``` says:

```
Wacom Pen and multitouch sensor Finger touch    id: 9   type: TOUCH     
Wacom Pen and multitouch sensor Pen stylus      id: 10  type: STYLUS    
Wacom Pen and multitouch sensor Pen eraser      id: 19  type: ERASER    
```

Just one example of xsetwacom doing nothing: 
```
~$ xsetwacom --get 10 all 2> /dev/null | grep "Rotate"
Option "Rotate" "none"
wojtek@x1:~$ xsetwacom set "10" Rotate "cw"
wojtek@x1:~$ xsetwacom --get 10 all 2> /dev/null | grep "Rotate"
Option "Rotate" "cw"
```

However, nothing whatsoever changes - all pen movement on the tablet still results in the same cursor movement as with "Rotate" set to "none". 

By the way, the tablet is also recognized in the KDE "Tablet" settings, and changing anything there has no effect either...

Any ideas? It would be really useful to have this working...

Popolon, 2021-01-12:
Did you try this?: https://digimend.github.io/support/howto/drivers/wacom/

teras, 2021-09-27:
@Popolon I think this link is not appropriate since it references a waltop device, which (according to the USB IDs) this isn't (?).
It is such a pity that M708 V2 almost works (also reported here https://github.com/DIGImend/digimend-kernel-drivers/issues/469) but most configuration is missing, i.e. can't set the third button or the keyboard events.

Using no drivers at all, the device produces some random keyboard events anyway (like saving, zoom in/out etc), but not being able to configure it.

nic3-14159, 2021-09-27:
@dabrowskiw Those USB ids don't look right. That ID is for a Wacom device, and seems to be the pen digitizer on the Thinkpad X1 Yoga. From the [DIGImend tablets page](https://digimend.github.io/tablets/), the ID for the Ugee M708 should either be 5543:0081 (M708 v1) or 28bd:0907 (M708 v2). Are you sure those are the right IDs?

dabrowskiw, 2021-09-27:
@nic3-14159 Hmm, I seem to remember that these reproducibly went away when I unplugged the device and came back when I plugged it back in. Then again, I was using an X1 Yoga at the time, so I might be wrong. However, I am not able too verify this - in the meantime, I have sold that tablet and I got a Huion that does work perfectly with the digimend drivers. I guess at this point it's unlikely that anything that will help others can happen on this issue, so I'll just close it. Sorry for the trouble and thanks for your great work on digimend!

closed 2021-09-27 by dabrowskiw