DIGImend/digimend-kernel-drivers #496: [BUG] Huion H1060P - Express Buttons not working as intended

J2xWo, 2020-12-22:
Hi all,
everything seems to be working besides the Express buttons.
I would imagine that the layout is like this:
```
 [1]  [2]                                 [1]  [2]
 [3]  [4]                                 [3]  [8]
 [5]  [6]     but instead its like this:  [9] [10]
 [7]  [8]                                [11] [12]
 [9] [10]                                 [ ]  [ ]
[11] [12]                                 [ ]  [ ]
```
The last four buttons don't write anything at all
When checking each button with `xsetwacom --get 'HUION Huion Tablet Pad pad' Button button-number`, I get this:
```
❯ xsetwacom --get 'HUION Huion Tablet Pad pad' Button 1
key +1 -1 
❯ xsetwacom --get 'HUION Huion Tablet Pad pad' Button 2
key +2 -2 
❯ xsetwacom --get 'HUION Huion Tablet Pad pad' Button 3
key +3 -3 
❯ xsetwacom --get 'HUION Huion Tablet Pad pad' Button 4
4
❯ xsetwacom --get 'HUION Huion Tablet Pad pad' Button 5
5
❯ xsetwacom --get 'HUION Huion Tablet Pad pad' Button 6
6
❯ xsetwacom --get 'HUION Huion Tablet Pad pad' Button 7
7
❯ xsetwacom --get 'HUION Huion Tablet Pad pad' Button 8
key +8 -8 
❯ xsetwacom --get 'HUION Huion Tablet Pad pad' Button 9
key +9 -9 
❯ xsetwacom --get 'HUION Huion Tablet Pad pad' Button 10
key +0 -0 
❯ xsetwacom --get 'HUION Huion Tablet Pad pad' Button 11
key +a -a 
❯ xsetwacom --get 'HUION Huion Tablet Pad pad' Button 12
key +b -b 
```
I'm not sure why its returning `key +X -X` for some and for others just `X`. Otherwise it looks how I configured it via the graphic tablet settings in the system settings. Maybe the issue is that the graphics tablet went through three upgrades? From the 1060pro -> New 1060plus -> H1060P. I got the latest one. And according to the supported tablets, the one listed is New 1060 Plus(which had a battery powered pen and SD card slot I believe. I don't know what other differences it has), so one generation before the one I got.
How do I go about solving this issue I have? Thanks in advance! 
Oh also how do I even use/configure the 16 soft buttons at the top? I assume manually via `xsetwacom --set` since they aren't listed in the settings?
```
❯ xsetwacom --list
HUION Huion Tablet Touch Strip pad      id: 19  type: PAD       
HUION Huion Tablet stylus               id: 20  type: STYLUS    
HUION Huion Tablet Pad pad              id: 21  type: PAD  
```

Duduz7, 2021-01-22:
As mentioned in the READ.me section, you can't configure keys 4-7 since they are reserved to be used as the up, down, right, and left arrows

shinich123, 2024-10-08:
for me, Its

[A] [B]                          ---                -------------  -------    ---       -----  [A] [A]
[C] [D]                             ---      -------------  -------       ---    -----    [B] [B]
[E] [F]        ---   but instead its like this:           -----        [C] [C]
[G] [H]          ---               -------------  -------      ---     -----    [D] [D]
[I] [J]                   ---              -------------  -------    ---    -   ----- [E] [G]
[K] [L]                   ---                -------------  -------    ---       -----      [F] [H]